OpenZWave/open-zwave #2063: Domoticz : OZW-driver Segmentation fault

candrea77, 2020-01-08:
Hello,
I'm trying to test my system (Domoticz) against the beta-release with ozw 1.6, but I'm running into some segmentation fault related to OZW-Driver.
I've opened issue in domoticz/domoticz repository, but i really do not know if this issue is specifically related to open-zwave or to domoticz.

I've tryied both pre-compiled binary from domoticz website (updating my stable system) , and also with a system fully compiled from source.

In both case I run into segmentation fault on a specific OZW device (i've got more of that type, but it crash anytime system interrogate that specific Device ID 105).
Device is a FIBARO System FGMS001 Motion Sensor | ID 0x1001 | Type 0x0800

I'm not a coder so I can't help you so much.

Here is the related log : let me know in case I've to provide more information or do some other tests.

Regards

```
2020-01-05 23:14:16.271 Always,
2020-01-05 23:14:17.027 Detail, Node105,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x69, 0x03, 0x30, 0x03, 0x00, 0xab
2020-01-05 23:14:17.027 Detail,
2020-01-05 23:14:17.027 Info, Node105, Received SensorBinary report: State=Off
2020-01-05 23:14:17.027 Detail, Node105, Initial read of value
2020-01-05 23:14:17.028 Detail, Node105, Notification: ValueChanged
2020-01-05 23:14:17.268 Detail, Node105,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x69, 0x03, 0x30, 0x03, 0x00, 0xab
2020-01-05 23:14:17.269 Detail,
2020-01-05 23:14:17.269 Info, Node105, Received SensorBinary report: State=Off
2020-01-05 23:14:17.269 Detail, Node105, Refreshed Value: old value=false, new value=false, type=bool
2020-01-05 23:14:17.269 Detail, Node105, Changes to this value are not verified
2020-01-05 23:14:17.269 Detail, Node105, Notification: ValueChanged
2020-01-05 23:14:17.358 Detail, Node105,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x69, 0x03, 0x30, 0x03, 0x00, 0xab
2020-01-05 23:14:17.358 Detail,
2020-01-05 23:14:17.358 Info, Node105, Received SensorBinary report: State=Off
2020-01-05 23:14:17.358 Detail, Node105, Refreshed Value: old value=false, new value=false, type=bool
2020-01-05 23:14:17.358 Detail, Node105, Changes to this value are not verified
2020-01-05 23:14:17.358 Detail, Node105, Notification: ValueChanged
2020-01-05 23:14:17.449 Detail, Node105,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x69, 0x03, 0x30, 0x03, 0x00, 0xab
2020-01-05 23:14:17.449 Detail,
2020-01-05 23:14:17.449 Info, Node105, Received SensorBinary report: State=Off
2020-01-05 23:14:17.449 Detail, Node105, Refreshed Value: old value=false, new value=false, type=bool
2020-01-05 23:14:17.449 Detail, Node105, Changes to this value are not verified
2020-01-05 23:14:17.449 Detail, Node105, Notification: ValueChanged
2020-01-05 23:14:18.715 Detail, Node105,   Received: 0x01, 0x09, 0x00, 0x04, 0x10, 0x69, 0x03, 0x30, 0x03, 0x00, 0xbb
2020-01-05 23:14:18.715 Detail,
2020-01-05 23:14:18.716 Info, Node105, Received SensorBinary report: State=Off
2020-01-05 23:14:18.716 Detail, Node105, Refreshed Value: old value=false, new value=false, type=bool
2020-01-05 23:14:18.716 Detail, Node105, Changes to this value are not verified
2020-01-05 23:14:18.716 Detail, Node105, Notification: ValueChanged
2020-01-05 23:14:18.753 Detail, Node105,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x69, 0x03, 0x20, 0x01, 0x00, 0xb9
2020-01-05 23:14:18.753 Detail,
2020-01-05 23:14:18.753 Info, Node105, Received Basic set from node 105: level=0.  Sending event notification.
2020-01-05 23:14:18.753 Detail, Node105, Notification: NodeEvent
2020-01-05 23:14:18.753  Error: Domoticz(pid:20823, tid:20845('OZW-driver')) received fatal signal 11 (Segmentation fault)
2020-01-05 23:14:18.753  Error: siginfo address=0x24, address=0x55b3ab46b17d
2020-01-05 23:14:18.765  Error: Failed to start gdb, will use backtrace() for printing stack frame

2020-01-05 23:14:18.769  Error: #0  ./domoticz :  + 0x363b36 [0x55b3aaf4eb36]
2020-01-05 23:14:18.769  Error: #1  ./domoticz : signal_handler(int, siginfo_t*, void*) + 0x2ad [0x55b3aaf4f69d]
2020-01-05 23:14:18.769  Error: #2  /lib/x86_64-linux-gnu/libpthread.so.0 :  + 0x12890 [0x7f297c0e4890]
2020-01-05 23:14:18.769  Error: #3  ./domoticz : OpenZWave::Node::GetValue(OpenZWave::ValueID const&) + 0x5d [0x55b3ab46b17d]
2020-01-05 23:14:18.769  Error: #4  ./domoticz : OpenZWave::Manager::IsValueValid(OpenZWave::ValueID const&) + 0x49 [0x55b3ab40f019]
2020-01-05 23:14:18.769  Error: #5  ./domoticz : COpenZWave::UpdateNodeEvent(OpenZWave::ValueID const&, int) + 0x3b [0x55b3ab1ed40b]
2020-01-05 23:14:18.769  Error: #6  ./domoticz : COpenZWave::OnZWaveNotification(OpenZWave::Notification const*) + 0x84f [0x55b3ab20863f]
2020-01-05 23:14:18.769  Error: #7  ./domoticz : OnNotification(OpenZWave::Notification const*, void*) + 0x31 [0x55b3ab2093f1]
2020-01-05 23:14:18.769  Error: #8  ./domoticz : OpenZWave::Manager::NotifyWatchers(OpenZWave::Notification*) + 0x8c [0x55b3ab40b46c]
2020-01-05 23:14:18.769  Error: #9  ./domoticz : OpenZWave::Driver::NotifyWatchers() + 0x82 [0x55b3ab44b422]
2020-01-05 23:14:18.769  Error: #10 ./domoticz : OpenZWave::Driver::DriverThreadProc(OpenZWave::Internal::Platform::Event*) + 0x348 [0x55b3ab458ea8]
2020-01-05 23:14:18.769  Error: #11 ./domoticz : OpenZWave::Internal::Platform::ThreadImpl::Run() + 0x13 [0x55b3ab4cebc3]
2020-01-05 23:14:18.769  Error: #12 ./domoticz : OpenZWave::Internal::Platform::ThreadImpl::ThreadProc(void*) + 0x9 [0x55b3ab4cebd9]
2020-01-05 23:14:18.769  Error: #13 /lib/x86_64-linux-gnu/libpthread.so.0 :  + 0x76db [0x7f297c0d96db]
2020-01-05 23:14:18.769  Error: #14 /lib/x86_64-linux-gnu/libc.so.6 : clone + 0x3f [0x7f297b43d88f]

```

Fishwaldo, 2020-01-09:
I've just pushed a potential fix - We added some additional debugging recently for another issue that could have caused this crash. Can you please try with the latest OZW version (not sure how its merged into domoticz)

candrea77, 2020-01-09:
Sure , I'll give it a try this evening ( gmt+1).
I will rebuild both ozw and domoticz in order to test it.

Thanks

petergebruers, 2020-01-09:
I wonder if there is something specific about your setup, eg something in Domoticz database or ozwcache... I cannot reproduce your crash. I've tested FGMS-001, OpenZwave Version 1.6-1006-gad05fecd (the one with the extra logging added by @Fishwaldo a few hours ago) and Domoticz V4.11605, Build Hash: 7de1f368f

This sensor reports motion and tamper through "Alarm Events", it is worth nothing that if you want the device called "Home Security" (of type Switch) to work in Domoticz, you'll have to enable legacy (sensor) reporting. You do that by adding "1" to group "4" on the OZW device setup page of Domoticz.

Fishwaldo, 2020-01-09:
Peter: does your test include a basic set? That’s what’s triggering the crash. Obviously something deeper went wrong, but with the fix, we should get a lot of what it’s looking for now.

petergebruers, 2020-01-09:
> Peter: does your test include a basic set?

No, did not think about sending a basic set to this device, being a sensor I am not sure what it is supposed to do. And In Domoticz I don't see a switch-like object that sends 'basic'. I have three of them but they are mapped to different CCs.

I can use ozwcp to do BasicCmd_Set, I see nothing special there...

```log
2020-01-09 10:34:31.938 Info, Node016, Sending (WakeUp) message (Callback ID=0x2a, Expected Reply=0x13) - BasicCmd_Set (Node=16): 0x01, 0x0a, 0x00, 0x13, 0x10, 0x03, 0x20, 0x01, 0x00, 0x25, 0x2a, 0xdb
2020-01-09 10:34:31.938 Info, Node016, Encrypted Flag is 0
2020-01-09 10:34:31.945 Detail, Node016,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-01-09 10:34:31.945 Detail, Node016,   ZW_SEND_DATA delivered to Z-Wave stack
2020-01-09 10:34:31.966 Detail, Node016,   Received: 0x01, 0x18, 0x00, 0x13, 0x2a, 0x00, 0x00, 0x02, 0x00, 0x7e, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0xa3
2020-01-09 10:34:31.966 Detail, Node016,   ZW_SEND_DATA Request with callback ID 0x2a received (expected 0x2a)
2020-01-09 10:34:31.966 Info, Node016, Request RTT 27 Average Request RTT 27
2020-01-09 10:34:31.966 Detail, Node016, Extended TxStatus: Time: 20, Hops: 0, Rssi: MAX --- --- --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 0 0 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
2020-01-09 10:34:31.966 Detail, Node016,   Expected callbackId was received
2020-01-09 10:34:31.966 Detail, Node016,   Expected reply was received
2020-01-09 10:34:31.966 Detail, Node016,   Message transaction complete
2020-01-09 10:34:31.966 Detail, 
2020-01-09 10:34:31.966 Detail, Node016, Removing current message
```

Fishwaldo, 2020-01-09:
@petergebruers ahh. I meant configuring the device to send basic set instead of something like binary sensor (or whatnot). 
OZW was crashing on a Node_Event which is triggered when we get a basic set from a device.

petergebruers, 2020-01-09:
Oh, right, that makes sense... To do that I have to add the controller to group 2...

`2nd association group – “Motion” is assigned to the motion sensor - sends Basic Set command frames to the associated devices.`

ozcwp ... 

```
2020-01-09 15:30:11.408 Detail, Node016,   Received: 0x01, 0x0b, 0x00, 0x04, 0x08, 0x10, 0x03, 0x20, 0x01, 0x00, 0x7e, 0x00, 0xb4
2020-01-09 15:30:11.408 Detail, 
2020-01-09 15:30:11.408 Info, Node016, Received Basic set from node 16: level=0. Treating it as a Basic report.
2020-01-09 15:30:11.408 Detail, Node016, Refreshed Value: old value=255, new value=0, type=byte
2020-01-09 15:30:11.409 Detail, Node016, Changes to this value are not verified
2020-01-09 15:30:11.409 Detail, Node016, Notification: ValueChanged
2020-01-09 15:30:11.409 Info, Notification: Value Changed Home 0xd96a77cd Node 16 Genre basic Class BASIC Instance 1 Index 0 Type byte IsSet True ValueAsString 0
```

I'll try that with domoticz...

petergebruers, 2020-01-09:
Wait a second... My ozwcache has `<SetAsReport>true</SetAsReport>` so not doing the "Sending event notification." part inBasic.cpp

Changed my ozwcp

`<SetAsReport>false</SetAsReport>`

Nothing special happens running ozwcp but Domoticz now logs...

```
2020-01-09 16:02:11.174 Info, Node016, Received Basic set from node 16: level=0.  Sending event notification.
2020-01-09 16:02:11.175 Detail, Node016, Notification: NodeEvent
2020-01-09 16:02:11.175 Warning, Node016, Node::GetValue - Couldn't find ValueID in Store: HomeID: 0xd96a77cd, ValueID: (Id 0x0000000010000010, NodeID 16, Genre basic, CC 0x00, Instance 1, Index 0, Type bool)
```

petergebruers, 2020-01-09:
Stack trace tells me this code runs in Domoticz because of the NodeEvent:

```cpp
void COpenZWave::UpdateNodeEvent(const OpenZWave::ValueID& vID, int EventID)
{
	if (m_pManager == NULL)
		return;
	if (!m_pManager->IsValueValid(vID))
		return;
```

IMHO the latest commit avoids the crash, but I'm not sure if anything else is needed. @Fishwaldo  I don't see where it goes wrong but I suppose CC 0x00 isn't right?

BTW sorry about the crash... I added the "if (value->GetID().GetId() != _id.GetId())" that caused the crash, to diagnose a different issue. I assumed we could not get there without a valid ID but that clearly can happen!

candrea77, 2020-01-09:
> I wonder if there is something specific about your setup, eg something in Domoticz database or ozwcache... I cannot reproduce your crash. I've tested FGMS-001, OpenZwave Version 1.6-1006-gad05fecd [...]

I can clean the ozwcache for sure ..... I'm doing make right now.
I let you know asap.
The strange thing is that I've got some other FGMS-001 and this is the only one that made crash my system.

candrea77, 2020-01-09:
> I can clean the ozwcache for sure ..... I'm doing make right now.
A little things : domoticz have same issue if updated from stable and also if builded from source : in that case I suppose that ozwcache was empty ... isn't it ?

Currently system is running ... I let you know if crash or if domoticz finish node discovery

candrea77, 2020-01-09:
> I've just pushed a potential fix - We added some additional debugging recently for another issue that could have caused this crash. Can you please try with the latest OZW version (not sure how its merged into domoticz)

Exactly, when this potential fix was pushed? I do git clone&pull in 2020-01-05

candrea77, 2020-01-09:
Node discovery finished and system is not crashed.
I've also forced a node wakeup in order to read device settings and it work ok.

Do you need a copy of the full OZW Log ?

Fishwaldo, 2020-01-10:
> Stack trace tells me this code runs in Domoticz because of the NodeEvent:
> 
> ```c++
> void COpenZWave::UpdateNodeEvent(const OpenZWave::ValueID& vID, int EventID)
> {
> 	if (m_pManager == NULL)
> 		return;
> 	if (!m_pManager->IsValueValid(vID))
> 		return;
> ```
> 
> IMHO the latest commit avoids the crash, but I'm not sure if anything else is needed. @Fishwaldo I don't see where it goes wrong but I suppose CC 0x00 isn't right?
> 
> BTW sorry about the crash... I added the "if (value->GetID().GetId() != _id.GetId())" that caused the crash, to diagnose a different issue. I assumed we could not get there without a valid ID but that clearly can happen!

I have to double check - but I don’t think NodeEvent publishes a Value. So that code looks wrong.

petergebruers, 2020-01-10:
> I have to double check - but I don’t think NodeEvent publishes a Value. So that code looks wrong.

If you're sure about that I can open a PR on Domoticz to remove that IsValueValid test

> Do you need a copy of the full OZW Log ?

I don't need one.

candrea77, 2020-01-11:
Currently no more crash related to this issue.

closed 2020-01-11 by candrea77