OpenZWave/open-zwave #2070: Schlage Link (Wintop) Mini Keypad RFID 0x4501 0x6131 stopped working

candrea77, 2020-01-11:
Migrating from ozw 1.4-3335-g74e05982-dirty to 1.6-1004-g71220f43 cause Schlage Link (Wintop) Mini Keypad RFID to stop from working.
Currently, when I try the Arm or Disarm action, in Domoticz I didn't get any devices updated.

NOTE : I'm using domoticz as home automation system and due to my bad understandginds I'm not sure if this issue is related to ozw or to domoticz.
I can provide the relevant OZW Log in order to investigate the issue.
When using ozw1.6, In OZWCP the information window of the Schlage Link Node miss some relevant information (see attached images)

In OLD Zwave , when I switch the device state to ARM I get :
```
_2020-01-11 00:05:29.968 Detail, Node067,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x43, 0x0a, 0x71, 0x05, 0x00, 0xff, 0x00, 0xff, 0x06, 0x05, 0x01, 0x01, 0xd5
2020-01-11 00:05:29.968 Info, Node067, Received Alarm report: type=0, level=255, sensorSrcID=0, type:Access Control event:5, status=255
2020-01-11 00:05:29.969 Detail, Node067, Refreshed Value: old value=0, new value=0, type=byte
2020-01-11 00:05:29.969 Detail, Node067, Changes to this value are not verified
2020-01-11 00:05:29.969 Detail, Node067, Refreshed Value: old value=0, new value=255, type=byte
2020-01-11 00:05:29.969 Detail, Node067, Changes to this value are not verified
2020-01-11 00:05:29.969 Detail, Node067, Refreshed Value: old value=0, new value=0, type=byte
2020-01-11 00:05:29.969 Detail, Node067, Changes to this value are not verified
2020-01-11 00:05:29.969 Detail, Node067, Refreshed Value: old value=6, new value=5, type=byte
2020-01-11 00:05:29.969 Detail, Node067, Changes to this value are not verified
2020-01-11 00:05:29.969 Detail, Node067, Notification: ValueChanged
2020-01-11 00:05:29.969 Detail, Node067, Notification: ValueChanged
2020-01-11 00:05:29.969 Detail, Node067, Notification: ValueChanged
2020-01-11 00:05:29.969 Detail, Node067, Notification: ValueChanged
2020-01-11 00:05:30.151 Detail, Node067,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x43, 0x02, 0x84, 0x07, 0x31
2020-01-11 00:05:30.151 Info, Node067, Received Wakeup Notification from node 67
2020-01-11 00:05:30.151 Info, Node067,   Node 67 has been marked as awake
2020-01-11 00:05:30.151 Detail, Node067, Queuing (WakeUp) WakeUpCmd_NoMoreInformation (Node=67): 0x01, 0x09, 0x00, 0x13, 0x43, 0x02, 0x84, 0x08, 0x25, 0x0b, 0x06
2020-01-11 00:05:30.151 Detail, Node067, Notification: Notification - Node Awake
2020-01-11 00:05:30.152 Info, Node067, Sending (WakeUp) message (Callback ID=0x0b, Expected Reply=0x13) - WakeUpCmd_NoMoreInformation (Node=67): 0x01, 0x09, 0x00, 0x13, 0x43, 0x02, 0x84, 0x08, 0x25, 0x0b, 0x06
2020-01-11 00:05:30.157 Detail, Node067,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-01-11 00:05:30.157 Detail, Node067,   ZW_SEND_DATA delivered to Z-Wave stack
2020-01-11 00:05:30.254 Detail, Node067,   Received: 0x01, 0x07, 0x00, 0x13, 0x0b, 0x00, 0x00, 0x0a, 0xea
2020-01-11 00:05:30.254 Detail, Node067,   ZW_SEND_DATA Request with callback ID 0x0b received (expected 0x0b)
2020-01-11 00:05:30.254 Info, Node067, Request RTT 102 Average Request RTT 843
2020-01-11 00:05:30.254 Info, Node067,   Node 67 has been marked as asleep
2020-01-11 00:05:30.254 Detail, Node067, Removing current message
2020-01-11 00:05:30.254 Detail, Node067, Notification: Notification - Node Asleep
2020-01-11 00:05:30.433 Detail, Node067,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x43, 0x03, 0x80, 0x03, 0xff, 0xce
2020-01-11 00:05:30.433 Info, Node067, Received Battery report from node 67: level=0
2020-01-11 00:05:30.433 Detail, Node067, Refreshed Value: old value=0, new value=0, type=byte
2020-01-11 00:05:30.433 Detail, Node067, Changes to this value are not verified
2020-01-11 00:05:30.433 Detail, Node067, Notification: ValueChanged_
```

In the new OZW , I get this :
```
2020-01-11 07:41:35.765 Detail, Node067,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x43, 0x0a, 0x71, 0x05, 0x00, 0xff, 0x00, 0xff, 0x06, 0x05, 0x01, 0x01, 0xd5
2020-01-11 07:41:35.766 Info, Node067, Received Notification report (>v1): Type: Access Control (6) Event: Keypad Lock Operation (5) Status: true, Param Length: 1
2020-01-11 07:41:35.766 Warning, Node067, Couldn't Find ValueID_Index_Alarm::Type_ParamUserCodeid
2020-01-11 07:41:35.766 Warning, Node067, Couldn't Find a ValueList for Notification Type 6 (1)
2020-01-11 07:41:35.766 Info, Node067, Automatically Clearing Alarm in 5000ms
2020-01-11 07:41:35.920 Detail, Node067,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x43, 0x02, 0x84, 0x07, 0x31
2020-01-11 07:41:35.921 Info, Node067, Received Wakeup Notification from node 67
2020-01-11 07:41:35.921 Detail, Node067, Queuing (WakeUp) WakeUpCmd_NoMoreInformation (Node=67): 0x01, 0x09, 0x00, 0x13, 0x43, 0x02, 0x84, 0x08, 0x25, 0x77, 0x7a
2020-01-11 07:41:35.921 Info, Node067, Sending (WakeUp) message (Callback ID=0x77, Expected Reply=0x13) - WakeUpCmd_NoMoreInformation (Node=67): 0x01, 0x09, 0x00, 0x13, 0x43, 0x02, 0x84, 0x08, 0x25, 0x77, 0x7a
2020-01-11 07:41:35.921 Info, Node067, Encrypted Flag is 0
2020-01-11 07:41:35.926 Detail, Node067,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-01-11 07:41:35.926 Detail, Node067,   ZW_SEND_DATA delivered to Z-Wave stack
2020-01-11 07:41:36.037 Detail, Node067,   Received: 0x01, 0x07, 0x00, 0x13, 0x77, 0x00, 0x00, 0x0b, 0x97
2020-01-11 07:41:36.037 Detail, Node067,   ZW_SEND_DATA Request with callback ID 0x77 received (expected 0x77)
2020-01-11 07:41:36.037 Info, Node067, Request RTT 116 Average Request RTT 105
2020-01-11 07:41:36.037 Info, Node067,   Node 67 has been marked as asleep
2020-01-11 07:41:36.037 Detail, Node067,   Expected callbackId was received
2020-01-11 07:41:36.037 Detail, Node067,   Expected reply was received
2020-01-11 07:41:36.037 Detail, Node067,   Message transaction complete
2020-01-11 07:41:36.037 Detail, Node067, Removing current message
2020-01-11 07:41:36.038 Detail, Node067, Notification: Notification - Node Asleep
2020-01-11 07:41:36.085 Detail, Node067,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x43, 0x03, 0x80, 0x03, 0xff, 0xce
2020-01-11 07:41:36.086 Info, Node067, Received Battery report from node 67: level=0
2020-01-11 07:41:36.086 Detail, Node067, Refreshed Value: old value=0, new value=0, type=byte
2020-01-11 07:41:36.086 Detail, Node067, Changes to this value are not verified
2020-01-11 07:41:36.086 Detail, Node067, Notification: ValueChanged
2020-01-11 07:41:40.766 Warning, Node067, Couldn't Find a ValueList to ClearAlarm for Notification Type 6 (1)
2020-01-11 07:41:52.539 Info, Node067, Cache Save for Node 67 as its QueryStage_CacheLoad
```
[image]
[image]

Fishwaldo, 2020-01-11:
You need to ensure the device has completed the interview process after restarting. Your logs say it’s been sleeping. 

(Ensure QueryStage is complete.)

candrea77, 2020-01-11:
> You need to ensure the device has completed the interview process after restarting. Your logs say it’s been sleeping.
> 
> (Ensure QueryStage is complete.)

Hi Fishwaldo,
sorry , my fault.

Now I've done a "_Refresh Node Info_" from OZWCP and after a manual wakeup (device is battery operated) somethings has changed (attached the relevant OZW log).

Inside OZWCP I can see the additional information and despite is compiled with wrong value now keypad has start to work.
If you take a look to the attached image , the field "Enrollment code" now is empy, the field "Code1" present a "T*" and field Code 2 and 3 are empty.
Despite this, KeyPad till continue to recognize my tags without problem.

Functionality is different comparing old (domoticz stable with 1.4)  and new system (domoticz beta with 1.6).

In Domoticz I've got 2 devices : **_Alarm Pad Keypad_** and **_Alarm Level Keypad_** (both type **_Light/Switch_**)

With old version the physical action on keypad trigger a change on "Alarm Level Keypad" , now the change is triggered on "Alarm Pad Keypad".
I suppose this is not an issue , i've just adapted my script inside domoticz to use new mode.

Due to the fact now I can arm/disarm the alarm , I think I can leave it up and running for some days.
I've got more then hundred zwave devicies .... it will take a little bit to test all of'em

Let me know If I can close this issue or if you want to investigate deeper for wrong value inside OZWCP.

Thank you

```
2020-01-11 13:03:09.285 Detail, Node067, Reloading Node
2020-01-11 13:03:09.341 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.390 Detail, Node067, AdvanceQueries queryPending=0 queryRetries=0 queryStage=None live=1
2020-01-11 13:03:09.390 Detail, Node067, QueryStage_ProtocolInfo
2020-01-11 13:03:09.390 Detail, Node067, Queuing (Query) Get Node Protocol Info (Node=67): 0x01, 0x04, 0x00, 0x41, 0x43, 0xf9
2020-01-11 13:03:09.390 Detail, Node067, Queuing (Query) Query Stage Complete (ProtocolInfo)
2020-01-11 13:03:09.390 Info, Node067, Initializing Node. New Node: false (false)
2020-01-11 13:03:09.394 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.394 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.394 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.394 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.394 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.394 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.394 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.394 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: ValueRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: NodeRemoved
2020-01-11 13:03:09.395 Detail, Node067, Notification: NodeAdded
2020-01-11 13:03:09.400 Info, Node067, Sending (Query) message (Callback ID=0x00, Expected Reply=0x41) - Get Node Protocol Info (Node=67): 0x01, 0x04, 0x00, 0x41, 0x43, 0xf9
2020-01-11 13:03:09.400 Info, Node067, Encrypted Flag is 0
2020-01-11 13:03:09.401 Detail, Node067,   Received: 0x01, 0x09, 0x01, 0x41, 0x53, 0x9c, 0x00, 0x04, 0x40, 0x00, 0x3d
2020-01-11 13:03:09.401 Info, Node067, Received reply to FUNC_ID_ZW_GET_NODE_PROTOCOL_INFO
2020-01-11 13:03:09.402 Info, Node067,   Protocol Info for Node 67:
2020-01-11 13:03:09.402 Info, Node067,     Listening     = false
2020-01-11 13:03:09.402 Info, Node067,     Frequent      = false
2020-01-11 13:03:09.402 Info, Node067,     Beaming       = true
2020-01-11 13:03:09.402 Info, Node067,     Routing       = true
2020-01-11 13:03:09.402 Info, Node067,     Max Baud Rate = 40000
2020-01-11 13:03:09.402 Info, Node067,     Version       = 4
2020-01-11 13:03:09.402 Info, Node067,     Security      = false
2020-01-11 13:03:09.402 Info, Node067,   Basic device class    (0x04) - Routing Slave
2020-01-11 13:03:09.402 Info, Node067,   Generic device Class  (0x40) - Entry Control
2020-01-11 13:03:09.402 Info, Node067,   No specific device class defined
2020-01-11 13:03:09.402 Info, Node067,     COMMAND_CLASS_BASIC is not mapped
[... 860 lines cut ...]
2020-01-11 13:03:43.604 Detail, Node067, Notification: Notification - TimeOut
2020-01-11 13:03:43.604 Info, Node067, Sending (Send) message (Callback ID=0xed, Expected Reply=0x04) - SwitchBinaryCmd_Get (Node=67): 0x01, 0x09, 0x00, 0x13, 0x43, 0x02, 0x25, 0x02, 0x25, 0xed, 0x4b
2020-01-11 13:03:43.604 Info, Node067, Encrypted Flag is 0
2020-01-11 13:03:43.609 Detail, Node067,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-01-11 13:03:43.609 Detail, Node067,   ZW_SEND_DATA delivered to Z-Wave stack
2020-01-11 13:03:46.604 Error, Node067, ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "SwitchBinaryCmd_Get (Node=67): 0x01, 0x09, 0x00, 0x13, 0x43, 0x02, 0x25, 0x02, 0x25, 0xed, 0x4b"
2020-01-11 13:03:46.604 Detail, Node067, Removing current message
[... cut ...]
2020-01-11 13:03:49.127 Info, Node067, Node not responding - moving message to Wake-Up queue: UserCodeCmd_Get (Node=67): 0x01, 0x0a, 0x00, 0x13, 0x43, 0x03, 0x63, 0x02, 0x02, 0x25, 0xeb, 0x0b
2020-01-11 13:03:49.127 Info, Node067, Node not responding - moving QueryStageComplete command to Wake-Up queue
2020-01-11 13:03:49.147 Info, Node067, Cache Save for Node 67 as its QueryStage_CacheLoad
2020-01-11 13:03:49.209 Detail, Node067, Notification: Notification - Node Asleep
2020-01-11 13:03:57.894 Info, Node067, Cache Save for Node 67 as its QueryStage_CacheLoad
```

[image]

Fishwaldo, 2020-01-11:
There is a change with 1.6 to make the usercode command class compliant with the specifications. 

Unfortunately RFID readers abuse the specifications so you need to set https://github.com/OpenZWave/open-zwave/wiki/CommandClass-Compatibility-Flags#usercode-commandclass this in your config file and delete ozwcache (or you can try refresh node info but I don’t know how domoticz will react. 

Alarm Type and Alarm Level has been depreciated from the specifications. You should have dedicated Alarm events now.

candrea77, 2020-01-22:
Sorry for late reply.
Currently my ozwcache for the rfis reader node have :
```
            <CommandClass id="99" name="COMMAND_CLASS_USER_CODE">
                <Compatibility>
                    <ExposeRawUserCodes>true</ExposeRawUserCodes>
                </Compatibility>
```

I need to set it to false and retry or is correct have it to true ?

Fishwaldo, 2020-01-29:
Nope. Thats correct. ozwcp doesn't display raw values correctly, hence you get that garbled text (its actually a raw byte array, it shouldn't be displayed as text - thats where the RFID readers are "incompatible" with the specifications. The UserCode's should be treated as ASCII).

So all is good - OZW is functioning as intended. As mentioned, Alarm Type and Alarm Level are obsolete and replaced by specific Alarm ValueID's now.

closed 2020-01-29 by Fishwaldo