OpenZWave/open-zwave #2071: SensorBasic set value from Basic CC when GetSupported is false

michalskrivanek, 2020-01-17:
This is a weird use case, but I believe a code change makes sense regardless. The point is to set the sensor's value via Basic CC Report without syncing the sensor value when Get is not supported/disabled.

My use case is following, I have a wenzhou/t56.xml wall switch which has 2 paddles, and the right one is designed to only control other zwave devices, it's not switching anything. In my installation that makes it effectively useless as I don't really have a zwave device I can control. What I can do, though, is to use it as a trigger for other stuff, like Hue. All I need is to be able to get the right paddle state. But there's no sensor or anything, all the paddle does is Basic Report

Solution I found was to define a fake binary sensor on Basic CC, and configure the right paddle association with controller, that way the received value attempts to SensorBinary::SetValueBasic the state of the paddle. But currently this fails because we always call a value refresh to get the real state of the (fake) sensor, that fails and value is not set. 
A simple fix to consider GetSupported can make this work, not trying to sync the value when it can't.

Again, a weird usecase, but hopefully useful for some...

Fishwaldo, 2020-01-18:
Sounds like the device isn’t quite compliant with the specs. 

You could play with Ignore Mapping to see if that does what you want. https://github.com/OpenZWave/open-zwave/wiki/CommandClass-Compatibility-Flags#basic-commandclass

michalskrivanek, 2020-03-03:
I think the problem is rather witch how I try to use it:) 
IIUC with IgnoreMapping I would be able to make it poll a different class, but there seems to be nothing on the device which actually holds the state of the right paddle. All it does is to send a report (0 or 255 value) without any state. 
With my workaround I just "trust" the sent value and skip value refresh (as there is nothing to poll). 

Either way, a patch is fairly simple and I believe it makes sense regardless my weird use case. IT just honors the GetSupported setting.

Fishwaldo, 2020-03-04:
Ok. Merged. Makes sense

PR #2138 (2020-03-04): do not sychronize a BASIC set/report when Get is not supported
fixes #2071
closed 2020-03-04 by Fishwaldo