OpenZWave/open-zwave #2079: Qubino devices do not get properly included in Home Assistant

darkbasic, 2020-01-23:
I'm trying to get the Qubino Shutter DC working properly in HA in venetian mode. Unfortunately the firmware happened to be bugged so I've sent it back to Qubino in order to get a fix, which landed in a new firmware on January 2020. With my surprise I still had issues with HA, to the point that I contacted Qubino in order to discover what was going on. They usually don't test against HA, so when they finally did they discovered that the inclusion was troublesome: of the 5 times that they included it, the gateway included properly only once. This is the same behaviour I always had, but initially I though it was because of the distance. So I set up a docker container in my laptop with my AEON Labs ZW090 Z-Stick Gen5 EU and Home Assistant 0.104.2 near to the zwave device: same behaviour, most of the times it misses something during the inclusion.
They tested the same device with Fibaro, Jeedom and Eedomus and it seemed to work well.
I'm still not sure what's the problem with the venetian mode and Qubino couldn't figure it out either, but all those inclusion issues seem strange to me.
Is there any way we can figure out what's going wrong and why it only happens with HA? What's the best openzwave interface I can use in order to test if the problem only pertains to HA?

Thanks

Dinth, 2020-01-24:
Home Assistant is using own fork of old branch of OZW. It's better if you log the issue on their bugtracker.
You can also try to run Zwave2MQTT docker container which uses the latest version of OZW to see if you can replicate the issue on latest OZW.

Fishwaldo, 2020-01-29:
HA is using OZW 1.4 which we don't support anymore. You either have to wait till the new integration arrives or consider Z2M.

closed 2020-01-29 by Fishwaldo