OpenZWave/open-zwave #2082: Devices with broken CommandClsas

Dinth, 2020-01-25:
OZW supports many devices which have broken/unfuntional CommandClasses. For example Fibaro binary switches often support SWITCH_MULTILEVEL command class which is not functional.

On HA/OZW1.4 all those CommandClasses were hidden - entities were not generated using them. 
On Z2M/OZW1.6 though, those CommandClasses generate values which while not functional, are visible in Z2M web interface and further passed to HA through autodiscovery.

The idea is that those devices should be hidden in a same way as in HA/OZW1.4, but dont know how this can be achieved. 
It seems that all such broken CommandClasses have _<GetSupported>false</GetSupported>_ tag in their config files and could be distinguished using it - but as far as i understand, GetSupported tag does not imply the CommandClass should be hidden, only that it shouldnt be updated.

I had a look in HA source code trying to find out how HA does it, but havent found anything besides CONF_IGNORED boolean in device config schema. I couldnt find how it is populated though. 

Could someone help us to figure out how this can be done?

This is just a question related to this: https://github.com/OpenZWave/Zwave2Mqtt/issues/219

Fishwaldo, 2020-01-29:
As mention - Nothing was ever "hidden" in these devices. What you are seeing is probably something HA did. It wasn't OZW or the pyozw wrappers. 

You can add "remove" to the CommandClass block to entirely remove that commandclass, but before you do that, I'd need confirmation you have the very latest firmware version (thus you don't have a buggy device) and confirmation from Fibaro that its not valid (@petergebruers may be able to help there if he has the bandwidth).

I wont "fix" a config file for a older version with buggy firmware if the vendor has fixed the issue in more recent firmware releases.

Dinth, 2020-01-29:
Hi @Fishwaldo, thanks for your comments. 

I just noticed that github has deleted one important bit of my original post:
> It seems that all such broken CommandClasses have _<GetSupported>false</GetSupported>_ tag in their config files and could be distinguished using it

should be 
> It seems that all such broken CommandClasses have 
`<GetSupported>false</GetSupported>`
>tag in their config files and could be distinguished using it.

All my Fibaro switches were recently updated to the latest FW, but i will contact Fibaro support to double check.

Dinth, 2020-02-05:
I have received the following response for Fibaro (google-translated)

M G-K 29 Jan 20
 
I warmly welcome.
I have several binary Fibaro switches (FGS-213 and FGS-223).
They are updated to the latest version using the Fibaro HC control unit
which I only use to update the firmware.
These devices report that, in addition to the usual switches they use
CommandClass SWITCH_MULTILEVEL which should normally be used for
dimmers.
Could you confirm if this SWITCH_MULTILEVEL class is for
something used in these devices?

Hello,
Sorry for the long response time, Switch 2 can activate scenes in the Z-Wave controller by sending the scene identifier and the action data attribute using the central Scene command class. By default, scenes are not activated, set parameters 28 and 29 to enable scene activation for selected actions.
The Switch Multilevel command class is also used to create and operate associations.
Association (pairing) - direct control of another device in the Z-Wave network, e.g. Dimmer, Switch (ON-OFF), Roller Shutter Controller or stage (only via the Z-Wave controller).

petergebruers, 2020-02-05:
Makes sense. They are talking about "controlling" vs "controlled", something OZW calls "after mark".

You don't happen to have an OZW_log.txt from an inclusion of one of those devices? I don't have one available for testing...

Fishwaldo, 2020-02-10:
According to your support response, nothing "unusual" about these devices or the CC's they expose. OZW is doing the "right thing" here as a library.

closed 2020-02-10 by Fishwaldo