OpenZWave/open-zwave #2083: Shenzhen PD03Z confused by PD02Z.. and PD03Z missing

billiaz, 2020-01-25:
Hi All,

The Device I have on my hands, is missing parameters. Per manufacturer manual, the PD03Z is a newer version of the old device with added Temperature control.
It also has parameters moved around.

In the past I helped another database to add this device (Example of this device: https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/401)

I would like to add the same details and correct the device recognition. Should I submit a pull request in this case?

Can I also alternatively try modify/test this on my local install? (Using now home assistant)

billiaz, 2020-01-25:
Some history on this: https://community.openhab.org/t/neo-coolcam-motion-sensor-pd02z-and-previous-models/60372/9

one of the devices changes these parameters on a firmware version

petergebruers, 2020-01-25:
It think it is not as easy as as you say.

I have investigated as similar case, or possibly the same case a while ago (July 2019)

Can you please have a look at this PR

https://github.com/OpenZWave/open-zwave/pull/1867

And please also read this:

https://github.com/OpenZWave/open-zwave/issues/1866

The main conclusion at that time was:

```
I suddenly realize all posts up to now do not properly explain the main issue and proposed fix.

So with same IDs there is a temperature model and a non-temperature model.

The device flashes a bright LED when it detects motion, and this can be disabled but they used a different parameter for this.

With the current nas-pd02z.xml it is not possible to turn off the LED if you have the newer "temperature" model.

I propose a new config file with these parameters

index="10" label="Ambient Temperature Differential Level Report or Led Blink Enable (0 or 1 - non-temperature version)"
index="11" label="Led Blink Enable (temperature reporting version)"
Parameter 10 is the only "conflicting" parameter but imho that explanation should help. Parameters 11,12 and 99 are new
```

aretakisv, 2020-01-25:
I have had the same investigation with another z-wave module.
That module had the way to filter on device firmware version.

We tested multiple devices and found that specific device code 0003:108d have this pattern:
Anything from firmware 3.80 has Temperature and the three extra params, anything else not!

The device with 3.80 firmware and higher is also called PD03Z and not PD02Z.
I suspect (but is only thoughts) manufacturer forgot to change the 108d to 208d in early devices.

should I update my pull request with your suggestions, in order to cover both cases?

petergebruers, 2020-01-25:
> We tested multiple devices and found that specific device code 0003:108d have this pattern:
> Anything from firmware 3.80 has Temperature and the three extra params, anything else not!

From the top of my memory, that is exactly the problem and OpenZWave does not distinguish devices based on firmware. It is a very small % that requires this kind of revision based discrimination. So you have to come up with cofig files that are compatible with all firmware versions, or you'll break compatibility with the older devices.

> should I update my pull request with your suggestions, in order to cover both cases?

There might be 3 cases instead of two if I understand you correctly

The July PR I referred to covers 2 cases, two types of devices, with same IDs... < 3.80 is called PD02Z and > = 3.80 is sometimes called "PD02Z with temperature". I see no way how you can improve this. I am not sure if it is ever called PD03Z. I have both models and "PD02Z with temperature" was deleverd with the old manual, not covering the firmware changes...

If same ID is also use for PD03Z then you can only add that to the name of the device entry that exists now

If you have device with a _new and different ID_ then this is a third case (of the Neo Motion PIR) and I'll accept a PR with a new config file, because there is no conflict there.

petergebruers, 2020-01-25:
@billiaz Are you talking about Home Assistant?

petergebruers, 2020-01-25:
I mean by this the Z-Wave integration offered by HASS? Not Zwave2Mqtt for example?

Hass native integration uses an old version 1.4 of OZW and forked the repository.

OpenZWave 1.6 correctly supports your device.

aretakisv, 2020-01-25:
not the zwave2MQTT, though because I am on raspberry PI, I am now slowly migrating it to the MQTT one (fresh install)

petergebruers, 2020-01-25:
In that case, imho, this issue can be closed, because OpenZWave 1.6 supports your device.

Are you able to test that?

aretakisv, 2020-01-25:
I've seen the XML and I am pretty sure it works!

petergebruers, 2020-01-25:
Thank you for reporting back.

closed 2020-01-25 by petergebruers