OpenZWave/open-zwave #2085: [bug] Cannot update values on SoundSwitch - Doorbell 6

billiaz, 2020-01-26:
Running 2.1.1 of zwave2mqtt and open-zwave 1.6.

Having an 3 days old Aeotec Doorbell 6. And with the latest firmware. Configuration of Volume/Type of sound seems to not work properly.

Example:
WebUI apply settings for Tamper Alert type to **Inactive**  (67-121-2-3)

Log output
```
2020-01-26T08:51:36.987Z z2m:App Zwave api call: setValue [ 67, 121, 2, 3, 'Inactive' ]
2020-01-26 08:51:36.989 Info, Node067, Value::Set - COMMAND_CLASS_SOUND_SWITCH - Default Tone - 3 - 2 - Inactive
2020-01-26 08:51:36.990 Detail, Node067, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_MULTI_INSTANCE/CHANNEL
2020-01-26 08:51:36.990 Detail, Node067, Queuing (Send) MultiChannel Encapsulated (instance=2): SoundSwitchCmd_Tones_Config_Set (Node=67): 0x01, 0x0f, 0x00, 0x13, 0x43, 0x08, 0x60, 0x0d, 0x01, 0x01, 0x79, 0x05, 0x64, 0x06, 0x25, 0x29, 0xd7
2020-01-26 08:51:36.991 Detail,
2020-01-26 08:51:36.991 Info, Node067, Processing (Send) Nonce Request message (Callback ID=0x29, Expected Reply=0x04)
2020-01-26 08:51:36.991 Info, Node067, Sending (Send) message (Callback ID=0x02, Expected Reply=0x04) - Nonce_Get(MultiChannel Encapsulated (instance=2): SoundSwitchCmd_Tones_Config_Set) - 0x01, 0x09, 0x00, 0x13, 0x43, 0x02, 0x98, 0x40, 0x05, 0x02:
2020-01-26T08:51:36.991Z z2m:Zwave Success zwave api call setValue
2020-01-26 08:51:36.997 Detail, Node067,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-01-26 08:51:36.997 Detail, Node067,   ZW_SEND_DATA delivered to Z-Wave stack
2020-01-26 08:51:37.015 Detail, Node067,   Received: 0x01, 0x18, 0x00, 0x13, 0x02, 0x00, 0x00, 0x02, 0x00, 0xd8, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x2d
2020-01-26 08:51:37.015 Detail, Node067,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)
2020-01-26 08:51:37.015 Info, Node067, Request RTT 24 Average Request RTT 166
2020-01-26 08:51:37.015 Detail, Node067, Extended TxStatus: Time: 20, Hops: 0, Rssi:  -40 --- --- --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 0 0 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
2020-01-26 08:51:37.033 Detail, Node067,   Received: 0x01, 0x12, 0x00, 0x04, 0x00, 0x43, 0x0a, 0x98, 0x80, 0x96, 0x4d, 0xa1, 0xc7, 0x34, 0x25, 0xc3, 0x87, 0xd8, 0x00, 0x88
2020-01-26 08:51:37.033 Info, Node067, Received SecurityCmd_NonceReport from node 67
2020-01-26 08:51:37.033 Info, Node067, Encrypted Flag is 1
2020-01-26 08:51:37.033 Info, Node067, Sending (Send) message (Callback ID=0x2a, Expected Reply=0x04) - MultiChannel Encapsulated (instance=2): SoundSwitchCmd_Tones_Config_Set (Node=67): 0x01, 0x0f, 0x00, 0x13, 0x43, 0x08, 0x60, 0x0d, 0x01, 0x01, 0x79, 0x05, 0x64, 0x06, 0x25, 0x2a, 0xd4
2020-01-26 08:51:37.042 Detail, Node067,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-01-26 08:51:37.043 Detail, Node067,   ZW_SEND_DATA delivered to Z-Wave stack
2020-01-26 08:51:37.063 Detail, Node067,   Received: 0x01, 0x18, 0x00, 0x13, 0x2a, 0x00, 0x00, 0x02, 0x00, 0xd8, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x05
2020-01-26 08:51:37.063 Detail, Node067,   ZW_SEND_DATA Request with callback ID 0x2a received (expected 0x2a)
2020-01-26 08:51:37.064 Info, Node067, Request RTT 73 Average Request RTT 119
2020-01-26 08:51:37.064 Detail, Node067, Extended TxStatus: Time: 20, Hops: 0, Rssi:  -40 --- --- --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 0 0 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
2020-01-26 08:51:37.065 Detail, Node067,   Expected callbackId was received
2020-01-26 08:51:46.992 Error, Node067, ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "MultiChannel Encapsulated (instance=2): SoundSwitchCmd_Tones_Config_Set (Node=67): 0x01, 0x0f, 0x00, 0x13, 0x43, 0x08, 0x60, 0x0d, 0x01, 0x01, 0x79, 0x05, 0x64, 0x06, 0x25, 0x2a, 0xd4"
2020-01-26 08:51:46.992 Detail, Node067, Removing current message
2020-01-26 08:51:46.992 Detail, Node067, Notification: Notification - TimeOut
2020-01-26T08:51:46.994Z z2m:Zwave Notification from node 67: Notification - TimeOut (1)
```

example of volume update doorbell 1:

```
2020-01-26T08:53:32.704Z z2m:App Zwave api call: setValue [ 67, 121, 3, 2, '6' ]
2020-01-26 08:53:32.707 Info, Node067, Value::Set - COMMAND_CLASS_SOUND_SWITCH - Volume - 2 - 3 - 6
2020-01-26 08:53:32.707 Detail, Node067, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_MULTI_INSTANCE/CHANNEL
2020-01-26 08:53:32.707 Detail, Node067, Queuing (Send) MultiChannel Encapsulated (instance=3): SoundSwitchCmd_Tones_Config_Set (Node=67): 0x01, 0x0f, 0x00, 0x13, 0x43, 0x08, 0x60, 0x0d, 0x01, 0x02, 0x79, 0x05, 0x64, 0x11, 0x25, 0x2b, 0xc1
2020-01-26T08:53:32.708Z z2m:Zwave Success zwave api call setValue
2020-01-26 08:53:32.709 Detail,
2020-01-26 08:53:32.711 Info, Node067, Processing (Send) Nonce Request message (Callback ID=0x2b, Expected Reply=0x04)
2020-01-26 08:53:32.712 Info, Node067, Sending (Send) message (Callback ID=0x02, Expected Reply=0x04) - Nonce_Get(MultiChannel Encapsulated (instance=3): SoundSwitchCmd_Tones_Config_Set) - 0x01, 0x09, 0x00, 0x13, 0x43, 0x02, 0x98, 0x40, 0x05, 0x02:
2020-01-26 08:53:32.719 Detail, Node067,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-01-26 08:53:32.719 Detail, Node067,   ZW_SEND_DATA delivered to Z-Wave stack
2020-01-26 08:53:32.737 Detail, Node067,   Received: 0x01, 0x18, 0x00, 0x13, 0x02, 0x00, 0x00, 0x02, 0x00, 0xda, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x2f
2020-01-26 08:53:32.738 Detail, Node067,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)
2020-01-26 08:53:32.738 Info, Node067, Request RTT 24 Average Request RTT 71
2020-01-26 08:53:32.738 Detail, Node067, Extended TxStatus: Time: 20, Hops: 0, Rssi:  -38 --- --- --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 0 0 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
2020-01-26 08:53:32.755 Detail, Node067,   Received: 0x01, 0x12, 0x00, 0x04, 0x00, 0x43, 0x0a, 0x98, 0x80, 0x02, 0x29, 0xe1, 0x1a, 0xd9, 0x39, 0x1e, 0x35, 0xda, 0x00, 0x79
2020-01-26 08:53:32.756 Info, Node067, Received SecurityCmd_NonceReport from node 67
2020-01-26 08:53:32.756 Info, Node067, Encrypted Flag is 1
2020-01-26 08:53:32.756 Info, Node067, Sending (Send) message (Callback ID=0x2c, Expected Reply=0x04) - MultiChannel Encapsulated (instance=3): SoundSwitchCmd_Tones_Config_Set (Node=67): 0x01, 0x0f, 0x00, 0x13, 0x43, 0x08, 0x60, 0x0d, 0x01, 0x02, 0x79, 0x05, 0x64, 0x11, 0x25, 0x2c, 0xc6
2020-01-26 08:53:32.766 Detail, Node067,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-01-26 08:53:32.766 Detail, Node067,   ZW_SEND_DATA delivered to Z-Wave stack
2020-01-26 08:53:32.786 Detail, Node067,   Received: 0x01, 0x18, 0x00, 0x13, 0x2c, 0x00, 0x00, 0x02, 0x00, 0xda, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x01
2020-01-26 08:53:32.787 Detail, Node067,   ZW_SEND_DATA Request with callback ID 0x2c received (expected 0x2c)
2020-01-26 08:53:32.787 Info, Node067, Request RTT 73 Average Request RTT 72
2020-01-26 08:53:32.787 Detail, Node067, Extended TxStatus: Time: 20, Hops: 0, Rssi:  -38 --- --- --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 0 0 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
2020-01-26 08:53:32.789 Detail, Node067,   Expected callbackId was received
2020-01-26 08:53:42.714 Error, Node067, ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "MultiChannel Encapsulated (instance=3): SoundSwitchCmd_Tones_Config_Set (Node=67): 0x01, 0x0f, 0x00, 0x13, 0x43, 0x08, 0x60, 0x0d, 0x01, 0x02, 0x79, 0x05, 0x64, 0x11, 0x25, 0x2c, 0xc6"
2020-01-26 08:53:42.715 Detail, Node067, Removing current message
2020-01-26 08:53:42.715 Detail, Node067, Notification: Notification - TimeOut
2020-01-26T08:53:42.717Z z2m:Zwave Notification from node 67: Notification - TimeOut (1)
```
```

Also should soundSwitch be a parameter accessible by the automation and not a configuration param, meaning switch/adjustment instead of configuration?
It is a soundswitch ( in light similar is an RGB light or dimmer right?)

billiaz, 2020-01-26:
Tested it also on the latest Domoticz (4.110 with OZW 1.6)
same issue

billiaz, 2020-01-27:
Testing with qt-openzwave running 1.6.1019 also similar results

from my setup the node value id is:
```
{
    "Label": "Doorbell 1 Volume",
    "Value": 100,
    "Units": "",
    "Min": 0,
    "Max": 255,
    "Type": "Byte",
    "Instance": 3,
    "CommandClass": "COMMAND_CLASS_SOUND_SWITCH",
    "Index": 2,
    "Node": 74,
    "Genre": "Config",
    "Help": "The Volume to play the tone at",
    "ValueIDKey": 562951205306417,
    "ReadOnly": false,
    "WriteOnly": false,
    "ValueSet": false,
    "ValuePolled": false,
    "ChangeVerified": false,
    "Event": "valueAdded",
    "TimeStamp": 1580115716
}
```
used the command:
```
OpenZWave/1/command/setvalue/
{ "ValueIDKey": 562951205306417, "Value": 10 }
```

Error similar:
```
[20200127 9:09:24.200 UTC] [qt.mqtt.connection.verbose] [debug]: Finalize PUBLISH: topic: QMqttTopicName("OpenZWave/1/node/73/statistics/")  payloadLength: 809
[20200127 9:09:24.200 UTC] [qt.mqtt.connection.verbose] [debug]: Received PUBLISH
[20200127 9:09:24.200 UTC] [qt.mqtt.connection.verbose] [debug]: Finalize PUBLISH: topic: QMqttTopicName("OpenZWave/1/node/74/statistics/")  payloadLength: 817
[20200127 9:09:29.725 UTC] [ozw.library] [critical]: Error - Node: 74 ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "MultiChannel Encapsulated (instance=3): SoundSwitchCmd_Tones_Config_Set (Node=74): 0x01, 0x0f, 0x00, 0x13, 0x4a, 0x08, 0x60, 0x0d, 0x01, 0x02, 0x79, 0x05, 0x64, 0x11, 0x25, 0xb0, 0x53"
[20200127 9:09:29.726 UTC] [ozw.library] [debug]: Detail - Node: 74 Removing current message
[20200127 9:09:29.727 UTC] [ozw.library] [debug]: Detail - Node: 74 Notification: Notification - TimeOut
[20200127 9:09:29.728 UTC] [ozw.notifications] [debug]: Notification pvt_ozwNotification NotificationTypes::Notification_Code_MsgTimeout
[20200127 9:09:29.729 UTC] [ozw.mqtt.publisher] [debug]: Publishing Event ozwNotification
[20200127 9:09:29.730 UTC] [qt.mqtt.connection] [debug]: qint32 QMqttConnection::sendControlPublish(const QMqttTopicName&, const QByteArray&, quint8, bool, const QMqttPublishProperties&) QMqttTopicName("OpenZWave/1/event/notification/")  Size: 92  bytes. QoS: 0  Retain: false
[20200127 9:09:29.730 UTC] [qt.mqtt.connection.verbose] [debug]: bool QMqttConnection::writePacketToTransport(const QMqttControlPacket&)  DataSize: 127
[20200127 9:09:29.732 UTC] [qt.mqtt.connection.verbose] [debug]: void QMqttConnection::transportReadReady()
[20200127 9:09:29.733 UTC] [qt.mqtt.connection.verbose] [debug]: Received PUBLISH
[20200127 9:09:29.734 UTC] [qt.mqtt.connection.verbose] [debug]: Finalize PUBLISH: topic: QMqttTopicName("OpenZWave/1/event/notification/")  payloadLength: 92
[20200127 9:09:53.185 UTC] [qt.mqtt.connection] [debug]: bool QMqttConnection::sendControlPingRequest()
[20200127 9:09:53.186 UTC] [qt.mqtt.connection.verbose] [debug]: bool QMqttConnection::writePacketToTransport(const QMqttControlPacket&)  DataSize: 2
[20200127 9:09:53.187 UTC] [qt.mqtt.connection.verbose] [debug]: void QMqttConnection::transportReadReady()
[20200127 9:09:53.188 UTC] [qt.mqtt.connection.verbose] [debug]: Received PINGRESP
[20200127 9:09:53.189 UTC] [qt.mqtt.connection.verbose] [debug]: Finalize PINGRESP
[20200127 9:09:53.189 UTC] [ozw.mqtt.publisher] [debug]: "Mon Jan 27 09:09:53 2020 PingResponse\n"
```

Also when joining encrypted the Ringtone values are broken
example:

```
{
    "Label": "Doorbell 1 Tones",
    "Value": {
        "List": [
            {
                "Value": 0,
                "Label": "Inactive"
            },
            {
                "Value": 1,
[... 121 lines cut ...]
            }
        ],
        "Selected": "Inactive"
    },
    "Units": "",
    "Min": 0,
    "Max": 0,
    "Type": "List",
    "Instance": 3,
    "CommandClass": "COMMAND_CLASS_SOUND_SWITCH",
    "Index": 1,
    "Node": 74,
    "Genre": "User",
    "Help": "A List of Available Tones on the Device",
    "ValueIDKey": 281476224401460,
    "ReadOnly": false,
    "WriteOnly": false,
    "ValueSet": false,
    "ValuePolled": false,
    "ChangeVerified": false,
    "Event": "valueAdded",
    "TimeStamp": 1580115813
}
```
when insecure they show without special characters.

Fishwaldo, 2020-01-29:
hrm... I'll have to pull mine out. It was working (but I'll check firmware versions). Maybe you can open a support case with Aeotec and ask them if it works. 

As far as the "corruption" with secure/non-secure - That doesn't make much sense. In the OZW Log file are the labels "corrupt" as well? what about ozwcache*.xml file etc.

billiaz, 2020-01-29:
The sound types names, seem to come from the doorbell itself, xml doesn't have any reference to these unzer zw162.xml

Unencrypted the sound names are ok,
S0 encryption not ok.

Tested with 3 qt-openzwave,zwave2mqtt and domoticz beta

Fishwaldo, 2020-01-29:
The names should be saved in ozwcache file after inclusion. Please check that and compare between encrypted/non-encrypted.

billiaz, 2020-01-29:
> The names should be saved in ozwcache file after inclusion. Please check that and compare between encrypted/non-encrypted.

Will do later today, and Update this thread. should I send you also inclusion logs?

billiaz, 2020-01-29:
Including a Secure inclusion, later when home I will re include with insecure inclusion
[ozwcache_secure.txt](https://github.com/OpenZWave/open-zwave/files/4126732/ozwcache_secure.txt)

billiaz, 2020-01-29:
attaching Inclusion log from qt-openzwave 2.1.1 insecure
[inclusion doorbell6 - insecure.txt](https://github.com/OpenZWave/open-zwave/files/4128160/inclusion.doorbell6.-.insecure.txt)

Ozwcache of node insecure:
[doorbell6 - insecure ozwcache.txt](https://github.com/OpenZWave/open-zwave/files/4128163/doorbell6.-.insecure.ozwcache.txt)

@Fishwaldo Are these  useful?
Thanks,
Vassilis

billiaz, 2020-01-29:
> hrm... I'll have to pull mine out. It was working (but I'll check firmware versions). Maybe you can open a support case with Aeotec and ask them if it works.

From my device data I do see Version is 1.04?

billiaz, 2020-01-29:
Tested the changes with z-way server (z-wave.me) and the changes work without issues. I suspect is not the device

from Z-way server logs:
```

[2020-01-29 16:06:24.346] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.6 = Empty
[2020-01-29 16:06:24.346] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.6.status = Empty
[2020-01-29 16:06:24.346] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.6.duration = Empty
[2020-01-29 16:06:24.347] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.6.moreStatusUpdates = Empty
[2020-01-29 16:06:24.347] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.6.lastEvent = Empty
[2020-01-29 16:06:24.347] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.6.status = 255 (0x000000ff)
[2020-01-29 16:06:24.348] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.6.duration = 0 (0x00000000)
[2020-01-29 16:06:24.348] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.6.moreStatusUpdates = False
[2020-01-29 16:06:24.348] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.6.lastEvent = 823177 (0x000c8f89)
[2020-01-29 16:06:24.349] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.6 = Empty
[2020-01-29 16:06:24.349] [D] [zway] SETDATA devices.76.instances.0.commandClasses.108.data.lastSession = 6 (0x00000006)
[2020-01-29 16:06:24.350] [I] [zway] Adding job: SoundSwitch Configuration Set
[2020-01-29 16:06:24.350] [D] [zway] SENDING (cb 0xd8): ( 01 13 00 13 4C 0C 60 0D 00 03 6C 01 86 04 79 05 00 00 25 D8 BF )
[2020-01-29 16:06:24.353] [D] [zway] RECEIVED ACK
[2020-01-29 16:06:24.357] [D] [zway] RECEIVED: ( 01 04 01 13 01 E8 )
[2020-01-29 16:06:24.358] [D] [zway] SENT ACK
[2020-01-29 16:06:24.358] [D] [zway] Delivered to Z-Wave stack
[2020-01-29 16:06:24.381] [D] [zway] RECEIVED: ( 01 18 00 13 D8 00 00 02 00 DD 7F 7F 7F 7F 00 00 03 00 00 00 00 03 01 00 00 F2 )
[2020-01-29 16:06:24.382] [D] [zway] SENT ACK
[2020-01-29 16:06:24.383] [I] [zway] Job 0x13 (SoundSwitch Configuration Set): Delivered
[2020-01-29 16:06:24.383] [D] [zway] SendData Response with callback 0xd8 received: received by recipient
[2020-01-29 16:06:24.384] [D] [zway] SETDATA devices.76.data.lastSendInternal = **********
[2020-01-29 16:06:24.384] [D] [zway] SETDATA devices.76.data.lastSend = 823181 (0x000c8f8d)
[2020-01-29 16:06:24.385] [D] [zway] Job 0x13 (SoundSwitch Configuration Set): success
[2020-01-29 16:06:24.385] [I] [zway] Waiting for job reply: SoundSwitch Configuration Set
[2020-01-29 16:06:24.412] [D] [zway] RECEIVED: ( 01 11 00 04 00 4C 09 60 0D 03 00 6C 02 06 FF 00 DD 00 8B )
[2020-01-29 16:06:24.413] [D] [zway] SENT ACK
[2020-01-29 16:06:24.413] [D] [zway] SETDATA devices.76.data.lastReceived = 0 (0x00000000)
[2020-01-29 16:06:24.414] [D] [zway] SETDATA devices.76.instances.3.commandClasses.108.data.6 = Empty
[2020-01-29 16:06:24.414] [D] [zway] SETDATA devices.76.instances.3.commandClasses.108.data.6.status = Empty
[2020-01-29 16:06:24.414] [D] [zway] SETDATA devices.76.instances.3.commandClasses.108.data.6.duration = Empty
[2020-01-29 16:06:24.415] [D] [zway] SETDATA devices.76.instances.3.commandClasses.108.data.6.moreStatusUpdates = Empty
[2020-01-29 16:06:24.415] [D] [zway] SETDATA devices.76.instances.3.commandClasses.108.data.6.lastEvent = Empty
[2020-01-29 16:06:24.416] [D] [zway] SETDATA devices.76.instances.3.commandClasses.108.data.6.status = 255 (0x000000ff)
[2020-01-29 16:06:24.416] [D] [zway] SETDATA devices.76.instances.3.commandClasses.108.data.6.duration = 0 (0x00000000)
[2020-01-29 16:06:24.416] [D] [zway] SETDATA devices.76.instances.3.commandClasses.108.data.6.moreStatusUpdates = False
[2020-01-29 16:06:24.417] [D] [zway] SETDATA devices.76.instances.3.commandClasses.108.data.6.lastEvent = 823184 (0x000c8f90)
[2020-01-29 16:06:24.417] [D] [zway] SETDATA devices.76.instances.3.commandClasses.108.data.6 = Empty
[2020-01-29 16:06:24.418] [D] [zway] Received reply on job (SoundSwitch Configuration Set)
[2020-01-29 16:06:24.418] [I] [zway] Node 76:3 CC Supervision: Setter packet was confirmed: SoundSwitch Configuration Set
[2020-01-29 16:06:24.418] [D] [zway] SETDATA devices.76.instances.3.commandClasses.121.data.defaultVolume = 0 (0x00000000)
[2020-01-29 16:06:24.419] [D] [zway] SETDATA devices.76.instances.3.commandClasses.121.data.defaultTone = 2 (0x00000002)
[2020-01-29 16:06:24.436] [I] [core] Notification: device-info (device-status): {"dev":"Volume (76.3)","l":"0%","location":""}
[2020-01-29 16:06:24.524] [D] [zway] RECEIVED: ( 01 10 00 04 00 4C 08 60 0D 03 00 79 07 00 02 DD 00 60 )
[2020-01-29 16:06:24.525] [D] [zway] SENT ACK
[2020-01-29 16:06:24.526] [D] [zway] SETDATA devices.76.data.lastReceived = 0 (0x00000000)
[2020-01-29 16:06:24.526] [D] [zway] SETDATA devices.76.instances.3.commandClasses.121.data.defaultVolume = 0 (0x00000000)
[2020-01-29 16:06:24.527] [D] [zway] SETDATA devices.76.instances.3.commandClasses.121.data.defaultTone = 2 (0x00000002)
[2020-01-29 16:06:26.528] [D] [zway] Job 0x13: deleted from queue

[2020-01-29 16:06:35.656] [I] [zway] Adding job: SoundSwitch Configuration Get
[2020-01-29 16:06:35.662] [D] [zway] SENDING (cb 0xd9): ( 01 0D 00 13 4C 06 60 0D 00 03 79 06 25 D9 46 )
[2020-01-29 16:06:35.664] [D] [zway] RECEIVED ACK
[2020-01-29 16:06:35.668] [D] [zway] RECEIVED: ( 01 04 01 13 01 E8 )
[2020-01-29 16:06:35.668] [D] [zway] SENT ACK
[... cut ...]
[2020-01-29 16:07:16.979] [D] [zway] SETDATA controller.data.statistics.RFRxCRC16Errors = 0 (0x00000000)
[2020-01-29 16:07:16.979] [D] [zway] SETDATA controller.data.statistics.RFRxForeignHomeID = 0 (0x00000000)
[2020-01-29 16:07:16.979] [I] [zway] Job 0x3a (Get statistics gathered by the Z-Wave protocol): Done
[2020-01-29 16:07:16.979] [D] [zway] Job 0x3a (Get statistics gathered by the Z-Wave protocol): success
[2020-01-29 16:07:16.979] [I] [zway] Removing job: Get statistics gathered by the Z-Wave protocol
[2020-01-29 16:07:18.362] [I] [zway] Adding job: Get background noise level
```

Fishwaldo, 2020-01-29:
Ok. It does help. Truly bizarre! (The corruption that is). Looks like a off by 1 error in the decoding of packets or I’m not zeroing out a buffer somewhere. 

I’m traveling right now and won’t be able to get to this till next week at the earliest. Leave it with me to investigate

billiaz, 2020-01-29:
Cool if you want me to switch over my installations I am happy to provide more logs/support/tests.

Thank you for the awesome library and work!

Fishwaldo, 2020-02-07:
In the latest commits I tidied up the SoundSwitch. I’ve also realized I had firmware 1.03 and I’ve asked for 1.04. 

I couldn’t duplicate the corruption but in 1.03 the tone names are truncated and not like your 1.04 so when I can upgrade I’ll test again.

billiaz, 2020-02-07:
The names is the least of my issue, the inability to configure it was the biggest. Insecure or secure same issue 
I will compile the latest commits and try.

aretakisv, 2020-02-07:
> In the latest commits I tidied up the SoundSwitch. I’ve also realized I had firmware 1.03 and I’ve asked for 1.04.
> 
> I couldn’t duplicate the corruption but in 1.03 the tone names are truncated and not like your 1.04 so when I can upgrade I’ll test again.

I can confirm insecure works!
in secure mode it doesn't recognise everything and corruption still there.

aretakisv, 2020-02-08:
@Fishwaldo 
I get this error once I restart my gateway and request device info.. this way I loose all my  CC 121 settings
```
Error, Node011, Received a MultiChannelEncap for endpoint 2 for Command Class 121, which we can't find
```

```
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08 18:35:11.533 Detail, Node011, Notification: ValueChanged
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08 18:35:11.533 Detail, Node011, Notification: ValueChanged
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08T18:35:11.534Z z2m:Zwave zwave node 11: changed: 113-1-0:Alarm Type:0 -> 0
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08T18:35:11.535Z z2m:Zwave zwave node 11: changed: 113-1-1:Alarm Level:0 -> 0
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08T18:35:11.537Z z2m:Zwave zwave node 11: changed: 113-1-2:SourceNodeId:0 -> 0
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08T18:35:11.539Z z2m:Zwave zwave node 11: changed: 113-1-10:Burglar:9 -> 9
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08 18:35:11.610 Detail, Node011,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x0b, 0x07, 0x60, 0x0d, 0x02, 0x01, 0x79, 0x0a, 0x00, 0xe7
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08 18:35:11.610 Detail,
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08 18:35:11.611 Error, Node011, Received a MultiChannelEncap for endpoint 2 for Command Class 121, which we can't find
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08T18:35:45.850Z z2m:App User disconnected qian6EK4fdpTZHBzAAAI
zwave2mqtt2-667f97b6f7-nwl7z zwave2mqtt 2020-02-08T18:35:48.928Z z2m:App New connection s_zjHHQzxGs3KvFSAAAJ
```

Fishwaldo, 2020-02-09:
I need the full logs to diagnose that.

Fishwaldo, 2020-02-09:
This is on 1.03 for me. Works fine:
>2020-02-09 12:17:44.976 Detail, Node029,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x1d, 0x07, 0x60, 0x0d, 0x03, 0x01, 0x79, 0x0a, 0x01, 0xf1
2020-02-09 12:17:44.976 Detail, 
2020-02-09 12:17:44.977 Info, Node029, Received a MultiChannelEncap from node 29, endpoint 3 for Command Class COMMAND_CLASS_SOUND_SWITCH
2020-02-09 12:17:44.977 Info, Node029, Received SoundSwitch Tone Play report: 1
2020-02-09 12:17:44.977 Detail, Node029, Refreshed Value: old value=0, new value=1, type=list
2020-02-09 12:17:44.977 Detail, Node029, Changes to this value are not verified
2020-02-09 12:17:44.977 Detail, Node029, Notification: ValueChanged CC: COMMAND_CLASS_SOUND_SWITCH Instance: 3 Index: 1
2020-02-09 12:17:44.977 Info, Notification: Value Changed Home 0xdd12ee37 Node 29 Genre user Class SOUND SWITCH Instance 3 Index 1 Type list IsSet True ValueAsString 01DING~1 (5 sec)
List Item: 1 Selection: 01DING~1 (5 sec)
2020-02-09 12:17:45.078 Detail, Node029,   Received: 0x01, 0x13, 0x00, 0x04, 0x00, 0x1d, 0x0d, 0x60, 0x0d, 0x03, 0x01, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x0e, 0x01, 0x00, 0x13
2020-02-09 12:17:45.078 Detail, 
2020-02-09 12:17:45.078 Info, Node029, Received a MultiChannelEncap from node 29, endpoint 3 for Command Class COMMAND_CLASS_NOTIFICATION
2020-02-09 12:17:45.078 Info, Node029, Got a AlarmCmd_Report Message.... 
2020-02-09 12:17:45.078 Info, Node029, Received Notification report (>v1): Type: Siren (14) Event: Active (1) Status: true, Param Length: 0
2020-02-09 12:17:45.078 Detail, Node029, Refreshed Value: old value=0, new value=1, type=list
2020-02-09 12:17:45.078 Detail, Node029, Changes to this value are not verified
2020-02-09 12:17:45.078 Detail, Node029, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 3 Index: 14
2020-02-09 12:17:45.078 Info, Notification: Value Changed Home 0xdd12ee37 Node 29 Genre user Class ALARM Instance 3 Index 14 Type list IsSet True ValueAsString Active
2020-02-09 12:17:50.576 Detail, Node029,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x1d, 0x07, 0x60, 0x0d, 0x03, 0x01, 0x79, 0x0a, 0x00, 0xf0
2020-02-09 12:17:50.576 Detail, 
2020-02-09 12:17:50.576 Info, Node029, Received a MultiChannelEncap from node 29, endpoint 3 for Command Class COMMAND_CLASS_SOUND_SWITCH
2020-02-09 12:17:50.577 Info, Node029, Received SoundSwitch Tone Play report: 0
2020-02-09 12:17:50.577 Detail, Node029, Refreshed Value: old value=1, new value=0, type=list
2020-02-09 12:17:50.577 Detail, Node029, Changes to this value are not verified
2020-02-09 12:17:50.577 Detail, Node029, Notification: ValueChanged CC: COMMAND_CLASS_SOUND_SWITCH Instance: 3 Index: 1
2020-02-09 12:17:50.577 Info, Notification: Value Changed Home 0xdd12ee37 Node 29 Genre user Class SOUND SWITCH Instance 3 Index 1 Type list IsSet True ValueAsString Inactive
List Item: 0 Selection: Inactive
2020-02-09 12:17:50.677 Detail, Node029,   Received: 0x01, 0x13, 0x00, 0x04, 0x00, 0x1d, 0x0d, 0x60, 0x0d, 0x03, 0x01, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x0e, 0x00, 0x00, 0x12
2020-02-09 12:17:50.677 Detail, 
2020-02-09 12:17:50.677 Info, Node029, Received a MultiChannelEncap from node 29, endpoint 3 for Command Class COMMAND_CLASS_NOTIFICATION
2020-02-09 12:17:50.677 Info, Node029, Got a AlarmCmd_Report Message.... 
2020-02-09 12:17:50.677 Info, Node029, Received Notification report (>v1): Type: Siren (14) Event: Clear (0) Status: true, Param Length: 0
2020-02-09 12:17:50.677 Detail, Node029, Refreshed Value: old value=1, new value=0, type=list
2020-02-09 12:17:50.677 Detail, Node029, Changes to this value are not verified
2020-02-09 12:17:50.677 Detail, Node029, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 3 Index: 14
2020-02-09 12:17:50.678 Info, Notification: Value Changed Home 0xdd12ee37 Node 29 Genre user Class ALARM Instance 3 Index 14 Type list IsSet True ValueAsString Clear

(thats pressing the doorbell button - So endpoint 3)

aretakisv, 2020-02-09:
@Fishwaldo Which ever device I do include encrypted.. I have corruption and/or issues with the device 

An example of my Eurotronic Spirit. secured doesn't work as should, where insecure do.

I am running on Raspberry PI (arm64 open-zwave) and also with the latest doorbell patches.
The issue above also found on previous versions

The secure part, might be a more generic issue?

Fishwaldo, 2020-02-09:
I need to see the full OZW log file after removing ozwcache or attempting to include a new device securely.

billiaz, 2020-02-12:
Latest version fixes the issue.
Still pending the DB6 issue with the encryption.

billiaz, 2020-02-22:
Doorbell 6 still has corruption under encrypted inclusion... 
after almost 3 failed inclusions (secure) the last is still give corruption.
I am re-including it insecure, in order to be able to handle it.

Pastebin of log: https://pastebin.com/zrsnJr2W

LordMike, 2020-04-01:
Just chiming in to say I'm seeing this corruption issue too - and anecdotally, the issue with setting an inactive sound.

Fishwaldo, 2020-04-15:
I can't reproduce it - So maybe Locale issue. Are any of you running your linux with something other that UTF8 or C locales?

LordMike, 2020-04-15:
Mine's run in docker. I can't see an `LC_*` environment variable set in the container.. how can I check it?

aee0, 2020-08-14:
Any news on this one? I am experiencing this issue both in a Docker container and also in a Debian 9 VM (both for zwave2mqtt). On the Debian install, my locale is en_CA.UTF-8:
```
$ locale
LANG=en_CA.UTF-8
LANGUAGE=en_CA:en
LC_CTYPE="en_CA.UTF-8"
LC_NUMERIC="en_CA.UTF-8"
LC_TIME="en_CA.UTF-8"
LC_COLLATE="en_CA.UTF-8"
LC_MONETARY="en_CA.UTF-8"
LC_MESSAGES="en_CA.UTF-8"
LC_PAPER="en_CA.UTF-8"
LC_NAME="en_CA.UTF-8"
LC_ADDRESS="en_CA.UTF-8"
LC_TELEPHONE="en_CA.UTF-8"
LC_MEASUREMENT="en_CA.UTF-8"
LC_IDENTIFICATION="en_CA.UTF-8"
LC_ALL=
```

LordMike, 2020-08-14:
No news. I have not since attempted an S2 setup.

Aohzan, 2021-01-01:
Hello,

I have the same problem, with a secure inclusion or not, event after a hard reset.
When I try to update the volume:
`ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "MultiChannel Encapsulated (instance=1): SoundSwitchCmd_Tones_Config_Set`

I use zwave2mqtt addon on home assistant.

Did someone find a solution ?

billiaz, 2021-01-01:
I moved to zwavejs2mqtt. Works as expected

Gilles-Clemenceau, 2021-01-06:
> Hello,
>
> I have the same problem, with a secure inclusion or not, event after a hard reset.
> When I try to update the volume:
> `ERROR: Dropping command, expected response not received after 1 [...]

Same for me with zwave2mqtt addon on home assistant.

Crinisen, 2021-02-14:
I am having this issue as well.  I have not spent any time with Node.js so can't really dive in myself but am willing to provide whatever information needed to try and track the issue down if you just let me know what command I would need to run where and / or what logs to grab.

woddou, 2021-03-31:
Same here and it looks to be the same issue I raised today.
Buggy decryption of dataload with lenght higher than 20 char in Secure mode (my guess following code investigation) leads to corrupted OZW XML files and stop OZW panel load (due to corrupted XML). Other than that Domoticz and OZW work like a charme despite this issue but as you guys I'm not able to configure DB6 and basically to use it correctly.