OpenZWave/open-zwave #2090: Problem with Fibaro Walli Switch

DatafoxChris, 2020-01-28:
By default the Walli Switch is configured as a double-switch, controlling two internal relays. 

Replicating the problem using double-switch mode: 
Manually switch ON both switches
From the software using OpenZWave Library switch OFF the first switch
Manually switch OFF the second switch

Resulting problem:
The second switch doesn't get updated in the software, it's status remains ON. 

Quick demonstration:
https://youtu.be/mvJkxCHSnOs

This has been tested and found to work correctly on Fibaro Home Center.

Fishwaldo, 2020-01-28:
Logs please. 

If this is OZW 1.4, it’s not supported anymore. Please use 1.6 if you can.

DatafoxChris, 2020-01-28:
Using version 1.6.1019.g0d94c942.dirty. 

Tonight i will be able to get the log and post it.

DatafoxChris, 2020-01-28:
[OZW_Log.zip](https://github.com/OpenZWave/open-zwave/files/4124368/OZW_Log.zip)

The Walli Switch is Node 87, i repeated the video while capturing the debug log.

Fishwaldo, 2020-01-29:
Node 87?

>2020-01-28 21:10:52.844 Detail, Node087,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x57, 0x07, 0x60, 0x0d, 0x01, 0x01, 0x25, 0x03, 0xff, 0x12
2020-01-28 21:10:52.844 Detail, 
2020-01-28 21:10:52.844 Info, Node087, Received a MultiChannelEncap from node 87, endpoint 1 for Command Class COMMAND_CLASS_SWITCH_BINARY
2020-01-28 21:10:52.844 Info, Node087, Received SwitchBinary report from node 87: level=On

There is the "report" for the first switch (endpoint 1)

>2020-01-28 21:10:56.078 Detail, Node087,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x57, 0x07, 0x60, 0x0d, 0x02, 0x01, 0x25, 0x03, 0xff, 0x11
2020-01-28 21:10:56.078 Detail, 
2020-01-28 21:10:56.078 Info, Node087, Received a MultiChannelEncap from node 87, endpoint 2 for Command Class COMMAND_CLASS_SWITCH_BINARY
2020-01-28 21:10:56.078 Info, Node087, Received SwitchBinary report from node 87: level=On

There is the "report" for the second switch (endpoint 2)

Doesn't appear to be anything wrong with OZW. I'd check with Domoticz in this case.

DatafoxChris, 2020-01-29:
The problem is in the switch off event of the second switch, after the first switch has been switched off from within the software and the second has been switched off manually. Then the status of the second fwitch-off is not updated into the software. Same problem in Home Assitant.

petergebruers, 2020-01-29:
I was going to say the same thing as @Fishwaldo - there is feedback from both endpoint 1 and endpoint 2 in your log file, so the relay does send data from both relays. If you say you expect _more_ feedback on OZW then the device is not sending it and it is really hard to say what is going on in your case... I don't own a "Walli", they look great but are rather expensive.

> This has been tested and found to work correctly on Fibaro Home Center.

Hard to tell if HC does something different, but it might use "supervision" class. So this does not completely rule out a firmware issue or peculiarity of the device.

I cannot think of any better way to see what is happening then to enable debug logging, toggle that switch and watch OZW_Log.txt and see if you get those "Received a MultiChannelEncap from node 87" followed by "Received SwitchBinary report from node 87". Is it consistent? Random?

DatafoxChris, 2020-01-30:
Perhaps this is an explanation:

This is the moment the second switch is switched Off by hand, which refers to 'Endpoint 2', while all other Electric messages refer to 'Endpoint 1': 

2020-01-28 21:11:13.829 Detail, Node087,   Received: 0x01, 0x14, 0x00, 0x04, 0x00, 0x57, 0x0e, 0x60, 0x0d, 0x02, 0x01, 0x32, 0x02, 0x21, 0x32, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xfb
2020-01-28 21:11:13.829 Detail, 
2020-01-28 21:11:13.829 Info, Node087, Received a MultiChannelEncap from node 87, endpoint 2 for Command Class COMMAND_CLASS_METER
2020-01-28 21:11:13.829 Info, Node087, Received Meter Report for Electric - W (1) with Units W (2) on Index 2: 0.0
2020-01-28 21:11:13.829 Detail, Node087, Refreshed Value: old value=0.0, new value=0.0, type=decimal
2020-01-28 21:11:13.829 Detail, Node087, Changes to this value are not verified
2020-01-28 21:11:13.829 Detail, Node087, Refreshed Value: old value=false, new value=false, type=bool
2020-01-28 21:11:13.829 Detail, Node087, Changes to this value are not verified
2020-01-28 21:11:13.829 Detail, Node087, Notification: ValueChanged
2020-01-28 21:11:13.829 Detail, Node087, Notification: ValueChanged

DatafoxChris, 2020-01-30:
petergebruers, fishwaldo: i am happy to send you a Walli Switch for further investigation. 
Just let me know.

DatafoxChris, 2020-01-30:
Tomorrow the Dutch Importer will update the firmware of one of my Walli Switches from 5.1 to 5.2, let's hope this solves the issue.

petergebruers, 2020-01-30:
> petergebruers, fishwaldo: i am happy to send you a Walli Switch for further investigation.
> Just let me know.

I have to think about that. I have a small issue, I have "health problems" and I am very slow at... well... at anything really!

> Tomorrow the Dutch Importer will update the firmware of one of my Walli Switches from 5.1 to 5.2, let's hope this solves the issue.

That might help, unfortunately we do not have a very good relationship with Fibaro, so it is very hard to get info from them. To be honest, I have never tried to make that relationship work, I only joined OZW last year. I am unsure how they treat open source these days but I do know they are very busy with the release of their HomeCenter 3...

DatafoxChris, 2020-02-06:
Today i am going to pick up the upgraded Walli Switches

DatafoxChris, 2020-02-06:
FW 5.2 did not solve this issue. The same Walli Switch works correctly on the Home Center 3.

petergebruers, 2020-02-06:
Yeah Fibaro HC2/3 Z-Wave implementation is not bad, contrary to what some people believe...

If you click that switch, do you see _anything_ from that node in OZW_Log? If you see nothing, then device is not sending data. What do you have in group 1? Does it say "1" or "1.1"?

DatafoxChris, 2020-02-07:
My test: switch both switches on by hand, switch off the first switch by software, and then switch off the second switch by hand: the software doesn't get informed about switch-off of the second button now. 

When switching off the second button by hand, there is a Power report logged.

DatafoxChris, 2020-02-07:
To answer your second question: The Walli has '1.1' in Group 1, and nothing in the other groups. 

The importer asked me to check scene settings of the Walli (will do this in the weekend):

40. First button – scenes sent: 0   (current: 0)
41. Second button – scenes sent: 0   (current: 0)

This parameter determines which actions result in sending scene IDs assigned to them. Values can be combined (e.g. 1+2=3 means that scenes for single and double click are sent). Enabling scenes for triple click disables entering the device in learn mode by triple clicking. 1 - Key pressed 1 time 2 - Key pressed 2 time 4 - Key pressed 3 time 8 - Key hold down and key released

petergebruers, 2020-02-07:
Thanks

> The importer asked me to check scene settings

Well, the "scene" settings could certainly offer some benefit: without them I expect the Walli to send _some status_ (but you see none, that is the unexpected part of this issue). By activating central scene support you get those extra click patterns (eg double click). That is a workaround, but a good one imho.

What happens if you add a another relay to group 4, can you control it?
What happens if you add 1.1 to Group 4?

4th association group – “On/Off (2)” is used to turn the associated devices on/off reflecting button operation for 2nd channel (uses Basic command class).

DatafoxChris, 2020-02-07:
The Walli only has 1.1 in group 1, the other groups (2,3,4,5) are empty. This has a good chance of explaining my problem. The current node configuration:

156. First channel – Switch ON value sent to 2nd and 3rd association groups: 255 (huidig: 255)

This parameter defines value sent with Switch ON command to devices associated in 2nd and 3rd association group for manual changes of Endpoint 1 state. 0-99 - 2nd association group 255 - 3rd association group

157. First channel – Switch OFF value sent to 2nd and 3rd association groups: 0 (huidig: 0)

This parameter defines value sent with Switch OFF command to devices associated in 2nd and 3rd association group for manual changes of Endpoint 1 state. 0-99 - 2nd association group 255 - 3rd association group

158. First channel – Double Click value sent to 2nd and 3rd association groups: 99 (huidig: 99)

This parameter defines value sent with Double Click command to devices associated in 2nd and 3rd association group for manual changes of Endpoint 1 state. 0-99 - 2nd association group 255 - 3rd association group

159. Second channel – Switch ON value sent to 4th and 5th association groups: 255 (huidig: 255)

This parameter defines value sent with Switch ON command to devices associated in 4th and 5th association group for manual changes of Endpoint 2 state. 0-99 - 4th association group 255 - 5th association group

160. Second channel – Switch OFF value sent to 4th and 5th association groups: 0 (huidig: 0)

This parameter defines value sent with Switch OFF command to devices associated in 4th and 5th association group for manual changes of Endpoint 2 state. 0-99 - 4th association group 255 - 5th association group

161. Second channel – Double Click value sent to 4th and 5th association groups:  99  (huidig: 99)

This parameter defines value sent with Double Click command to devices associated in 4th and 5th association group for manual changes of Endpoint 2 state. 0-99 - 4th association group 255 - 5th association group

DatafoxChris, 2020-02-10:
Very busy making preparations for placement of new glass windows in my home on wednesday. Tonight i hope to find time for the zwave groups-test.

DatafoxChris, 2020-02-11:
Tested adding the main zwave controller into groups 2 and 4: unfortunatly this doesn't solve the issue.

Fishwaldo, 2020-02-16:
Can you post a new log file with ozwcache removed before starting? Lets double check - But smells like a problem with the device unfortuantly. Lets confirm.

DatafoxChris, 2020-02-17:
[OZW_Log.zip](https://github.com/OpenZWave/open-zwave/files/4216340/OZW_Log.zip)

This is the log recorded after cache had been removed. Same problem with the Walli Switch.

Fishwaldo, 2020-02-19:
Ok - I see Node 1.1 is added to Group 1 (lifeline), 2, and 4

At 22:41:06 I see we get a COMMAND_CLASS_SWITCH_BINARY report for Endpoint 1
at 22:41:09 we get a COMMAND_CLASS_METER report for endpoint 1
at 22:41:10 we get another COMMAND_CLASS_SWITCH_BINARY for endpoint 1
at 22:41:14 another COMMAND_CLASS_METER report for endpoint 1
at 22:41:25 we get a COMMAND_CLASS_SWITCH_BINARY for endpoint *2*
at 22:41:28 we get a COMMAND_CLASS_METER for endpoint *2*
at 22:41:29 we get a COMMAND_CLASS_SWITCH_BINARY for endpoint 1
at 22:41:37 you switch both instances off (instance 2 - endpoint 1, instance 3 - endpoint 2)
Futher down, we do a get to confirm the changes were made. Instance 3 returns endpoint 2 - Good.
*BUT* Instance 2 returns endpoint 2:
```
2020-02-17 22:41:37.979 Info, Node089, Sending (Send) message (Callback ID=0xb4, Expected Reply=0x13) - MultiChannel Encapsulated (instance=2): SwitchBinaryCmd_Set (Node=89): 0x01, 0x0e, 0x00, 0x13, 0x59, 0x07, 0x60, 0x0d, 0x01, 0x02, 0x25, 0x01, 0x00, 0x25, 0xb4, 0x67
2020-02-17 22:41:37.979 Info, Node089, Encrypted Flag is 0
2020-02-17 22:41:37.979 Detail, Node089,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-02-17 22:41:37.979 Detail, Node089,   ZW_SEND_DATA delivered to Z-Wave stack
2020-02-17 22:41:37.995 Detail, Node089,   Received: 0x01, 0x07, 0x00, 0x13, 0xb4, 0x00, 0x00, 0x02, 0x5d
2020-02-17 22:41:37.995 Detail, Node089,   ZW_SEND_DATA Request with callback ID 0xb4 received (expected 0xb4)
2020-02-17 22:41:37.995 Info, Node089, Request RTT 15 Average Request RTT 17
2020-02-17 22:41:37.995 Detail, Node089,   Expected callbackId was received
2020-02-17 22:41:37.995 Detail, Node089,   Expected reply was received
2020-02-17 22:41:37.995 Detail, Node089,   Message transaction complete
2020-02-17 22:41:37.995 Detail, 
2020-02-17 22:41:37.995 Detail, Node089, Removing current message
2020-02-17 22:41:37.995 Detail, 
2020-02-17 22:41:37.995 Info, Node089, Sending (Send) message (Callback ID=0xb5, Expected Reply=0x04) - MultiChannel Encapsulated (instance=2): SwitchBinaryCmd_Get (Node=89): 0x01, 0x0d, 0x00, 0x13, 0x59, 0x06, 0x60, 0x0d, 0x01, 0x02, 0x25, 0x02, 0x25, 0xb5, 0x67
2020-02-17 22:41:37.995 Info, Node089, Encrypted Flag is 0
2020-02-17 22:41:38.011 Detail, Node089,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-02-17 22:41:38.011 Detail, Node089,   ZW_SEND_DATA delivered to Z-Wave stack
2020-02-17 22:41:38.026 Detail, Node089,   Received: 0x01, 0x07, 0x00, 0x13, 0xb5, 0x00, 0x00, 0x02, 0x5c
2020-02-17 22:41:38.026 Detail, Node089,   ZW_SEND_DATA Request with callback ID 0xb5 received (expected 0xb5)
2020-02-17 22:41:38.026 Info, Node089, Request RTT 31 Average Request RTT 24
2020-02-17 22:41:38.026 Detail, Node089,   Expected callbackId was received
2020-02-17 22:41:38.026 Detail, Node089,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x59, 0x07, 0x60, 0x0d, 0x02, 0x01, 0x25, 0x03, 0x00, 0xe0
2020-02-17 22:41:38.026 Detail, 
2020-02-17 22:41:38.026 Info, Node089, Response RTT 31 Average Response RTT 30
2020-02-17 22:41:38.026 Info, Node089, Received a MultiChannelEncap from node 89, endpoint 2 for Command Class COMMAND_CLASS_SWITCH_BINARY
2020-02-17 22:41:38.026 Info, Node089, Received SwitchBinary report from node 89: level=Off
```

It should return instance 1 - This seems like a bug

You could try [MapRootToEndpoint](https://github.com/OpenZWave/open-zwave/wiki/CommandClass-Compatibility-Flags#multiinstance-commandclass)

DatafoxChris, 2020-02-19:
Walli Switch is Node 093

Node 089 is a relay:  FIBARO System FGS222 Double Relay Switch 2x1.5kW

This relay is copying the on/off states of the Walli Switch through a script.

Fishwaldo, 2020-02-20:
Sorry - I was in a rush when I was looking at your logs:

At 22:41:37 we set COMMAND_CLASS_SWITCH_BINARY endpoint 2 and get a Report for Endpoint 2
At 22:41:42 we get a METER report for endpoint 2
At 22:41:45 we get a METER report for endpoint 1

Fishwaldo, 2020-02-20:
So looking at your logs, OZW is actually getting messages COMMAND_CLASS_SWITCH_BINARY for endpoints 1 and 2 (I'm assuming this is where you physically flicked the switch). 

Can you confirm this?

Again, I would try the MapRootToEndpoint config option if its not working as you expect. Otherwise I'm at a loss.

DatafoxChris, 2020-02-20:
MapRootToEndpoint can be added to existing ozw cache ? 

I can make another debug-log and write down the exact timeframe of executing the test.

Much easier for you: Would you like me to send one Walli Switch to you ?

I can just buy some other double-swich: Which up/down zwave switch is supported by OZW ? The purpose is to control a sun-screen.

DatafoxChris, 2020-02-21:
[OZW_Log.zip](https://github.com/OpenZWave/open-zwave/files/4236800/OZW_Log.zip)

The test has been executed at 16:42 in this debug-logfile using Node093.

Romke10, 2020-10-18:
I have the same problem with my Walli switch in combination with open z-wave. Open Z-wave is not receviving any status updates when I press the second switch on my Walli.

The first switch works.

Did you find a solution?

tobiasgraf, 2021-10-21:
I found a solution for this that works for me.

I simply added 
```
<CommandClass id="142" ForceInstances="true"/>
```
to the configuration xml file (and remove and re-add the device)

With this the device reports the power and switch status for the 2nd channel.

note: i'm still on a 1.4 version :see_no_evil: 

so for a more recent version it should be something like i guess

```
  <CommandClass id="142">
    <Compatibility>
      <ForceInstances>true</ForceInstances>
    </Compatibility>
  </CommandClass>
```