OpenZWave/open-zwave #2099: Find alternative way to "version" non-git source

petergebruers, 2020-02-05:
When I added a version test to the framework, the build server claimed "1.6.0".

That's why the test has this part:

```cpp
	// Allow ozw_vers_revision to be zero, to avoid build server failure:
	// "ValueID_test.cpp:42: Failure"
	// "Expected: (ozw_vers_revision) >= (900), actual: 0 vs 900 [ FAILED ] OpenZWave.Version (0 ms)""
	if (ozw_vers_revision != 0)
	{
		EXPECT_GE(ozw_vers_revision, 992);
```

So I added debug code to and it finally dawned upon me that it depends not only on the presence of git, but also on the presence of the .git folder... Which is absent if you download a release version.

If you download and build the "zip"...

"git describe returned an empty result, setting GITVERSION to VERSION_MAJ.VERSION_MIN.-1 and VERSION_REV to 0"

The code path hasn't changed, I only added debugging...

Fishwaldo, 2020-02-05:
I thought at one stage - we packaged vers.cpp with the “make dist” command (won’t work downloading the zip from GitHub). 

Maybe we need a test - if vers.cpp exists and no .git directory, be quiet. 

We are getting that error in the PR build tests using github actions as well.

Fishwaldo, 2020-02-06:
See the GitVersionFix Branch... :)

Fishwaldo, 2020-02-06:
(you can test a "download" with make dist)

petergebruers, 2020-02-06:
Thanks. Changes make sense to me. "make dist" and version work OK on my Mac.

LGTM

Because Windows does not use make, that is still broken (Version String:       1.6.0.NoGit). Maybe I can fix it but I haven't touched .bat files in 10 years ;)

Fishwaldo, 2020-02-06:
Yeah.. Windows... :)

closed 2020-02-06 by Fishwaldo