OpenZWave/open-zwave #2102: [bug] AEOTEC Z-stick Gen5 cannot change device parameters (class 112)

aretakisv, 2020-02-07:
I have an aeotec Z-stick. I cannot enable or disable led (where with zensys I can)
I used both Qt-openzwave and zwave2mqtt. Same behaviour

An example with qt-openzwave.
Led parameter json:
```
{
    "Label": "LED indicator configuration",
    "Value": {
        "List": [
            {
                "Value": 0,
                "Label": "Disable"
            },
            {
                "Value": 1,
                "Label": "Enable"
            }
        ],
        "Selected": "Enable"
    },
    "Units": "",
    "Min": 0,
    "Max": 1,
    "Type": "List",
    "Instance": 1,
    "CommandClass": "COMMAND_CLASS_CONFIGURATION",
    "Index": 81,
    "Node": 1,
    "Genre": "Config",
    "Help": "Enable/Disable LED indicator when plugged in",
    "ValueIDKey": 22799473140563988,
    "ReadOnly": false,
    "WriteOnly": false,
    "ValueSet": false,
    "ValuePolled": false,
    "ChangeVerified": false,
    "Event": "valueAdded",
    "TimeStamp": 1581072955
}
```

Command to set value:
```
sent to: OpenZWave/1/command/setvalue/
Payload:
{
  "ValueIDKey": 22799473140563988,
  "Value": 1
}
```
Expected behaviour: led to work. Doesn't work

Debug log
```
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.732 UTC] [qt.mqtt.connection.verbose] [debug]: Received PUBLISH
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.732 UTC] [qt.mqtt.connection.verbose] [debug]: Finalize PUBLISH: topic: QMqttTopicName("OpenZWave/1/command/setvalue/")  payloadLength: 51
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.732 UTC] [ozw.mqtt.commands] [debug]: Got  "OpenZWave/1/command/setvalue/"  Message:  "{\n  \"ValueIDKey\": 22799473140563988,\n  \"Value\": 1\n}"
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.733 UTC] [ozw.mqtt.commands.setValue] [info]: Setting  22799473140563988  to Value  QVariant(QTOZW_ValueIDList, )
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.733 UTC] [ozw.values] [debug]: setData Called for Row 6  With Value QVariant(QTOZW_ValueIDList, )
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.733 UTC] [ozw.values] [debug]: valueModel Changed! QTOZW_ValueIds::Value - 6 :  QVariant(QTOZW_ValueIDList, )
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.733 UTC] [ozw.values] [debug]: valueModel Changed! vidKey: 22799473140563988 QVariant(QTOZW_ValueIDList, )
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.734 UTC] [qt.remoteobjects.models] [debug]: void QAbstractItemModelSourceAdapter::sourceDataChanged(const QModelIndex&, const QModelIndex&, const QVector<int>&) const start= (ModelIndex[row=6, column=1]) end= (ModelIndex[row=6, column=1]) neededRoles= QVector(0)
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.734 UTC] [qt.mqtt.connection] [debug]: qint32 QMqttConnection::sendControlPublish(const QMqttTopicName&, const QByteArray&, quint8, bool, const QMqttPublishProperties&) QMqttTopicName("OpenZWave/1/event/setvalue/")  Size: 51  bytes. QoS: 0  Retain: false
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.734 UTC] [qt.mqtt.connection.verbose] [debug]: bool QMqttConnection::writePacketToTransport(const QMqttControlPacket&)  DataSize: 82
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.734 UTC] [ozw.mqtt.commands] [info]: Processed Message for  "SetValue" :  "{\n  \"ValueIDKey\": 22799473140563988,\n  \"Value\": 1\n}"
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.735 UTC] [qt.mqtt.connection.verbose] [debug]: void QMqttConnection::transportReadReady()
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.735 UTC] [qt.mqtt.connection.verbose] [debug]: Received PUBLISH
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:12:57.736 UTC] [qt.mqtt.connection.verbose] [debug]: Finalize PUBLISH: topic: QMqttTopicName("OpenZWave/1/event/setvalue/")  payloadLength: 51
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:13:25.282 UTC] [qt.mqtt.connection] [debug]: qint32 QMqttConnection::sendControlPublish(const QMqttTopicName&, const QByteArray&, quint8, bool, const QMqttPublishProperties&) QMqttTopicName("OpenZWave/1/statistics/")  Size: 441  bytes. QoS: 0  Retain: false
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:13:25.283 UTC] [qt.mqtt.connection.verbose] [debug]: bool QMqttConnection::writePacketToTransport(const QMqttControlPacket&)  DataSize: 469
ozwdaemon-55499fcb78-l6c6w ozwdaemon [20200207 11:13:25.284 UTC] [qt.mqtt.connection] [debug]: qint32 QMqttConnection::sendControlPublish(const QMqttTopicName&, const QByteArray&, quint8, bool, const QMqttPublishProperties&) QMqttTopicName("OpenZWave/1/node/1/statistics/")  Size: 774  bytes. QoS: 0  Retain: false
```

Fishwaldo, 2020-02-07:
its a limitation of the Protocol. You can't set params on your Controller. (if you have a secondary controller, you could setup config params on it).

There is a hack floating around the internet to disable them.

closed 2020-02-07 by Fishwaldo

aretakisv, 2020-02-07:
didn't know that. to enable encryption I need a secondary controller to send the command through.
As I understand Z-stick uses it's own setting for encryption and ignores the apps.

petergebruers, 2020-02-07:
If you are on linux, stop OZW and try this to disable "disco light". I don't own one but it is my docs :)

echo -e -n "\x01\x08\x00\xF2\x51\x01\x00\x05\x01\x51" > /dev/serial/by-id/usb-0658_0200-if00

> to enable encryption I need a secondary controller to send the command through.
> As I understand Z-stick uses it's own setting for encryption and ignores the apps.

Never heard of that before, OZW handles S0 encryption with any stick. Set a key in your gateway software and do "secure inclusion"...