OpenZWave/open-zwave #2106: Lock wants to be added securely, but it doesn't

alex-savin, 2020-02-09:
I tried to include a Kwikset 910 lock with secure inclusion a couple times and it was included but as a regular device and I it doesn't work properly

```
2020-02-08 13:15:23.579 Detail, Node037, Notification: NodeAdded
2020-02-08 13:15:23.579 Detail, Node037, Notification: NodeProtocolInfo
2020-02-08 13:15:23.579 Detail, Node037, Notification: AddDevice ControllerCommand - Completed
2020-02-08 13:15:23.580 Info, Node037, Sending (Command) message (Callback ID=0xa3, Expected Reply=0x04) - SecurityCmd_SchemeGet (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x98, 0x04, 0x00, 0x25, 0xa3, 0xda
2020-02-08 13:15:23.580 Info, Node037, Encrypted Flag is 0
2020-02-08 13:15:23.590 Detail, Node037,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-02-08 13:15:23.590 Detail, Node037,   ZW_SEND_DATA delivered to Z-Wave stack
2020-02-08 13:15:27.519 Detail, Node037,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x25, 0x03, 0x98, 0x05, 0x00, 0x49
2020-02-08 13:15:27.520 Info, Node037, Response RTT 3940 Average Response RTT 3940
2020-02-08 13:15:27.520 Info, Node037, Received SecurityCmd_SchemeReport from node 37: 0
2020-02-08 13:15:27.520 Info, Node037,     Security scheme agreed.
2020-02-08 13:15:27.520 Detail, Node037, Queuing (Command) SecurityCmd_NetworkKeySet (Node=37): 0x01, 0x19, 0x00, 0x13, 0x25, 0x12, 0x98, 0x06, 0x06, 0x07, 0xd6, 0x41, 0x8c, 0x25, 0xe5, 0xa2, 0x23, 0x52, 0x37, 0xed, 0x59, 0x72, 0xab, 0x9d, 0x25, 0xa4, 0x13
2020-02-08 13:15:27.940 Detail, Node037,   Received: 0x01, 0x09, 0x00, 0x04, 0x10, 0x25, 0x03, 0x98, 0x05, 0x00, 0x59
2020-02-08 13:15:27.940 Info, Node037, Response RTT 4360 Average Response RTT 4150
2020-02-08 13:15:27.940 Info, Node037, Received SecurityCmd_SchemeReport from node 37: 0
2020-02-08 13:15:27.940 Warning, Node037,    Already Received a SecurityCmd_SchemeReport from the node. Ignoring
2020-02-08 13:15:28.705 Detail, Node037,   Received: 0x01, 0x09, 0x00, 0x04, 0x10, 0x25, 0x03, 0x98, 0x05, 0x00, 0x59
2020-02-08 13:15:28.705 Info, Node037, Response RTT 5125 Average Response RTT 4637
2020-02-08 13:15:28.705 Info, Node037, Received SecurityCmd_SchemeReport from node 37: 0
2020-02-08 13:15:28.705 Warning, Node037,    Already Received a SecurityCmd_SchemeReport from the node. Ignoring
2020-02-08 13:15:28.756 Detail, Node037,   Received: 0x01, 0x09, 0x00, 0x04, 0x10, 0x25, 0x03, 0x98, 0x05, 0x00, 0x59
2020-02-08 13:15:28.756 Info, Node037, Response RTT 5176 Average Response RTT 4906
2020-02-08 13:15:28.756 Info, Node037, Received SecurityCmd_SchemeReport from node 37: 0
2020-02-08 13:15:28.756 Warning, Node037,    Already Received a SecurityCmd_SchemeReport from the node. Ignoring
2020-02-08 13:15:29.721 Detail, Node037,   Received: 0x01, 0x09, 0x00, 0x04, 0x10, 0x25, 0x03, 0x98, 0x05, 0x00, 0x59
2020-02-08 13:15:29.721 Info, Node037, Response RTT 6141 Average Response RTT 5523
2020-02-08 13:15:29.721 Info, Node037, Received SecurityCmd_SchemeReport from node 37: 0
2020-02-08 13:15:29.721 Warning, Node037,    Already Received a SecurityCmd_SchemeReport from the node. Ignoring
2020-02-08 13:15:31.408 Detail, Node037,   Received: 0x01, 0x07, 0x00, 0x13, 0xa3, 0x00, 0x03, 0x0e, 0x45
2020-02-08 13:15:31.408 Detail, Node037,   ZW_SEND_DATA Request with callback ID 0xa3 received (expected 0xa3)
2020-02-08 13:15:31.408 Info, Node037, Request RTT 7829 Average Request RTT 7829
2020-02-08 13:15:31.408 Detail, Node037,   Expected callbackId was received
2020-02-08 13:15:33.580 Error, Node037, ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "SecurityCmd_SchemeGet (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x98, 0x04, 0x00, 0x25, 0xa3, 0xda"
2020-02-08 13:15:33.580 Detail, Node037, Removing current message
2020-02-08 13:15:33.580 Detail, Node037, Notification: Notification - TimeOut
2020-02-08 13:15:33.580 Detail, Node037, Notification: ControllerCommand - Error - Failed
2020-02-08 13:15:33.580 Info, Node037, Processing (Command) Nonce Request message (Callback ID=0xa4, Expected Reply=0x04)
2020-02-08 13:15:33.581 Info, Node037, Sending (Command) message (Callback ID=0x02, Expected Reply=0x04) - Nonce_Get(SecurityCmd_NetworkKeySet) - 0x01, 0x09, 0x00, 0x13, 0x25, 0x02, 0x98, 0x40, 0x05, 0x02:
2020-02-08 13:15:33.587 Detail, Node037,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-02-08 13:15:33.588 Detail, Node037,   ZW_SEND_DATA delivered to Z-Wave stack
2020-02-08 13:15:33.659 Detail, Node037,   Received: 0x01, 0x07, 0x00, 0x13, 0x02, 0x00, 0x00, 0x07, 0xee
2020-02-08 13:15:33.659 Detail, Node037,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)
2020-02-08 13:15:33.659 Info, Node037, Request RTT 78 Average Request RTT 3953
2020-02-08 13:15:33.733 Detail, Node037,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x25, 0x0a, 0x98, 0x80, 0xf6, 0x46, 0x04, 0x4a, 0xea, 0x37, 0x5a, 0x42, 0xe7
2020-02-08 13:15:33.734 Info, Node037, Received SecurityCmd_NonceReport from node 37
2020-02-08 13:15:33.734 Info, Node037, Encrypted Flag is 1
2020-02-08 13:15:33.734 Info, Node037, Sending (Command) message (Callback ID=0xa5, Expected Reply=0x04) - SecurityCmd_NetworkKeySet (Node=37): 0x01, 0x19, 0x00, 0x13, 0x25, 0x12, 0x98, 0x06, 0x06, 0x07, 0xd6, 0x41, 0x8c, 0x25, 0xe5, 0xa2, 0x23, 0x52, 0x37, 0xed, 0x59, 0x72, 0xab, 0x9d, 0x25, 0xa5, 0x12
2020-02-08 13:15:33.746 Detail, Node037,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-02-08 13:15:33.746 Detail, Node037,   ZW_SEND_DATA delivered to Z-Wave stack
2020-02-08 13:15:33.905 Detail, Node037,   Received: 0x01, 0x07, 0x00, 0x13, 0xa5, 0x00, 0x00, 0x10, 0x5e
2020-02-08 13:15:33.905 Detail, Node037,   ZW_SEND_DATA Request with callback ID 0xa5 received (expected 0xa5)
2020-02-08 13:15:33.905 Info, Node037, Request RTT 323 Average Request RTT 2138
2020-02-08 13:15:33.905 Detail, Node037,   Expected callbackId was received
2020-02-08 13:15:43.581 Error, Node037, ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "SecurityCmd_NetworkKeySet (Node=37): 0x01, 0x19, 0x00, 0x13, 0x25, 0x12, 0x98, 0x06, 0x06, 0x07, 0xd6, 0x41, 0x8c, 0x25, 0xe5, 0xa2, 0x23, 0x52, 0x37, 0xed, 0x59, 0x72, 0xab, 0x9d, 0x25, 0xa5, 0x12"
2020-02-08 13:15:43.581 Detail, Node037, Removing current message
2020-02-08 13:15:43.581 Detail, Node037, Notification: Notification - TimeOut
2020-02-08 13:15:43.581 Detail, Node037, Notification: ControllerCommand - Error - Failed
2020-02-08 13:15:53.095 Info, Node037, Skipping Cache Save for Node 37 as its not past QueryStage_CacheLoad
2020-02-08 13:15:53.180 Info, Node037, Sending (Query) message (Callback ID=0x00, Expected Reply=0x41) - Get Node Protocol Info (Node=37): 0x01, 0x04, 0x00, 0x41, 0x25, 0x9f
2020-02-08 13:15:53.180 Info, Node037, Encrypted Flag is 0
2020-02-08 13:15:53.182 Detail, Node037,   Received: 0x01, 0x09, 0x01, 0x41, 0x53, 0xdc, 0x00, 0x04, 0x40, 0x03, 0x7e
2020-02-08 13:15:53.182 Info, Node037, Received reply to FUNC_ID_ZW_GET_NODE_PROTOCOL_INFO
2020-02-08 13:15:53.182 Info, Node037,   Protocol Info for Node 37:
2020-02-08 13:15:53.182 Info, Node037,     Listening     = false
2020-02-08 13:15:53.182 Info, Node037,     Frequent      = true
2020-02-08 13:15:53.182 Info, Node037,     Beaming       = true
2020-02-08 13:15:53.182 Info, Node037,     Routing       = true
[... 377 lines cut ...]
2020-02-08 13:19:34.343 Detail, Node037, Queuing (Query) ConfigurationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x70, 0x05, 0x17, 0x25, 0xca, 0x4d
2020-02-08 13:19:34.344 Detail, Node037, Queuing (Query) ConfigurationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x70, 0x05, 0x18, 0x25, 0xcb, 0x43
2020-02-08 13:19:34.344 Detail, Node037, Queuing (Query) ConfigurationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x70, 0x05, 0x19, 0x25, 0xcc, 0x45
2020-02-08 13:19:34.344 Detail, Node037, Queuing (Query) ConfigurationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x70, 0x05, 0x1a, 0x25, 0xcd, 0x47
2020-02-08 13:19:34.344 Detail, Node037, Queuing (Query) ConfigurationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x70, 0x05, 0x1b, 0x25, 0xce, 0x45
2020-02-08 13:19:34.344 Detail, Node037, Queuing (Query) ConfigurationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x70, 0x05, 0x1c, 0x25, 0xcf, 0x43
2020-02-08 13:19:34.344 Detail, Node037, Queuing (Query) ConfigurationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x70, 0x05, 0x1d, 0x25, 0xd0, 0x5d
[... cut ...]
2020-02-08 13:19:35.958 Info, Node037, Request RTT 1613 Average Request RTT 1673
2020-02-08 13:19:35.958 Detail, Node037,   Expected callbackId was received
2020-02-08 13:19:44.346 Error, Node037, ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "ConfigurationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x70, 0x05, 0x01, 0x25, 0xb4, 0x25"
2020-02-08 13:19:44.346 Detail, Node037, Removing current message
2020-02-08 13:19:44.346 Detail, Node037, Notification: Notification - TimeOut
```

Fishwaldo, 2020-02-09:
I need log files that have not been filtered.

Probably related to #1981

petergebruers, 2020-02-09:
> Probably related to #1981

FWIW I do not own a door lock, but I can simulate one by using a Z-Uno, so I've tried my sketch with "master branch" which may be a bit more recent then what @alex-savin is running... So mine is "OpenZwave Version 1.6-1032-g38ff332f "

I see no issues.

COMMAND_CLASS_DOOR_LOCK -> OK
Received SecurityCmd_SchemeReport from node 7: 0 -> OK
Setting Up Inclusion Network Key for Secure Communications -> OK

Status report from lock to controller:

Info, Notification: Value Changed Home 0xd96a77cd Node 7 Genre user Class DOOR LOCK Instance 1 Index 0 Type bool IsSet True ValueAsString False -> OK
Info, Notification: Value Changed Home 0xd96a77cd Node 7 Genre user Class DOOR LOCK Instance 1 Index 0 Type bool IsSet True ValueAsString True -> OK

@alex-savin as @Fishwaldo said, please restart OZW, try to include your lock then zip and drag&drop the OZW_Log.txt here please... BTW could it be the lock is simply a bit too far away (unreliable communication during inclusion)?

alex-savin, 2020-02-09:
Here is a full log file
https://file.io/HSzHhL

First time lock was included as a Node036, that I excluded it and re add one more time as a Node037

Fishwaldo, 2020-02-10:
So this appears to be a OLD ZWave Device. (Pre-Zwave+). As @petergebruers older locks required inclusion to be done "very close" to the device (they don't support NWI and wont use the mesh when including). Can you confirm how close you are to the Lock when trying to include?

Its timing out when trying to set the new Network Key.

(its also acting a bit strange, sending the same message over and over again - SchemaGet)

Also - You have a lot of activity going on while including this device. the Security Protocol has some timeouts involved, and all the other traffic might mean your hitting a timeout... Any way you can "not move around" as I see a lot of motion sensors going off!

Protocol wise, OZW is acting according to specs... but then this device... I don't know what quirks it might have.

Regardless, it does not appear to be related to #1981 after all.

Fishwaldo, 2020-02-10:
based on a quick search on google - you need to be very physically close to pair it correctly:
https://community.hubitat.com/t/kwikset-910-z-wave-lock/1118/18

alex-savin, 2020-02-10:
@Fishwaldo the lock it pretty close to hub, something about 6 feet.
I was messing around and successfully connected it to SmartThings with the same distance.

Anyway I will try to shutdown other devices (I mean switches which send power/energy reports) to minify network traffic and put it as close as it possible.

Fishwaldo, 2020-02-10:
If you read some of the comments online, they talk about inches, not feet and a lot of people say its really troublesome to get included (not just with OZW). 

So assuming you can quieten your network down, and it still doesn't work - try moving closer.

(FYI, prior to Z-Wave Plus - The instructions were pretty adamant it had to be "next to the controller" - A "Security" feature was that this all happened with very low RF power so no one could sniff your keys)

alex-savin, 2020-02-12:
@Fishwaldo You were right!
When I moved lock as close as possible to hub, it successfully securely included to the network.

Thank you!

closed 2020-02-12 by alex-savin