OpenZWave/open-zwave #2123: Reporting crash during initialization

elcosmelco, 2020-02-18:
Ello!

In trying the 'latest' version of OZW using Zwave2Mqtt and the OZWDaemon from Qt-OpenZwave I get a consistent crash during initialization. 
 
I am a Windows dot net guy (used to the old dot net wrapper) but out of necessity I had to venture out and try different ways to use OZW - so both attempts are Docker containers on an Rpi running Raspbian. Both are using the 'latest' version of OZW, which I'm assuming is 1.6?
 
I was wondering what the easiest way to report the crash is. When the latter crashed it uploaded the minidump somewhere which I'm hoping will be of use to someone.
 
Any help here would be greatly appreciated as I am keen on using an up-to-date version of OZW in my software.

Regards,
Elco

Fishwaldo, 2020-02-19:
Please upload a log on of the crash. Depending upon the host you can get it from the docker log command or journalctl command. 

I’ve seen the mini dump but without the log I can’t figure out where it went wrong.

elcosmelco, 2020-02-19:
Here is the log file I got using the docker logs command:

[OZW_Crash.log](https://github.com/OpenZWave/open-zwave/files/4227338/OZW_Crash.log)

Hope it's of use!

Regards,
Elco

Fishwaldo, 2020-02-20:
What device is Node 63?

Fishwaldo, 2020-02-20:
There is something really weird with your Network. Do you have a lot of "dead" nodes etc? 
What type of Controller/Stick are you using?

In this case, Node 62 and Node 63 seem to be very confused:
We ask for some data from Node 63:
```[20200217 22:28:26.218 UTC] [ozw.library] [info]: Info - Node: 63 Sending (Query) message (Callback ID=0xa9, Expected Reply=0x04) - ManufacturerSpecificCmd_Get (Node=63): 0x01, 0x09, 0x00, 0x13, 0x3f, 0x02, 0x72, 0x04, 0x25, 0xa9, 0x22
```
and get a response from Node 62:
```[20200217 22:28:28.370 UTC] [ozw.library] [debug]: Detail - Node: 62   Received: 0x01, 0x0e, 0x00, 0x04, 0x00, 0x3e, 0x08, 0x72, 0x05, 0x01, 0x31, 0x00, 0x02, 0x00, 0x02, 0x84
[20200217 22:28:28.372 UTC] [ozw.library] [debug]: Detail - Node: 0 
[20200217 22:28:28.372 UTC] [ozw.library] [info]: Info - Node: 62   Opening config param file ./config/zipato/RGBBulb.xml
```

I see so many timeouts and issues on your network, if you have a lot of dead/removed nodes, I'm inclined to tell you to factory reset your stick and the nodes and reinclude and try again.

elcosmelco, 2020-02-20:
Thanks for looking at the Log!

The controller is a AEON labs z-stick S2 and I have 2 legit dead nodes.
Node 63 is a 'Greenwave Node6' and node 62 is a 'zipato RGBW bulb'.

With the older version of the library I am using now it largely works fine and I have no dead nodes reported. The 2 actual dead nodes (or rather just not connected right now) which are node 54 and 75 but they aren't marked as such using the older lib. The rest works though, including node 62 and node 63. Besides that using my network is responsive and reliable - I just have a few nodes that aren't recognized properly and some missing CC support but that's about it. 

Regardless of the state of my network - it shouldn't crash the library though should it?

Regards,
Elco