OpenZWave/open-zwave #2126: 700 Series Bridge Firmware - ProcessMsg Updates

Fishwaldo, 2020-02-20:
Update ProcessMsg to handle FUNC_ID_APPLICATION_COMMAND_HANDLER_BRIDGE the same as FUNC_ID_APPLICATION_COMMAND_HANDLER

FUNC_ID_APPLICATION_COMMAND_HANDLER:
```
<?> <?> <status> <srcnode> <length> <CC> <DATA....>
```
FUNC_ID_APPLICATION_COMMAND_HANDLER_BRIDGE:
```
<?> <?> <status> <dstnode> <srcnode> <length> <CC> <DATA...>
```

Fishwaldo, 2020-02-24:
This is Done in Dev Branch Now.

closed 2020-02-24 by Fishwaldo

kpine, 2020-04-29:
Do have an idea on when this will land in master? Is it waiting for 1.7?

gizmocuz, 2020-05-03:
I think the dev branch is closed. Would be great to have bridge support (for ZStick-7 for example)
Maybe this patch could be merged in master ?

m-p-goodwin, 2020-05-11:
I have built the Dev branch to use with domoticz, and the controller comes up and can include nodes, but no nodes appear in domoticz. In the console I get a number of errors about information for NodeID 2:
Error: OpenZWave: Type_ValueChanged, NodeID not found internally!. HomeID: 3915538418, NodeID: 2 (0x02)
I know that the bridge has a slightly different format for the new node messages, that seem to not be picked up by OpenZWave. Thus the dev branch is clearly not ready for merge with master. I'd like to provide more information/debug for this, if desired. Is there a preferred application to use if domoticz is using it unwieldy.

Fishwaldo, 2020-05-11:
Dev branch is the upcoming version 1.7. Its active development and has series 700 support. 

I don't recommend using it for anything other than "what does this look like" as its no where ready for prime time/testing yet.

jcam, 2021-01-04:
Hi folks! I just purchased a 700 series stick by mistake (Zooz now sells two devices with the ZST10 name, one 700 and one 500...). Is 1.7 development dead, or is it happening somewhere else? The dev branch has zero commits in the last 9 months....
Thanks!

ljfh001, 2021-03-15:
Any plans to merge the changes from dev branch to master, so people can start testing? There are more 700 sticks now on the market.

HeFeng1947, 2021-09-17:
Is there any update? At present, 700 series has been the mainstream on the market.

gizmocuz, 2021-09-17:
@HeFeng1947 , I have tested this branch quite some time ago and it was working... 
Not sure why it was closed as the close action mentions another PR/branch, and that mentions this PR
As you might have seen, there is (unfortunately) no active development anymore (c++ wise) to the OZW code
ZWaveJS2MQTT seems to support it which should work with HASS and latest Domoticz

larstobi, 2021-09-17:
@HeFeng1947 
[zwave-js](https://github.com/zwave-js/) supports it, if you are able to use that instead.