OpenZWave/open-zwave #2129: "IsNodeZWavePlus" flag is set to false for Z-Wave Plus controller node

kpine, 2020-02-20:
The IsNodeZWavePlus flag for the primary controller node is always false. Is it possible to set it to true for controllers that are actually z-wave plus?

In Home Assistant, the flag is exported as an attribute for the zwave entity. A common support question is something along the lines of "Why is my controller reporting that it doesn't support Z-Wave Plus, do I need to enable something"?

If it were possible to have the flag set correctly, it would solve a lot of confusion.

Fishwaldo, 2020-02-22:
Technically - OZW is correct, but due to OZW not exposing the Z-Wave+ CommandClass on the Controller Node. 
(ZW+ is both a hardware and a software thing). 

The other challenge is the controller behaves differently from other devices on the network and we can not query 90% of the capabilities via the Serial Port (if you had another controller on the Network, you would be able to, for instance, turn off the disco lights etc).

I understand the confusion. I might look into "fudging" that flag for Controllers in the Future, if i can tell how to differentiate between ZW+ and ZW+v2)

kpine, 2020-02-22:
I noticed the MQTT data is showing some "NodePlusType" information:
```json
{
    "NodeManufacturerName": "Unknown: id=0000",
    "NodeProductName": "Unknown: type=0003, id=0008",
    "NodeBasicString": "Static Controller",
    "NodeBasic": 2,
    "NodeGenericString": "Static Controller",
    "NodeGeneric": 2,
    "NodeSpecificString": "Static PC Controller",
    "NodeSpecific": 1,
    "NodeManufacturerID": "0x0000",
    "NodeProductType": "0x0003",
    "NodeProductID": "0x0008",
    "NodeBaudRate": 100000,
    "NodeVersion": 4,
    "NodeGroups": 0,
    "NodeDeviceTypeString": "Unknown Type (0x0000)",
    "NodeDeviceType": 0,
    "NodeRole": 0,
    "NodeRoleString": "Central Controller",
    "NodePlusType": 0,
    "NodePlusTypeString": "Z-Wave+ node",
}
```
Could `NodePlusType` value be used to "fudge" the flag for a controller? I only have 2 non-zwave-plus devices and they're part of my OZW 1.4 network, but I did notice that in the zwcfg*.xml file those nodes are missing the `nodetype="0"` attribute, while all the plus nodes have it.

Or, maybe the application should not even expose this flag for primary controllers.

Fishwaldo, 2020-02-25:
Actually - that is a bug - I need to find a way to represent “not a plus device or unknown” other than returning 0 (as that’s a valid plus type according to the specs) - that information comes from the zwave plus CC which as I mentioned is not implemented on the OZW for the controller. 

So in short - nope. Can’t fudge it that way.

petergebruers, 2020-03-07:
For a controller I would argue the "Communication Interface Version" is a lot more interesting than Z-Wave Plus type but there might actually be some sort of mapping between that value and Z-Wave Plus

INS12350-17 Serial API Host Appl. Prg. Guide

FUNC_ID_SERIAL_API_GET_INIT_DATA

HOST->ZW: REQ | 0x02
    (Controller) ZW->HOST: RES | 0x02 | ver | capabilities | 29 | nodes[29] | chip_type | chip_version
    (Slave) ZW->HOST: RES | 0x02 | ver | capabilities | 0 | chip_type | chip_version
For example

4.1.1 Version 1-3
The differences between serial API communication interface versions 1 to 3 is not documented.

4.1.2 Version 4
The SDK 4.23-28 are based on the serial API communication interface version 4. This version introduces Serial API commands to better support a host application, especially the Serial API Capabilities Command. The Serial API Capabilities Command determines the Serial API functions that a specific Serial API Z-Wave Module supports.

(...)

This goes up to 8:

4.1.6 Version 8
SDK 6.80.0x changes the Serial API communication interface to version 8 enabling the host software to ensure that this version of the serial API supports Smart Start.

It's decode as m_initVersion:

```cpp
void Driver::HandleSerialAPIGetInitDataResponse(uint8* _data)
(...)
	Log::Write(LogLevel_Info, GetNodeNumber(m_currentMsg), "Received reply to FUNC_ID_SERIAL_API_GET_INIT_DATA:");
	m_initVersion = _data[2];
	m_initCaps = _data[3];
```

What's also exposed is the chip version but that's not decoded yet... I would say >= series 500 means Z-Wave Plus (little white lie).

Fishwaldo, 2020-03-07:
That could be one way. I was also thinking of fudging it based on Baud Rate.

Fishwaldo, 2020-03-23:
@petergebruers any thoughts on what version we should "consider" zw+? I'm inclined to go with your approach.

petergebruers, 2020-03-24:
Hi Justin, I see you've been busy with OZW 1.7, that's awesome! I'll have a more thorough look at this question today. I have to check my notes on the "chip version" and test with ZW500/700 and I think I have a ZW300 as well (the controller used in a HC2).

Fishwaldo, 2020-03-25:
Ok. I've gone with ChipType >= 5 to fake the Controller IsZWavePlus attribute

closed 2020-03-25 by Fishwaldo