OpenZWave/open-zwave #2130: Cannot always include with Security

billiaz, 2020-02-22:
I have multiple Eurotronics Spirit thermostats. 
Some of these are not able to be included with security on and some work like charm.

I always include these near the Z-stick.
This struggle is noticed also on other devices. but this is the most obvious

The thermostat bellow took 4 times to be included with security.

Example of error when this happens:

```
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.052 Detail, Node048,   Received: 0x01, 0x07, 0x00, 0x13, 0x02, 0x00, 0x00, 0x0c, 0xe5
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.052 Detail, Node048,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.053 Info, Node048, Request RTT 132 Average Request RTT 132
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.185 Detail, Node048,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x30, 0x0a, 0x98, 0x80, 0x13, 0x3c, 0x7e, 0x87, 0xe3, 0xd2, 0x20, 0xdf, 0xd1
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.185 Info, Node048, Received SecurityCmd_NonceReport from node 48
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.185 Info, Node048, Encrypted Flag is 1
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.186 Info, Node001, Setting Up Inclusion Network Key for Secure Communications
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.186 Info, Node001, Setting Up Inclusion Network Key for Secure Communications
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.186 Info, Node001, Setting Up Inclusion Network Key for Secure Communications
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.186 Info, Node001, Setting Up Inclusion Network Key for Secure Communications
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.186 Info, Node001, Setting Up Inclusion Network Key for Secure Communications
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.186 Info, Node001, Setting Up Inclusion Network Key for Secure Communications
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.186 Info, Node001, Setting Up Inclusion Network Key for Secure Communications
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.186 Info, Node001, Setting Up Inclusion Network Key for Secure Communications
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.187 Info, Node048, Sending (Command) message (Callback ID=0x2c, Expected Reply=0x04) - SecurityCmd_NetworkKeySet (Node=48): 0x01, 0x19, 0x00, 0x13, 0x30, 0x12, 0x98, 0x06, 0x11, 0x33, 0x13, 0x53, 0x75, 0x06, 0xa7, 0xb8, 0x09, 0x0a, 0xfb, 0x0c, 0x0d, 0x0e, 0x0f, 0x1f, 0x25, 0x2c, 0xa9
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.199 Detail, Node048,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.199 Detail, Node048,   ZW_SEND_DATA delivered to Z-Wave stack
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.672 Detail, Node048,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x30, 0x02, 0x98, 0x40, 0x19
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.672 Info, Node048, Received SecurityCmd_NonceGet from node 48
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.672 Info, NONCES: 0x18, 0x0b, 0xd1, 0x1d, 0xbb, 0x00, 0xd1, 0xb5
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.673 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.673 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.673 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.673 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.673 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.673 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.673 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.673 Info, Node048, Sending (Command) message (Callback ID=0x01, Expected Reply=0x04) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x30, 0x0a, 0x98, 0x80, 0x18, 0x0b, 0xd1, 0x1d, 0xbb, 0x00, 0xd1, 0xb5, 0x05, 0x01, 0xdb:
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.676 Detail, Node048,   Received: 0x01, 0x04, 0x01, 0x13, 0x00, 0xe9
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.677 Error, Node048, ERROR: ZW_SEND_DATA could not be delivered to Z-Wave stack
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.754 Detail, Node048,   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x08, 0xe2
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.755 Detail, Node048,   ZW_SEND_DATA Request with callback ID 0x01 received (expect

```

billiaz, 2020-02-22:
Failed secure inclusion 1: https://pastebin.com/6daR3N7x
Failed secure inclusion 2: https://pastebin.com/iuPkB8Mk

Successful: https://pastebin.com/iXVqKcyn

Fishwaldo, 2020-02-25:
>zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.676 Detail, Node048,   Received: 0x01, 0x04, 0x01, 0x13, 0x00, 0xe9
zwave2mqtt-589b8db8ff-2cfhz z2m 2020-02-22 15:46:50.677 Error, Node048, ERROR: ZW_SEND_DATA could not be delivered to Z-Wave stack

Until the full specifications are opened up by Silabs - we can't know why its failing. So nothing I can do about it right now. 

My only suspect is you have a "busy" network. 

My advise - Unless you absolutely have to include Securely - I would only include most devices as non-secure. The overhead introduced by security is pretty high, and adds a lot of traffic on your network. re-including non-secure should reduce the load on your network. 

its a trade off - Security versus Latency/network load - you have to decide the risk and make a decision. 

closing this - as I can't do anything about it in OZW right now.

closed 2020-02-25 by Fishwaldo