OpenZWave/open-zwave #2132: Steinel 140-2: No motion messages using OZW 1.6

CaCu15, 2020-02-24:
I'm using Openzwave2MQTT (Version 2.2.0) in a docker container. Since the migration to OZW 1.6 I've a problem with my Steinel 140-2 motion sensor: If it detects motion, it does not send any messages. Using Openzwave2MQTT  versions based on OZW 1.4 it used to send messages to the endpoint 113-1-256 containing either  "8" (motion detected) or "0" (cleared).

Other functions of the sensor work as expected, e.g. it does send messages containing illumination info (lux values).

In the container log I see warnings like 
`2020-02-24 16:36:29.970 Warning, Node015, Got a AlarmCmd_Report Message - Greater than Version 2, but size was less than 7?

Attached you will find my OZW_log.txt. It starts with the startup of the zwave2mqtt container and end with the (erroneous) motion detection.

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/4245656/OZW_Log.txt)
`

Fishwaldo, 2020-02-25:
Can you please post a full log with debug enabled. 

But initial suspect - The device is not compatible with the latest command class version used in OZW 1.6. This would be a device bug, but *maybe* we can do a workaround.

CaCu15, 2020-02-27:
Sorry, took me a while to provide this log, because I was not at home. Here comes the full log with zwave logging enabled.  Thanks for your support!

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/4260094/OZW_Log.txt)

Fishwaldo, 2020-03-04:
>2020-02-27 07:23:40.337 Info, Node015, Received CommandClass Version report from node 15: CommandClass=COMMAND_CLASS_SWITCH_BINARY, Version=0
2020-02-27 07:23:40.337 Warning, Node015, CommandClass Version is 0, Removing CommandClass COMMAND_CLASS_SWITCH_BINARY

Buggy Device already... :(

>2020-02-27 07:23:40.822 Info, Node015, Received CommandClass Version report from node 15: CommandClass=COMMAND_CLASS_SENSOR_MULTILEVEL, Version=0
2020-02-27 07:23:40.823 Warning, Node015, CommandClass Version is 0, Removing CommandClass COMMAND_CLASS_SENSOR_MULTILEVEL

More Bugs

>2020-02-27 07:23:41.796 Info, Node015, Received CommandClass Version report from node 15: CommandClass=COMMAND_CLASS_NOTIFICATION, Version=0
2020-02-27 07:23:41.796 Warning, Node015, CommandClass Version is 0, Removing CommandClass COMMAND_CLASS_NOTIFICATION

Boom. Device is Broken Big Time. Its basically saying the Alarm CC (Renamed to Notification) doesn't exist on the device, hence why OZW gets confused as hell.

Sorry, many other issues with this device. I suggest you get something more compliant with the specifications, this is just way to broken.

(it just happened to work ok with OZW 1.4 as the discovery was much more limited in that version, in 1.6 we did a lot of work to bring OZW inline with the Z-Wave Specifications, and this device is just broken too many ways)

Fishwaldo, 2020-03-04:
(or ask if there is a firmware update for the device - maybe they fixed it eventually)

closed 2020-03-04 by Fishwaldo