OpenZWave/open-zwave #2133: ZMNHHDx Mini Dimmer switching flapping

UmbraMalison, 2020-02-26:
ZMNHHDx Mini Dimmer is in OZW.

my experience with it so far has transcended OZW 1.4 to latest 1.6.

That is the switch state flaps. I have 4 of these and they all perform the same.

STR - i'm using roberts zwave2mqtt
1. device is off
2. switch device on (observe switch flaps to on and straight back to off and device turns on)
3. switch device on (observe switch moves to on and device stays on)
4. switch device off (observe switch flaps to off and straight back to on and device turns off)
5. switch device off (observe switch moves to off and device stays off).

i'm using roberts landing zwave2mqtt 2.2.0 which I believe includes OZW 1.6.1038.

I've wondered if perhaps the qubino mini just takes too long to adjust it's state. but had not had any success from changing it's dimming time. 

I've assumed and probbably incorrectly that this is best here in OZW and not in Zwave2mqtt.

i'm keen to provide more info, but please include how as this is a new domain for me.

kind regards,

Fishwaldo, 2020-02-27:
https://github.com/OpenZWave/open-zwave/wiki/OpenZWave-1.6-Release-Notes#switchmultilevel-commandclass

UmbraMalison, 2020-02-27:
Thanks. 
That link talks about using target state. Is this something ozw needs to do, or is it something in the device itself that might not exist, Or maybe configuration that I can do?

Fishwaldo, 2020-03-04:
Its something you have to take into account in the "Application". If you have a "target" variable present in your app, then do a test like 
```
if (level != target) then refreshvalue()
```

Its a quirk of Z-Wave unfortunately, and not all devices did the same thing till the specifications clarified exactly how things are meant to operate. 
(the issue was some devices reported the "instant" status when they were transitioning between states, versus other devices reported the final state the device was transitioning too - Untill recently, the specifications didn't say what a device should report hence vendors doing their own thing)

closed 2020-03-04 by Fishwaldo