OpenZWave/open-zwave #2137: [feat] Identify secured devices

robertsLando, 2020-03-03:
Actually the `COMMAND_CLASS_SECURITY` could be used to to identify secured devices (https://github.com/OpenZWave/open-zwave/blob/master/config/Localization.xml#L844)

Unfortunally almost all devices doens't support it. What about to create a method `IsSecure(nodeId) : Boolean` or even better a `virtual` valueId for those who doesn't support this command_class?

Fishwaldo, 2020-03-04:
The presence of the Security CC does not indicate if encryption is enabled (only that the device possibly supports encryption). 

Please see index 1 of the Security CC (label is "Secured"). There will be multiple for each instance on a device indicating if that endpoint is secured as well.

I'll look at a Node based variable when S2 is introduced.

robertsLando, 2020-03-04:
> The presence of the Security CC does not indicate if encryption is enabled (only that the device possibly supports encryption).

@Fishwaldo I know this, the value index 1 should be true/false based on if the communication is encrypted or not. 

> I'll look at a Node based variable when S2 is introduced.

Keep me updated on this, many users are asking me this feature :) Actually I rely on that command class but as said before it isn't widely supported