OpenZWave/open-zwave #2140: [REQUEST] Support new Aeotec Z-Stick 7 USB-dongle & Z-P17 GPIO-module

Gamester17, 2020-03-05:
Can OpenZWave be made to support the new Aeotec Z-Stick 7 USB-dongle & Z-P17 GPIO-module?

This is a feature request for OpenZWave to support the new Aeotec Z-Stick 7 USB-dongle (a.k.a. Z-Stick Gen7) as well as their new Z-P17 GPIO-module, both supporting Z-Wave 700 Platform series.

Aeotec "Z-Stick 7" USB-dongle (a.k.a. Z-Stick Gen7) is listed with part-number ZWA010 on FCC (with model "ZWA010-A" specifically being the region model for frequencies used North America). 

https://fccid.io/2AOGIZWA010

Aeotec announced several new Z-Wave 700 Platform series devices at CES, but I had missed that it included an announcement of "Z-Stick 7" USB-dongle and "Z-P17" GPIO-model for Raspberry Pi.

https://9to5toys.com/2020/01/10/aeotec-z-wave-ces/

PS: Hopefull that Aeotec as a company are willing to sponsor OpenZWave devs with its hardware?

Fishwaldo, 2020-03-05:
This is already supported in the Dev Branch. (Any 700 series stick will work with the dev branch). 

I don’t believe these are available for purchase yet though, as Aeotec have not released to the public. (They sent me pre-release versions earlier this month). Hence the testing of Dev is minimal right now. 

https://fccid.io/2AOGIZWA010

https://9to5toys.com/2020/01/10/aeotec-z-wave-ces/

closed 2020-03-05 by Fishwaldo

Gamester17, 2020-03-05:
OK that is very cool but why close this request when not in master? Normally an issue is closed when the issue has been resolved in master or?

Yes you are correct that these can't be bought yet but I was hoping that by highlighting this issue that Aeotec as a company might get an incentive to ship these dongles to you OpenZWave developers for testing and more, or?

Fishwaldo, 2020-03-05:
Because the work is complete and tracked at https://github.com/OpenZWave/open-zwave/projects/2 
I only keep issues open that require work. This specific feature does not require any work. (Unless bugs are found)

Fishwaldo, 2020-03-05:
Aeotec and Goap are both very active supporters of OZW. We generally get prelease or beta products to test etc from them automatically without asking. 

Other vendors cooperate if we ask nicely, some just plan act like we don’t exist.

Gamester17, 2020-03-05:
OK that is great! I guess it should be obvious that my follow-up question is when will that hit master? :)

Fishwaldo, 2020-03-05:
How long is a piece of string :)

Sorry - Not sure. I'm dividing my time between the core library, the new HA integration, and the new Gui. 

You can track progress here - https://github.com/OpenZWave/open-zwave/milestone/3
(its going to be a major release)