OpenZWave/open-zwave #2152: FGWPF-102 ZW5 missing group 1 after update domoticz to 4.11776

GravityRZ, 2020-03-10:
maybe this problem  appeared earlier but i discovered it when i deleted the cache file
somehow the cache is rebuild without recognizing the correct groups

it fails to add 1 or 1.1 to group 1 (device 17,18,19)
whatever you do(refresh node, disable/enable zwave) i can not get it to add 1 or 1.1 to group 1 anymore.

devices seem to work but if 1 is not in group 1(this can cause problems)
any idea how this is related.
what can i do to get the groups back(did refresh node in domoticz and in zwave panel)

[image]

after restoring the old cache file everything looks normal

[image]

i inspected the cache file and it is missing this entry for all 3 FGWPF devices
```
> <CommandClass id="96" name="COMMAND_CLASS_MULTI_INSTANCE/CHANNEL">
>                 <Compatibility />
>                 <State />
>             </CommandClass>

domoticz used to crash when you delete the cache file and let it rebuild.
this was caused by the way devices got interviewed.
i noticed it is not crashing anymore so i suspect we changed the way devices get interviewed(solving the crash but causing this)

units seem to work without association to group 1 but i do not think it is smart to keep this setup.
FGWPF is multi instance/channel capable so i do not know what caused this because the device files seem unchanged.

```

Fishwaldo, 2020-03-13:
Logs?

GravityRZ, 2020-03-13:
it is device 17,18,19

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/4328347/OZW_Log.txt)

Fishwaldo, 2020-03-13:
Another Buggy Implementation:
```
2020-03-13 08:26:35.635 Info, Node017, UPDATE_STATE_NODE_INFO_RECEIVED from node 17
2020-03-13 08:26:35.635 Info, Node017,   Optional CommandClasses for node 17:
2020-03-13 08:26:35.635 Info, Node017,     COMMAND_CLASS_ZWAVEPLUS_INFO
2020-03-13 08:26:35.635 Info, Node017,     COMMAND_CLASS_APPLICATION_STATUS
2020-03-13 08:26:35.635 Info, Node017,     COMMAND_CLASS_ASSOCIATION
2020-03-13 08:26:35.635 Info, Node017,   CommandClass 0x59 - NOT REQUIRED
2020-03-13 08:26:35.635 Info, Node017,     COMMAND_CLASS_CONFIGURATION
2020-03-13 08:26:35.635 Info, Node017,     COMMAND_CLASS_CRC_16_ENCAP
2020-03-13 08:26:35.635 Info, Node017,     COMMAND_CLASS_DEVICE_RESET_LOCALLY
2020-03-13 08:26:35.635 Info, Node017,   CommandClass 0x7a - NOT REQUIRED
2020-03-13 08:26:35.635 Info, Node017,     COMMAND_CLASS_MANUFACTURER_SPECIFIC (Existing)
2020-03-13 08:26:35.635 Info, Node017,     COMMAND_CLASS_METER
2020-03-13 08:26:35.635 Info, Node017,     COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION
2020-03-13 08:26:35.636 Info, Node017,     COMMAND_CLASS_NOTIFICATION
2020-03-13 08:26:35.636 Info, Node017,     COMMAND_CLASS_POWERLEVEL
2020-03-13 08:26:35.636 Info, Node017,     COMMAND_CLASS_SECURITY
2020-03-13 08:26:35.636 Info, Node017,     COMMAND_CLASS_SENSOR_MULTILEVEL
2020-03-13 08:26:35.636 Info, Node017,     COMMAND_CLASS_SWITCH_BINARY (Existing)
2020-03-13 08:26:35.636 Info, Node017,     COMMAND_CLASS_VERSION
```
Supports MultiChannelAssociation, but not MultiChannel... sigh.

Regardless:
```
2020-03-13 08:27:14.176 Info, Node017, Received MULTI_CHANNEL_ASSOCIATION_REPORT from node 17, group 1
2020-03-13 08:27:14.176 Info, Node017,   The group contains:
2020-03-13 08:27:14.176 Info, Node017,     Node 1 End Point 1
2020-03-13 08:27:14.176 Info, Node017, Get MultiChannelAssociation for group 2 of node 17
```
```
2020-03-13 08:27:14.575 Info, Node017, Received MULTI_CHANNEL_ASSOCIATION_REPORT from node 17, group 2
2020-03-13 08:27:14.575 Info, Node017,   The group contains:
2020-03-13 08:27:14.576 Info, Node017,     Node 1 End Point 1
2020-03-13 08:27:14.576 Info, Node017, Get MultiChannelAssociation for group 3 of node 17
```
```
2020-03-13 08:27:14.976 Info, Node017, Received MULTI_CHANNEL_ASSOCIATION_REPORT from node 17, group 3
2020-03-13 08:27:14.976 Info, Node017,   The group contains:
2020-03-13 08:27:14.976 Info, Node017,     Node 1 End Point 1
2020-03-13 08:27:14.976 Info, Node017, Querying associations for node 17 is complete.
```

Then due to the bug above:
```
2020-03-13 08:27:14.976 Info, Node017, Adding the controller to group 1 (Lifeline) of node 17
2020-03-13 08:27:14.976 Warning, Node017, MultiChannelAssociation is Present, but MultiChannel CC is not. Trying Plain Association...
2020-03-13 08:27:14.976 Info, Node017, Association::Set - Adding node 1 to group 1 of node 17
```

But the Node doesn't take it:
```
2020-03-13 08:27:15.540 Info, Node017, Received Association report from node 17, group 1, containing 0 associations
```

(that work around is for the Qubino devices that exhibit a similar bug, and it works for them....)

Would be interesting to see some unsolicited reports from this device, if they are indeed encapsulated in a MultiChanneEncap message or not....

Fishwaldo, 2020-03-13:
Try this:
https://github.com/OpenZWave/open-zwave/blob/d8fddb47b3ac12f495101691a42ff38f3b61f296/cpp/src/Group.cpp#L252

Add the following lines after that log message:
```
					cc->Set(m_groupIdx, _nodeId, 0);
					cc->QueryGroup(m_groupIdx, 0);
					return;
```

GravityRZ, 2020-03-13:
@Fishwaldo , do i need to try this. if so, how?

Fishwaldo, 2020-03-13:
Yes - I dont own these devices so someone needs to debug.

You need to make the changes and recompile.

GravityRZ, 2020-03-13:
ok, have asked Jumbotroll to do this for me and will get back after i have tested.

GravityRZ, 2020-03-13:
Jumbotroll was so kind to create a 782 build with the modifications you suggested.
it looks like it is working. 1.1 in group 1 again. also deleting 1.1 from group 1 and adding it again works and show 1.1 in group 1 again.

here is the log in case you want to inspect if things really stuck
do i need to check anything else in case this might have broken something else?

[OZW_Log-afterfix.txt](https://github.com/OpenZWave/open-zwave/files/4330668/OZW_Log-afterfix.txt)

Fishwaldo, 2020-03-13:
I need to test that with Quibino Devices, which the original fix broke your fibaro devices. both of which, IMHO are not compliant with the specifications. *sigh*

GravityRZ, 2020-03-13:
device 11 is a qubino zmnhadx
it seems to work in my situation but standard association so 1 in group 1(fix in the config file)

GravityRZ, 2020-04-16:
any news on this issue. i do not see any changes regarding this. at thr moment i keep a couple copies of my good working cache file

GravityRZ, 2020-06-20:
@Fishwaldo 

i installed 2020.2.12178 but the issue is still there with the FGWPF modules.
eveything seems to work but those modules not being present in group 1 (lifeline) is not good.

for now i still keep on using a very old ozwcache file which does work but there will come a time when this file needs to be refreshed and then i will have a problem.

any news on when we will fix this (or undo the other fix)

since the FGWPF gen5 is a widely used module i am suprized nobody else is complaining about this.
maybee everybody is upgrading and never tested a fresch ozw cache file