OpenZWave/open-zwave #2154: Corruption of ozwcache

kevinkahn, 2020-03-11:
I have now seen a couple of instances at apparently random times (i.e., I can't relate occurrence to any event) of the ozwcache becoming corrupted.  This relates to issue 2139 (closed) here and to  https://community.home-assistant.io/t/the-future-of-z-wave-in-ha-qt-openzwave/148541/134 discussion.  The characteristic is that instead of my Homeseer devices returning "Scene 1", "Scene 2", etc. they return things like "Top Button Scene".  These latter terms seem to originate (from the previous discussion) from the Inovelli switch in my system.  Clearing the cache and restarting seems to repair things.  I have now switched to using the scene_id instead of the label so hopefully this will fix my personal problem but there does seem to be something periodically happening with the cache.   This is an example of what monitoring the event zwave_mqtt.scene_activated within HA showed:
{
    "event_type": "zwave_mqtt.scene_activated",
    "data": {
        "node_id": 9,
        "scene_id": 1,
        "scene_label": "Bottom Button Scene",
        "scene_value_id": 4,
        "scene_value_label": "Pressed 2 Times"
    },
    "origin": "LOCAL",
    "time_fired": "2020-03-11T16:01:20.266437+00:00",
    "context": {
        "id": "eca47b588e9f44e39a8152084cf9b994",
        "parent_id": null,
        "user_id": null
    }
}

After restarting OZW and HA and deleting the cache while they were down the same scene activation yielded:
{
    "event_type": "zwave_mqtt.scene_activated",
    "data": {
        "node_id": 9,
        "scene_id": 1,
        "scene_label": "Scene 1",
        "scene_value_id": 4,
        "scene_value_label": "Pressed 2 Times"
    },
    "origin": "LOCAL",
    "time_fired": "2020-03-11T16:14:04.589816+00:00",
    "context": {
        "id": "53a788d4f0344c57972c3247d0235c10",
        "parent_id": null,
        "user_id": null
    }
}

An example of what the corrupted cache looked like the first time I saw this is referenced in the thread referenced above.

Of perhaps note: the one Inovelli switch does not seem to generate any scene activation messages.  I don't use it for scenes so perhaps there is some switch setting I haven't set up for it but I would generally expect that it would cause reports in its default state (it handles up to 5 taps as I remember).

RicardP, 2020-03-12:
I believe HA is still on the obsolete OZW 1.4 ?  If so I would say there is a lot of flaws that is solved or much improved in 1.6, also related to Central Scene as in your case. Now development is even focused on next level 1.7  :)

kevinkahn, 2020-03-12:
Note; this is happening with the new HA integration via mqtt which is on
OZW 1.6 currently
________________________
Kevin Kahn
kevin@coral-zone.com
cell 503-701-8781

Fishwaldo, 2020-03-13:
Should be fixed in latest commit. Will update the ozwdaemon later. Please reopen if you see this issue after the new image is pushed out.

closed 2020-03-13 by Fishwaldo