OpenZWave/open-zwave #2158: Hank HKZW-SCN04 unhandled commands

uSpike, 2020-03-17:
Hello, I have some of these devices: https://www.thesmartesthouse.com/products/hank-z-wave-plus-four-button-scene-controller-hkzw-scn04

Config here: https://github.com/OpenZWave/open-zwave/blob/master/config/hank/scenecontroller4.xml

When I press buttons on the remote, I see messages such as:
```
[20200316 4:03:25.010 UTC] [ozw.library] [info]: Info - Node: 14 ApplicationCommandHandler - Unhandled Command Class 0x5b
[20200316 4:03:25.463 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x0e, 0x03, 0x80, 0x03, 0x64, 0x18
[20200316 4:03:25.463 UTC] [ozw.library] [info]: Info - Node: 14 ApplicationCommandHandler - Unhandled Command Class 0x80
[20200316 4:05:30.355 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x0e, 0x05, 0x5b, 0x03, 0x1d, 0x00, 0x03, 0xbd
[20200316 4:05:30.355 UTC] [ozw.library] [info]: Info - Node: 14 ApplicationCommandHandler - Unhandled Command Class 0x5b
[20200316 4:05:30.848 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x0e, 0x03, 0x80, 0x03, 0x64, 0x18
[20200316 4:05:30.848 UTC] [ozw.library] [info]: Info - Node: 14 ApplicationCommandHandler - Unhandled Command Class 0x80
```
and events are not forwarded to MQTT.  Not sure where the issue is, any ideas?  Thank you.

Fishwaldo, 2020-03-17:
Have you ensured that the device was woken up after restarting ozwdaemon (or the Docker image). Battery Powered devices wont be fully operational till you get to "QueryStage_Complete"

uSpike, 2020-03-17:
Hm, interesting.  It seems that the device has passed `QueryStage_Complete`.  Here's some cherry-picked logs:
```
[20200308 18:28:26.542 UTC] [ozw.library] [debug]: Detail - Node: 14 Queuing (Query) Get Node Protocol Info (Node=14): 0x01, 0x04, 0x00, 0x41, 0x0e, 0xb4
[20200308 18:28:26.542 UTC] [ozw.library] [debug]: Detail - Node: 14 Queuing (Query) Query Stage Complete (ProtocolInfo)
[20200308 18:28:26.542 UTC] [ozw.library] [info]: Info - Node: 14 Initializing Node. New Node: false (false)
...
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14 Received reply to FUNC_ID_ZW_GET_NODE_PROTOCOL_INFO
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14   Protocol Info for Node 14:
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14     Listening     = false
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14     Frequent      = false
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14     Beaming       = true
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14     Routing       = true
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14     Max Baud Rate = 100000
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14     Version       = 4
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14     Security      = false
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14   Basic device class    (0x04) - Routing Slave
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14   Generic device Class  (0x18) - Wall Controller
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14   Specific device class (0x01) - Basic Wall Controller
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14   Mandatory Command Classes for Node 14:
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14     COMMAND_CLASS_BASIC
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14     COMMAND_CLASS_MANUFACTURER_SPECIFIC
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14     COMMAND_CLASS_WAKE_UP
[20200308 18:28:42.830 UTC] [ozw.library] [info]: Info - Node: 14   Mandatory Command Classes controlled by Node 14:
[20200308 18:28:42.831 UTC] [ozw.library] [info]: Info - Node: 14     None
...
[20200308 18:28:42.831 UTC] [ozw.library] [debug]: Detail - Node: 14 Queuing (Query) Query Stage Complete (Probe)
[20200308 18:28:42.831 UTC] [ozw.library] [info]: Info - Node: 14 Sending (NoOp) message (Callback ID=0x2c, Expected Reply=0x13) - NoOperation_Set (Node=14): 0x01, 0x09, 0x00, 0x13, 0x0e, 0x02, 0x00, 0x00, 0x25, 0x2c, 0xe0
[20200308 18:28:42.831 UTC] [ozw.library] [info]: Info - Node: 14 Encrypted Flag is 0
[20200308 18:28:42.843 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
[20200308 18:28:42.843 UTC] [ozw.library] [debug]: Detail - Node: 14   ZW_SEND_DATA delivered to Z-Wave stack
[20200308 18:28:52.831 UTC] [ozw.library] [critical]: Error - Node: 14 ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "NoOperation_Set (Node=14): 0x01, 0x09, 0x00, 0x13, 0x0e, 0x02, 0x00, 0x00, 0x25, 0x2c, 0xe0"
[20200308 18:28:52.832 UTC] [ozw.library] [debug]: Detail - Node: 14 Removing current message
[20200308 18:28:52.832 UTC] [ozw.library] [debug]: Detail - Node: 14 Notification: Notification - TimeOut
[20200308 18:28:53.137 UTC] [ozw.library] [info]: Info - Node: 14 Skipping Cache Save for Node 14 as its not past QueryStage_CacheLoad
[20200308 18:28:57.771 UTC] [ozw.library] [info]: Info - Node: 14 Skipping Cache Save for Node 14 as its not past QueryStage_CacheLoad
[20200308 18:29:01.677 UTC] [ozw.library] [info]: Info - Node: 14 Skipping Cache Save for Node 14 as its not past QueryStage_CacheLoad
[20200308 18:29:02.380 UTC] [ozw.library] [info]: Info - Node: 14 Skipping Cache Save for Node 14 as its not past QueryStage_CacheLoad
[20200308 18:29:02.452 UTC] [ozw.library] [debug]: Detail - Node: 14 Query Stage Complete (Probe)
```
Do I need to wake up all of my devices each time I restart the ozwdaemon?

uSpike, 2020-03-22:
I pressed the buttons on the controller while starting `qt-openzwave` to make sure it was awake while querying was occurring.

```
[20200322 19:04:45.386 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x0e, 0x05, 0x5b, 0x03, 0x33, 0x00, 0x03, 0x93
[20200322 19:04:45.386 UTC] [ozw.library] [debug]: Detail - Node: 0 
[20200322 19:04:45.386 UTC] [ozw.library] [info]: Info - Node: 14 Received Central Scene set from node 14: scene id=3 with key Attribute 0. Sending event notification.
[20200322 19:04:45.386 UTC] [ozw.library] [warning]: Warning - Node: 14 No ValueID created for Scene 3
[20200322 19:04:45.386 UTC] [ozw.library] [warning]: Warning - Node: 14 CommandClass COMMAND_CLASS_CENTRAL_SCENE HandlerMsg Returned False
[20200322 19:04:45.879 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x0e, 0x03, 0x80, 0x03, 0x5a, 0x26
[20200322 19:04:45.879 UTC] [ozw.library] [debug]: Detail - Node: 0 
[20200322 19:04:45.879 UTC] [ozw.library] [info]: Info - Node: 14 Received Battery report from node 14: level=90
```

Fishwaldo, 2020-03-23:
Pressing a button does not wake up a device. You need to consult the device manual to figure out how to wake it up so that OZW can complete the interview. 

>Do I need to wake up all of my devices each time I restart the ozwdaemon?

Each sleeping device will eventually (in 90% of cases) wake up. You can consult the Value "Wake Up Interval" - This is how often the device will wake up. If you cant wait that long, then you can manually wake it up.

Be Warning - Reducing the Wake Up Interval will affect battery life.

uSpike, 2020-03-24:
It seems that my issues is actually a duplicate of https://github.com/OpenZWave/qt-openzwave/issues/45
I pulled a fresh image of `qt-openzwave` that included https://github.com/OpenZWave/open-zwave/commit/f3920e88d731577dbe058e64ebb91aa8e386bd8c and my issue seems to have gone away.

closed 2020-03-24 by uSpike

robotictacos, 2021-01-28:
I have the same issue (same logfile entries) with the same device.  I am running the latest docker stable (ozwdaemon build 182) so I assume that it includes the fix mentioned from nearly a year ago.

robotictacos, 2021-01-28:
I created a device-specific workaround.  I don't think it solves the root, but it got my device working.  A bit of background; I got two of these devices -> one for my side of the bed and one for my wife's side, to be able to operate the ceiling fan and lights in our bedroom.  The one for my side worked in OZW straightaway (home assistant required a little more finagling, but that's outside the scope of the discussion).  What I found was that once the unit on my side of the bed was added to the network, it created this entry in ozwcache_0xxxx.xml:

`<CommandClass id="91" name="COMMAND_CLASS_CENTRAL_SCENE">
<Compatibility />
<State>
    <CCVersion>2</CCVersion>
    <InNif>true</InNif>
    <SceneCount>4</SceneCount>
    <StaticRequests>2</StaticRequests>
</State>
<Instance index="1" />
<Value type="list" genre="user" instance="1" index="1" label="Scene 1" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="4">
    <Help></Help>
    <Item label="Inactive" value="0" />
    <Item label="Pressed 1 Time" value="1" />
    <Item label="Key Released" value="2" />
    <Item label="Key Held down" value="3" />
</Value>
<Value type="list" genre="user" instance="1" index="2" label="Scene 2" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="4">
    <Help></Help>
    <Item label="Inactive" value="0" />
    <Item label="Pressed 1 Time" value="1" />
    <Item label="Key Released" value="2" />
    <Item label="Key Held down" value="3" />
</Value>
<Value type="list" genre="user" instance="1" index="3" label="Scene 3" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="4">
    <Help></Help>
    <Item label="Inactive" value="0" />
    <Item label="Pressed 1 Time" value="1" />
    <Item label="Key Released" value="2" />
    <Item label="Key Held down" value="3" />
</Value>
<Value type="list" genre="user" instance="1" index="4" label="Scene 4" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="4">
    <Help></Help>
    <Item label="Inactive" value="0" />
    <Item label="Pressed 1 Time" value="1" />
    <Item label="Key Released" value="2" />
    <Item label="Key Held down" value="3" />
</Value>
<Value type="int" genre="user" instance="1" index="256" label="Scene Count" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-2147483648" max="2147483647" value="4">
    <Help></Help>
</Value>
<Value type="int" genre="config" instance="1" index="257" label="Scene Reset Timeout" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="-2147483648" max="2147483647" value="1000">
    <Help></Help>
</Value>
</CommandClass>`

Note that this is quite different from the wiki posted by HA [here](https://www.home-assistant.io/docs/z-wave/device-specific/#hank-one-key-scene-controller-hkzn-scn01) 

I don't actually know how it got the command class 91 information, since it's completely missing from the scenecontroller4.xml file in the hank folder that ships with ozw.  I can only assume the device itself sent it when it was added to the network?  Anyway, it was there for the working node, and not there for the non-working node.  I copied the snippet above and pasted into scenecontroller4.xml (the zwcfg file), deleted the non-working node and added it back.  Now, it populates the command class 91 into the ozwcache file and everything works as expected.  Like I said though, I'm not sure if this is a fix per se, or only a workaround, but thought I would share.