OpenZWave/open-zwave #2165: Value change does not trigger mqtt message

erikced, 2020-03-23:
Triggering a multi-level set to some value on my Qubino Flush Dimmers does not cause a `valueChanged` message to be sent, for my (single) Fibaro Dimmer 2 (that, for some reason 🤷‍♂  is included with S0 security unlike the Qubinos) it does.
Looking at the logs it appears that for both devices ozw issues multilevel set followed by multilevel get calls, but the result of the qubino's multilevel report is does not cause a value change whereas it does for the fibaro.

Qubino "level" value:
```json
{
  "Label": "Level",
  "Value": 0,
  "Units": "",
  "Min": 0,
  "Max": 255,
  "Type": "Byte",
  "Instance": 1,
  "CommandClass": "COMMAND_CLASS_SWITCH_MULTILEVEL",
  "Index": 0,
  "Node": 2,
  "Genre": "User",
  "Help": "The Current Level of the Device",
  "ValueIDKey": 38371345,
  "ReadOnly": false,
  "WriteOnly": false,
  "ValueSet": false,
  "ValuePolled": false,
  "ChangeVerified": false,
  "Event": "valueAdded",
  "TimeStamp": 1583269799
}
```

Fibaro "level" value:
```json
{
  "Label": "Instance 1: Level",
  "Value": 75,
  "Units": "",
  "Min": 0,
  "Max": 255,
  "Type": "Byte",
  "Instance": 1,
  "CommandClass": "COMMAND_CLASS_SWITCH_MULTILEVEL",
  "Index": 0,
  "Node": 54,
  "Genre": "User",
  "Help": "The Current Level of the Device",
  "ValueIDKey": 910786577,
  "ReadOnly": false,
  "WriteOnly": false,
  "ValueSet": false,
  "ValuePolled": false,
  "ChangeVerified": false,
  "Event": "valueChanged",
  "TimeStamp": 1583362100
}
```

Logs, `ozw.logs.gz` is _not_ complete, it contains network startup but not the actual failing calls, they are located in the other two log files.
[ozw.log.gz](https://github.com/OpenZWave/qt-openzwave/files/4290034/ozw.log.gz)
[qubino.log.gz](https://github.com/OpenZWave/qt-openzwave/files/4290036/qubino.log.gz)
[fibaro.log.gz](https://github.com/OpenZWave/qt-openzwave/files/4290033/fibaro.log.gz)

Fishwaldo, 2020-03-23:
Please test Release > 73 when it becomes available on Docker

erikced, 2020-03-23:
I didn't have the patience to wait for builds > 73 so I built an ozwdaemon image containing openzwave `55770df`, the value appears to get updated when it should. Thanks for taking the time to look into this.

closed 2020-03-27 by erikced