OpenZWave/open-zwave #2166: 26933-motion-dimmer.xml Association Group 2 and 3 have same label

Botched1, 2020-03-24:
open-zwave/config/ge/26933-motion-dimmer.xml

Association groups 2 and 3 have the same label. This causes issues in zwave2mqtt, or any interface that uses the label for item enumeration, as you can't tell which "Basic Set"  association group is number 2 and which is number 3.

Request: Consider adding a prefix or suffix to the label to make it clear which is which.

Fishwaldo, 2020-03-24:
This is a application related issue. Nothing todo with OZW. We provide API’s to iterate the group numbers and labels. It’s upto the application to determine the best way to present it to the user.

robertsLando, 2020-03-24:
@Fishwaldo I think he means, there is an error on configuration file, there are 2 associatons group with the same name

Fishwaldo, 2020-03-24:
It’s not strictly a error. 

Regardless, PR is welcome. I don’t have own this device, can’t test it and with over 900 devices in the database, I rely on the community to help maintain and update config files.

closed 2020-03-24 by Fishwaldo

Botched1, 2020-03-24:
> @Fishwaldo I think he means, there is an error on configuration file, there are 2 associatons group with the same name

I'm not really sure it is a bug, per se. This is from the zwave cert report. Notice it lists 2xbasic set association groups:
[image]

robertsLando, 2020-03-25:
@Botched1 Find the config file here and submit a PR to fix that label please