OpenZWave/open-zwave #2173: Receiving value with wrong measurement unit. Maybe a Bug ?

candrea77, 2020-03-31:
Dear,

SYSTEM INFO :
VM on QNAP 862
OS : Ubuntu 18.04.4 LTS
ZWAVE : open-zwave branch master commit 544347
HOME AUTOMATION : domoticz branch master release 2020.1

PROBLEM : 
Some times I'm receiving error on my home ha system : "Error: OpenZWave: Value_Changed: Tried adding value, not succeeded!. Node: 60 (0x3c), CommandClass: SENSOR MULTILEVEL, Label: Distance, Instance: 1, Index: 20"

A quick serach on the OZW_Log highlight this strange behaviour : 
 - Take a look at 07:49:04.951 you can see Nodo060 send the power consumption as distance (meter).
```
2020-03-31 07:49:01.193 Info, Node072, Received Wakeup Notification from node 72
2020-03-31 07:49:01.193 Info, Node072,   Node 72 has been marked as awake
2020-03-31 07:49:01.193 Detail, Node072, Queuing (WakeUp) WakeUpCmd_NoMoreInformation (Node=72): 0x01, 0x09, 0x00, 0x13, 0x48, 0x02, 0x84, 0x08, 0x25, 0x0f, 0x09
2020-03-31 07:49:01.193 Info, Saving Cache
2020-03-31 07:49:01.194 Info, Node001, Cache Save for Node 1 as its QueryStage_CacheLoad
2020-03-31 07:49:01.196 Info, Node009, Cache Save for Node 9 as its QueryStage_CacheLoad
2020-03-31 07:49:01.197 Info, Node010, Cache Save for Node 10 as its QueryStage_CacheLoad
2020-03-31 07:49:01.198 Info, Node012, Cache Save for Node 12 as its QueryStage_CacheLoad
2020-03-31 07:49:01.199 Info, Node018, Cache Save for Node 18 as its QueryStage_CacheLoad
2020-03-31 07:49:01.199 Info, Node019, Cache Save for Node 19 as its QueryStage_CacheLoad
2020-03-31 07:49:01.199 Info, Node020, Cache Save for Node 20 as its QueryStage_CacheLoad
2020-03-31 07:49:01.200 Info, Node021, Cache Save for Node 21 as its QueryStage_CacheLoad
2020-03-31 07:49:01.200 Info, Node023, Cache Save for Node 23 as its QueryStage_CacheLoad
2020-03-31 07:49:01.201 Info, Node024, Cache Save for Node 24 as its QueryStage_CacheLoad
2020-03-31 07:49:01.201 Info, Node026, Cache Save for Node 26 as its QueryStage_CacheLoad
2020-03-31 07:49:01.202 Info, Node027, Cache Save for Node 27 as its QueryStage_CacheLoad
2020-03-31 07:49:01.202 Info, Node028, Cache Save for Node 28 as its QueryStage_CacheLoad
2020-03-31 07:49:01.203 Info, Node032, Cache Save for Node 32 as its QueryStage_CacheLoad
2020-03-31 07:49:01.203 Info, Node033, Cache Save for Node 33 as its QueryStage_CacheLoad
2020-03-31 07:49:01.203 Info, Node035, Cache Save for Node 35 as its QueryStage_CacheLoad
2020-03-31 07:49:01.204 Info, Node039, Cache Save for Node 39 as its QueryStage_CacheLoad
2020-03-31 07:49:01.205 Info, Node040, Cache Save for Node 40 as its QueryStage_CacheLoad
2020-03-31 07:49:01.206 Info, Node041, Cache Save for Node 41 as its QueryStage_CacheLoad
2020-03-31 07:49:01.206 Info, Node042, Cache Save for Node 42 as its QueryStage_CacheLoad
2020-03-31 07:49:01.206 Info, Node043, Cache Save for Node 43 as its QueryStage_CacheLoad
2020-03-31 07:49:01.207 Info, Node044, Cache Save for Node 44 as its QueryStage_CacheLoad
2020-03-31 07:49:01.207 Info, Node048, Cache Save for Node 48 as its QueryStage_CacheLoad
2020-03-31 07:49:01.208 Info, Node052, Cache Save for Node 52 as its QueryStage_CacheLoad
2020-03-31 07:49:01.208 Info, Node054, Cache Save for Node 54 as its QueryStage_CacheLoad
2020-03-31 07:49:01.209 Info, Node055, Cache Save for Node 55 as its QueryStage_CacheLoad
2020-03-31 07:49:01.209 Info, Node056, Cache Save for Node 56 as its QueryStage_CacheLoad
2020-03-31 07:49:01.209 Info, Node057, Cache Save for Node 57 as its QueryStage_CacheLoad
2020-03-31 07:49:01.210 Info, Node059, Cache Save for Node 59 as its QueryStage_CacheLoad
2020-03-31 07:49:01.211 Info, Node060, Cache Save for Node 60 as its QueryStage_CacheLoad
2020-03-31 07:49:01.211 Info, Node062, Cache Save for Node 62 as its QueryStage_CacheLoad
2020-03-31 07:49:01.211 Info, Node063, Cache Save for Node 63 as its QueryStage_CacheLoad
2020-03-31 07:49:01.212 Info, Node064, Cache Save for Node 64 as its QueryStage_CacheLoad
2020-03-31 07:49:01.212 Info, Node065, Cache Save for Node 65 as its QueryStage_CacheLoad
2020-03-31 07:49:01.212 Info, Node066, Cache Save for Node 66 as its QueryStage_CacheLoad
2020-03-31 07:49:01.214 Info, Node067, Cache Save for Node 67 as its QueryStage_CacheLoad
2020-03-31 07:49:01.214 Info, Node072, Cache Save for Node 72 as its QueryStage_CacheLoad
2020-03-31 07:49:01.215 Info, Node073, Cache Save for Node 73 as its QueryStage_CacheLoad
2020-03-31 07:49:01.215 Info, Node076, Cache Save for Node 76 as its QueryStage_CacheLoad
2020-03-31 07:49:01.215 Info, Node077, Cache Save for Node 77 as its QueryStage_CacheLoad
2020-03-31 07:49:01.216 Info, Node081, Cache Save for Node 81 as its QueryStage_CacheLoad
2020-03-31 07:49:01.216 Info, Node083, Cache Save for Node 83 as its QueryStage_CacheLoad
2020-03-31 07:49:01.216 Info, Node084, Cache Save for Node 84 as its QueryStage_CacheLoad
2020-03-31 07:49:01.217 Info, Node086, Cache Save for Node 86 as its QueryStage_CacheLoad
2020-03-31 07:49:01.217 Info, Node088, Cache Save for Node 88 as its QueryStage_CacheLoad
2020-03-31 07:49:01.217 Info, Node091, Cache Save for Node 91 as its QueryStage_CacheLoad
2020-03-31 07:49:01.218 Info, Node094, Cache Save for Node 94 as its QueryStage_CacheLoad
2020-03-31 07:49:01.218 Info, Node095, Cache Save for Node 95 as its QueryStage_CacheLoad
2020-03-31 07:49:01.218 Info, Node096, Cache Save for Node 96 as its QueryStage_CacheLoad
2020-03-31 07:49:01.219 Info, Node098, Cache Save for Node 98 as its QueryStage_CacheLoad
2020-03-31 07:49:01.219 Info, Node100, Cache Save for Node 100 as its QueryStage_CacheLoad
2020-03-31 07:49:01.220 Info, Node102, Cache Save for Node 102 as its QueryStage_CacheLoad
[... 89 lines cut ...]
2020-03-31 07:49:05.755 Info, Node122, Received Notification report (>v1): Type: Home Security (7) Event: Clear (0) Status: true, Param Length: 1
2020-03-31 07:49:05.755 Detail, Node122, Refreshed Value: old value=0, new value=8, type=byte
2020-03-31 07:49:05.755 Detail, Node122, Changes to this value are not verified
2020-03-31 07:49:05.755 Detail, Node122, Refreshed Value: old value=3, new value=0, type=list
2020-03-31 07:49:05.755 Detail, Node122, Changes to this value are not verified
2020-03-31 07:49:05.755 Detail, Node122, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 256
2020-03-31 07:49:05.755 Detail, Node122, Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 7
2020-03-31 07:49:05.858 Detail, Node122,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x7a, 0x03, 0x30, 0x03, 0x00, 0xb8
2020-03-31 07:49:05.858 Detail,
2020-03-31 07:49:05.858 Info, Node122, Received SensorBinary report: State=Off
2020-03-31 07:49:05.858 Detail, Node122, Refreshed Value: old value=true, new value=false, type=bool
2020-03-31 07:49:05.858 Detail, Node122, Changes to this value are not verified
2020-03-31 07:49:05.858 Detail, Node122, Notification: ValueChanged CC: COMMAND_CLASS_SENSOR_BINARY Instance: 1 Index: 0
2020-03-31 07:49:07.928 Detail, Node060,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x3c, 0x06, 0x31, 0x05, 0x04, 0x22, 0x00, 0x8d, 0x52
2020-03-31 07:49:07.928 Detail,
2020-03-31 07:49:07.928 Info, Node060, Received SensorMultiLevel report from node 60, instance 1, Power: value=14.1Watt
2020-03-31 07:49:07.928 Detail, Node060, Refreshed Value: old value=331.1, new value=14.1, type=decimal
2020-03-31 07:49:07.928 Detail, Node060, Changes to this value are not verified
2020-03-31 07:49:07.928 Detail, Node060, Notification: ValueChanged CC: COMMAND_CLASS_SENSOR_MULTILEVEL Instance: 1 Index: 4
2020-03-31 07:49:15.725 Detail, Node095,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x5f, 0x03, 0x80, 0x03, 0x2d, 0x00
2020-03-31 07:49:15.725 Detail,
```

This behaviour can happen mulpiple times a day (on version 1.4 I've never had something like this).
Is this a bug of open-zwave or something else ?

RicardP, 2020-04-01:
Good catch, I had many such false but no logs. I beleiv it is caused by corrupt-demodulated radio datapackets. I am just guessing but the CRC level is not good enough in ZWave... so if a user have not very good radio link quality, (bad c/n) between controller and nodes, it will increase the chance of ghost CCs added as well false values.

candrea77, 2020-04-01:
I dubt might be a proble of good link quality (may be, but not sure) : I've hundred of nodes creating a very wide mesh network.

Node060 is an active nodes with repeating functionality : in case of link quality problem I think I had to have problems most of times on that node (becouse is supposed to be a node with low quality index)

In my case problems is random two or three times a week on different nodes every times.

How can I check link quality ?

RicardP, 2020-04-01:
I don't think you can check link quality within OZW, maybe not even not supported by Z-Wave. This is a BIG flaw as you are not really able to error track or try optimize nodes with a cable antenna. 
However, bad link quality can happen randomly by many reasons, interference from other nodes or other equipment's, even a defective switched power supply "polluting the air", etc. etc. Even the node itself may degrade over time... not very easy, in other words.
I would say I have less issues with false data since OZW 1.6, still it happens... but as I meanwhile also got better meshing by some new nodes I cannot say for sure.

Fishwaldo, 2020-04-04:
The Payload structure for these messages is very precise and complex. Having a RF issue that "corrupts" a message, while still having a the payload valid *AND* the CRC message correct is less than 0.1% IMHO. 

Every report I've seen of this with logs, indicates that its a bug in the SiLabs SDK that vendors are using to develop their products on. I've reported it to both vendors and SiLabs and some vendors acknowledge it and said they also reported it upstream. 

Please see #1687

closed 2020-04-04 by Fishwaldo