OpenZWave/open-zwave #2177: Configuration.cpp output may be truncated before the last format character error

TheNetopyr, 2020-04-03:
Dear all,

I am not sure if I'll inform you correctly this way, however since a few weeks I am unable to compile the open-zwave build. The build hangs on the following error.

Building src/command_classes/Configuration.cpp
/home/netopyr/open-zwave-read-only/cpp/src/command_classes/Configuration.cpp: In member function ‘virtual bool OpenZWave::Internal::CC::Configuration::HandleMsg(const uint8*, uint32, uint32)’:
/home/netopyr/open-zwave-read-only/cpp/src/command_classes/Configuration.cpp:263:41: error: ‘snprintf’ output may be truncated before the last format character [-Werror=format-truncation=]
  263 |       snprintf(label, 16, "Parameter #%d", parameter);
      |                                         ^
/home/netopyr/open-zwave-read-only/cpp/src/command_classes/Configuration.cpp:263:15: note: ‘snprintf’ output between 13 and 17 bytes into a destination of size 16
  263 |       snprintf(label, 16, "Parameter #%d", parameter);
      |       ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[1]: *** [/home/netopyr/open-zwave-read-only/cpp/build/support.mk:192: /home/netopyr/open-zwave-read-only/.lib/Configuration.o] Error 1
make[1]: Leaving directory '/home/netopyr/open-zwave-read-only/cpp/build'
make: *** [Makefile:23: all] Error 2

It is build with:

gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-pkgversion='Arch Linux 9.3.0-1' --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++,d --enable-shared --enable-threads=posix --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --enable-cet=auto gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
gcc version 9.3.0 (Arch Linux 9.3.0-1) 

If you need anything additional, please let me know

Fishwaldo, 2020-04-04:
Related to #2170 I'll merge into master soon.

Fishwaldo, 2020-04-04:
Cherrypicked from Dev Branch.

closed 2020-04-04 by Fishwaldo

TheNetopyr, 2020-04-04:
Perfect, now it works again, thanks a lot for your great work!