OpenZWave/open-zwave #2179: ZW132 Dual Nano Switch

skoppe, 2020-04-08:
In the boolean variant of SetValue, OZW sends a SET message and queues a GET message to be send immediately thereafter.

The only problem is that the ZW132 often reports the old value.

I found that by setting the configuration 80 to Basic Report, the ZW132 will send a basic report message a few seconds after the first SET. This message will trigger OZW to send another GET message and this one will retrieve the correct value.

The problem is that the value reported thus flips back and forth, resulting in weird gui behavior.

futuros, 2020-04-13:
I have the same problem, with both the dual nano (ZW140) and nano (ZW139) switch

skoppe, 2020-04-14:
Setting the configuration 80 to basic report does help. With it the correct value will eventually arrive.

I am curious why OZW sends a GET message directly after the SET. Likely because some units don't report anything otherwise.

I chatted with the aeotec engineers and they were very helpful. It appears the nano switches aren't build to be read out like OZW does, but were designed to have the controller wait for a basic report and then issue a GET. I said I considered the behaviour a bug, since OZW does wait until the SET has been replied to. At which point anyone would expect a GET to return the current or the to-be value.

It is anyones guess why the switches would still report the old value after a SET.

Fishwaldo, 2020-04-15:
This is a issue with the Specifications that has been addressed in more recent releases (specifically the Binary/MultiLevel Switch CommandClasses Version 4 and above from memory).

The Specifications originally did not state if a device should return the "instant" value (at the time it receives a GET message) or the final value (if its in the midst of transitioning to a new value). Some vendors return the "future" state (which wouldn't result in the weird behavior you see) while others return current state (which is what you see).

You can enable SetChangeVerified on the ValueID in question (if your application exposes that API) to get OZW to query the state of the device twice (in most cases) before confirming the new value to the application. 

Newer devices will expose both a "Level" and a "Target" value, where Level is the "instant" value, and the target is what the device is transitioning too. 

>I am curious why OZW sends a GET message directly after the SET. Likely because some units don't report anything otherwise.

Because there is no way at the protocol level to see if the device has "accepted" the new value with a SET message. Hence we need todo a GET to ensure its been done. Its uncommon for devices to return a update to associated nodes like this device does.

closed 2020-04-15 by Fishwaldo

skoppe, 2020-04-15:
> This is a issue with the Specifications that has been addressed in more recent releases (specifically the Binary/MultiLevel Switch CommandClasses Version 4 and above from memory).

Thanks for replying.

> You can enable SetChangeVerified on the ValueID in question (if your application exposes that API) to get OZW to query the state of the device twice (in most cases) before confirming the new value to the application.

That is a good workaround. Thanks. I probably just fork OZW though, and remove the deferred GET.

> Because there is no way at the protocol level to see if the device has "accepted" the new value with a SET message. Hence we need todo a GET to ensure its been done. Its uncommon for devices to return a update to associated nodes like this device does.

How would that work in the case where you have 2 controllers? If the device doesn't send a report how would the other controller know it changed?

I think it makes sense to send a basic report in an async network, especially considering the nano has an electrical switch as well.