OpenZWave/open-zwave #2197: [question] Skipping Cache Save for Node XX as its not past QueryStage_CacheLoad

m4dm4rtig4n, 2020-04-27:
Hi,

I have a problem with battery device :/

When i restart my docker zwave2mqtt, all battery device have this message :

...
2020-04-26 21:49:22.173 Info, Node011, Cache Save for Node 11 as its QueryStage_CacheLoad
2020-04-26 21:49:22.173 Info, Node012, Skipping Cache Save for Node 12 as its not past QueryStage_CacheLoad
2020-04-26 21:49:22.173 Info, Node013, Skipping Cache Save for Node 13 as its not past QueryStage_CacheLoad
2020-04-26 21:49:22.173 Info, Node014, Skipping Cache Save for Node 14 as its not past QueryStage_CacheLoad
2020-04-26 21:49:22.173 Info, Node015, Skipping Cache Save for Node 15 as its not past QueryStage_CacheLoad
2020-04-26 21:49:22.173 Info, Node016, Skipping Cache Save for Node 16 as its not past QueryStage_CacheLoad
2020-04-26 21:49:22.176 Info, Node018, Cache Save for Node 18 as its QueryStage_CacheLoad
...
And have 0 data in User, Configuration, System and device not worker until they haven't had "wakeup".

Any idea why battery device can't load cache ?

Docker configuration :
docker run --rm -it -p 8091:8091 --device=/dev/ttyACM0 -v $(pwd)/store:/usr/src/app/store robertslando/zwave2mqtt:latest
=> OZW 1.6.1080

thx

m4dm4rtig4n, 2020-04-27:
And when module is dead without command :
```
2020-04-27 23:10:24.250 Detail, Node012,   Received: 0x01, 0x0f, 0x00, 0x04, 0x00, 0x19, 0x09, 0x71, 0x05, 0x00, 0x00, 0x00, 0xff, 0x07, 0x08, 0x00, 0x60
2020-04-27 23:10:24.250 Detail, 
2020-04-27 23:10:24.250 Info, Node012, ApplicationCommandHandler - Unhandled Command Class 0x71
```

:(

Fishwaldo, 2020-04-30:
You need to "wake up" battery powered devices after restarting OZW. Consult the manual of your device on how to do that.

Alternatively, most devices will automatically wake up after hours (or days) and "become ready"

closed 2020-04-30 by Fishwaldo