OpenZWave/open-zwave #2199: New device - D-Link DCH-Z210 ZW+ Plug

jefft4, 2020-04-27:
<Product config="dlink/dch-z210.xml" id="0011" name="DCH-Z210 Gen5 Plug" type="0001"/>
[dlink April2020.zip](https://github.com/OpenZWave/open-zwave/files/4540834/dlink.April2020.zip)

Fishwaldo, 2020-04-30:
Please open a PR - We rely on the community to maintain the database.

closed 2020-04-30 by Fishwaldo

jefft4, 2020-04-30:
@Fishwaldo , the instructions on the Wiki about adding new devices say to EITHER raise a PR if you use git for coding (I don't, I'm not actively programming anything) or load the changes into an issue.  I did the latter.  If the team only wants to deal with PRs now, then please will you correct the instructions.  
I've raised three new devices with issues and you're telling me to do it all again - after setting up GIT, working out how to do the PR, etc etc... when I will have time to do that, I don't know!

Fishwaldo, 2020-04-30:
I'm essentially the only one working on OZW now. I don't have the bandwidth unfortunately with commitments in real life, family and work. if someone else picks it up, then I'll merge it.

jefft4, 2020-04-30:
OK, understood.  I'll see what I can do with these configs, ought not to be difficult ;-)
Being down to one person isn't a great situation for the project; have you tried reaching out to the developer communities of the projects that depend on OZW - Home Asst., OpenHAB, domoticz, etc etc?  They may well have the right skills and they need OZW to be well-supported.  
If it helps, I'd be happy to post on the domoticz forum and some more general home automation groups to ask if anyone has the right skills to help you.

jefft4, 2020-04-30:
@Fishwaldo - I got Git Desktop up and running and created the PR.  Can't complete it, though; xmltest online is failing with an error that doesn't look related to my changes...  Any idea?

##[error]/home/runner/work/open-zwave/open-zwave/cpp/build/support.mk:69: git describe returned an empty result, setting GITVERSION to VERSION_MAJ.VERSION_MIN.-1 and VERSION_REV to 0

jefft4, 2020-05-01:
Sussed it; first error is weird but went away when I found the second.  PR checked and ready for action -> #2203 .