OpenZWave/open-zwave #2201: Segmentation Fault in libopenzwave

pboers1988, 2020-04-30:
According to @robertsLando om the Z2M project the following issue [link](https://github.com/OpenZWave/Zwave2Mqtt/issues/353) is a problem in this project. 

When updating a fibaro dimmer I experience a segfault in libopenzwave. 

**This is the log of Zwave to MQTT**

As you can see it receives the message and then just dies and respawns.

```bash
2020-04-29T19:04:14.904Z z2m:Mqtt Message received on homeassistant/Woonkamer/Woonkamer-Dimmer/38/1/0/set
[29/Apr/2020:21:04:14 +0200] 200 -(10.0.0.40) GET /socket.io/?EIO=3&transport=polling&t=N77KoIk&sid=hSMatRfN9cd4AdlcAAAD HTTP/1.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_4) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1 Safari/605.1.15)
2020-04-29 21:04:14.910 Info, Node004, Value::Set - COMMAND_CLASS_SWITCH_MULTILEVEL - Level - 0 - 1 - 51
2020-04-29 21:04:14.911 Info, Node004, SwitchMultilevel::Set - Setting to level 51
2020/04/29 21:04:15 [error] 415#415: *205 recv() failed (104: Connection reset by peer) while reading response header from upstream, client: 10.0.0.40, server: a0d7b954-zwave2mqtt, request: "GET /socket.io/?EIO=3&transport=polling&t=N77KqCp&sid=hSMatRfN9cd4AdlcAAAD HTTP/1.1", upstream: "http://127.0.0.1:44920/socket.io/?EIO=3&transport=polling&t=N77KqCp&sid=hSMatRfN9cd4AdlcAAAD", host: "10.0.0.32:8091", referrer: "http://10.0.0.32:8091/"
[29/Apr/2020:21:04:15 +0200] 502 -(10.0.0.40) GET /socket.io/?EIO=3&transport=polling&t=N77KqCp&sid=hSMatRfN9cd4AdlcAAAD HTTP/1.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_4) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1 Safari/605.1.15)
2020/04/29 21:04:15 [error] 415#415: *205 connect() failed (111: Connection refused) while connecting to upstream, client: 10.0.0.40, server: a0d7b954-zwave2mqtt, request: "POST /socket.io/?EIO=3&transport=polling&t=N77KqKo&sid=hSMatRfN9cd4AdlcAAAD HTTP/1.1", upstream: "http://127.0.0.1:44920/socket.io/?EIO=3&transport=polling&t=N77KqKo&sid=hSMatRfN9cd4AdlcAAAD", host: "10.0.0.32:8091", referrer: "http://10.0.0.32:8091/"
[29/Apr/2020:21:04:15 +0200] 502 -(10.0.0.40) POST /socket.io/?EIO=3&transport=polling&t=N77KqKo&sid=hSMatRfN9cd4AdlcAAAD HTTP/1.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_4) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1 Safari/605.1.15)
[21:04:15] INFO: Starting the Z-Wave to MQTT...
2020-04-29T19:04:16.046Z z2m:Store scenes.json not found
2020-04-29T19:04:16.873Z z2m:Gateway No customDevices file found
2020-04-29T19:04:16.878Z z2m:App Application path:/opt
2020/04/29 21:04:16 [error] 415#415: *205 connect() failed (111: Connection refused) while connecting to upstream, client: 10.0.0.40, server: a0d7b954-zwave2mqtt, request: "GET /socket.io/?EIO=3&transport=polling&t=N77KqiE HTTP/1.1", upstream: "http://127.0.0.1:44920/socket.io/?EIO=3&transport=polling&t=N77KqiE", host: "10.0.0.32:8091", referrer: "http://10.0.0.32:8091/"
[29/Apr/2020:21:04:16 +0200] 502 -(10.0.0.40) GET /socket.io/?EIO=3&transport=polling&t=N77KqiE HTTP/1.1 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_4) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1 Safari/605.1.15)
Initialising OpenZWave 1.6.974 binary addon for Node.JS.
	OpenZWave Security API is ENABLED
	ZWave device db    : etc/openzwave/
	User settings path : /opt/store
	Option Overrides : --Logging true --ConsoleOutput true --QueueLogLevel 8 --DriverMaxAttempts 9999 --SaveConfiguration true --PollInterval 60000 --AutoUpdateConfigFile true
2020-04-29T19:04:17.061Z z2m:Zwave Connecting to /dev/ttyACM0
```

**This is the corresponding log entry in the dmesg of Ubuntu**

```bash
[wo apr 29 21:03:56 2020] node[5685]: segfault at 63 ip 00007f0bf6da9bc4 sp 00007ffd71293450 error 4 in libopenzwave.so.1.6[7f0bf6d5b000+9c000]
```

I'm running Z2M 3.0.2 as packaged by Home Assistant in the most recent add-on.

@robertsLando indicated that @Fishwaldo would have a look at this issue.

Thanks

Fishwaldo, 2020-04-30:
You need to get a OZW Log file, and optionally a backtrace of the crash (running it under GDB) to figure it out. 

Those logs are not useful.

pboers1988, 2020-04-30:
The OZW logs didn't really give any meaningful extra info.

This is the output of gdb

```bash
Thread 1 "node" received signal SIGSEGV, Segmentation fault.
0x00007f2e2d376bc4 in OpenZWave::Internal::CC::SwitchMultilevel::SetLevel(unsigned char, unsigned char) ()
   from target:/usr/lib/libopenzwave.so.1.6
(gdb)
Continuing.
Couldn't get registers: No such process.
```

pboers1988, 2020-04-30:
```bash
hread 1 "node" received signal SIGSEGV, Segmentation fault.
0x00007f2b8ded2bc4 in OpenZWave::Internal::CC::SwitchMultilevel::SetLevel(unsigned char, unsigned char) ()
   from /usr/lib/libopenzwave.so.1.6
(gdb) backtrace
#0  0x00007f2b8ded2bc4 in OpenZWave::Internal::CC::SwitchMultilevel::SetLevel(unsigned char, unsigned char) ()
   from /usr/lib/libopenzwave.so.1.6
#1  0x00007f2b8ded3277 in OpenZWave::Internal::CC::SwitchMultilevel::SetValue(OpenZWave::Internal::VC::Value const&)
    () from /usr/lib/libopenzwave.so.1.6
#2  0x00007f2b8deda847 in OpenZWave::Internal::VC::Value::Set() () from /usr/lib/libopenzwave.so.1.6
#3  0x00007f2b8ded5d1f in OpenZWave::Internal::VC::ValueByte::Set(unsigned char) () from /usr/lib/libopenzwave.so.1.6
#4  0x00007f2b8def64c9 in OpenZWave::Manager::SetValue(OpenZWave::ValueID const&, unsigned char) ()
   from /usr/lib/libopenzwave.so.1.6
#5  0x00007f2b8dfa665d in OZW::OZW::SetValue(Nan::FunctionCallbackInfo<v8::Value> const&) ()
   from /opt/node_modules/openzwave-shared/build/Release/openzwave_shared.node
#6  0x00007f2b8df8fb18 in Nan::imp::FunctionCallbackWrapper(v8::FunctionCallbackInfo<v8::Value> const&) ()
   from /opt/node_modules/openzwave-shared/build/Release/openzwave_shared.node
#7  0x00005605548c907a in v8::internal::FunctionCallbackArguments::Call(v8::internal::CallHandlerInfo) ()
#8  0x00005605548c9499 in ?? ()
#9  0x00005605548c9be8 in ?? ()
#10 0x00005605548ca4c4 in v8::internal::Builtin_HandleApiCall(int, unsigned long*, v8::internal::Isolate*) ()
#11 0x0000560554e1ca99 in ?? ()
#12 0x00007ffecc9660d0 in ?? ()
#13 0x000000000000002a in ?? ()
#14 0x00007ffecc966170 in ?? ()
#15 0x0000560554d9cfe4 in ?? ()
#16 0x00001dd2db3c04a9 in ?? ()
#17 0x000020e71f737c11 in ?? ()
#18 0x0000000700000000 in ?? ()
#19 0x00001dd2db3c0589 in ?? ()
#20 0x0000003100000000 in ?? ()
#21 0x000001c215db7e11 in ?? ()
#22 0x00001162eacb8541 in ?? ()
#23 0x00001dd2db3c04a9 in ?? ()
#24 0x00001dd2db3c04a9 in ?? ()
#25 0x00001162eacb8541 in ?? ()
#26 0x000020e71f737c11 in ?? ()
#27 0x0000319afd718171 in ?? ()
#28 0x0000004b00000000 in ?? ()
#29 0x00001a4e7251fb51 in ?? ()
#30 0x000020e71f73b979 in ?? ()
#31 0x0000319afd718171 in ?? ()
#32 0x00007ffecc9661e8 in ?? ()
#33 0x0000560554d9cfe4 in ?? ()
#34 0x0000003100000000 in ?? ()
#35 0x000001c215db7e11 in ?? ()
#36 0x00001162eacb8329 in ?? ()
#37 0x00001dd2db3c04a9 in ?? ()
#38 0x000001c215db7e11 in ?? ()
#39 0x0000003100000000 in ?? ()
#40 0x00001162eacb8329 in ?? ()
#41 0x000020e71f73b979 in ?? ()
#42 0x000001c215db7e11 in ?? ()
#43 0x0000007a00000000 in ?? ()
#44 0x00001a4e7251ee69 in ?? ()
#45 0x0000319afd73d6f9 in ?? ()
#46 0x0000319afd735041 in ?? ()
#47 0x00007ffecc966250 in ?? ()
#48 0x00001d14dd1033e7 in ?? ()
#49 0x0000003100000000 in ?? ()
```

Fishwaldo, 2020-04-30:
>The OZW logs didn't really give any meaningful extra info.

Yes the do - They tell me the state of OZW prior to the crash. 

Can you please tell me the if the "QueryStage" of the node is "Complete" when the crash happens?

pboers1988, 2020-04-30:
@Fishwaldo Ah you need the full logs, the node is "Complete" at the time of the crash. 

attached you can find the startup log of OZW

and here is the snippet during the crash
```bash
020-04-30 16:05:39.386 Warning, NotificationCCTypes::GetAlarmNotificationTypes - Unknown Alarm Type 7
2020-04-30 16:05:39.386 Warning, NotificationCCTypes::GetEventForAlarmType - Unknown AlarmType/Event 7/d
2020-04-30 16:05:39.386 Warning, NotificationCCTypes::GetAlarmType - Unknown AlarmType 7
2020-04-30 16:05:39.387 Info, Node010, Received Notification report (>v1): Type: Unknown (7) Event: Unknown (8) Status: true, Param Length: 0
2020-04-30 16:05:39.387 Detail, Node010, Refreshed Value: old value=0, new value=3, type=list
2020-04-30 16:05:39.387 Detail, Node010, Changes to this value are not verified
2020-04-30 16:05:39.387 Detail, Node010, Notification: ValueChanged
2020-04-30 16:05:39.442 Detail, Node010,   Received: 0x01, 0x0b, 0x00, 0x04, 0x08, 0x0a, 0x03, 0x30, 0x03, 0xff, 0xbe, 0x00, 0x83
2020-04-30 16:05:39.442 Detail,
2020-04-30 16:05:39.442 Info, Node010, Received SensorBinary report: State=On
2020-04-30 16:05:39.442 Detail, Node010, Refreshed Value: old value=false, new value=true, type=bool
2020-04-30 16:05:39.442 Detail, Node010, Changes to this value are not verified
2020-04-30 16:05:39.443 Detail, Node010, Notification: ValueChanged

2020-04-30 16:05:39.507 Detail, Node010,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x0a, 0x03, 0x30, 0x03, 0xff, 0xbe, 0x00, 0x8b
2020-04-30 16:05:39.508 Detail,
2020-04-30 16:05:39.508 Info, Node010, Received SensorBinary report: State=On
2020-04-30 16:05:39.508 Detail, Node010, Refreshed Value: old value=true, new value=true, type=bool
2020-04-30 16:05:39.508 Detail, Node010, Changes to this value are not verified
2020-04-30 16:05:39.509 Detail, Node010, Notification: ValueChanged

2020-04-30 16:05:45.466 Info, Node004, Value::Set - COMMAND_CLASS_SWITCH_MULTILEVEL - Level - 0 - 1 - 50
2020-04-30 16:05:45.467 Info, Node004, SwitchMultilevel::Set - Setting to level 50
tail: /usr/share/hassio/addons/data/a0d7b954_zwave2mqtt/OZW_Log.txt: file truncated
```

[ozw.log](https://github.com/OpenZWave/open-zwave/files/4558801/ozw.log)

pboers1988, 2020-05-05:
any ideas @Fishwaldo ?

closed 2022-02-08 by pboers1988