OpenZWave/open-zwave #2204: Logic Soft ZDB5100: Missing 3 of 4 0x33 COMMAND_CLASS_SWITCH_COLOR

LordMike, 2020-05-01:
Hi,

I've acquired a Logic Soft ZDB5100, a wall-switch with 4 pushbuttons (each with an RGB LED), and one dimming output (for lights). In Zwave2MQTT, I see the device with a few inconsistencies.. such as more devices than I'd expect. But that's fine. 

For the RGB LED's, I can only control the first of the 4 LED's.. which sucks :).

I refreshed the node using Zwave2MQTT, and I got this [log snippet](https://github.com/OpenZWave/open-zwave/files/4565176/OZW_Log2.txt) out of it. In it, I can see messages like this:

> 2020-05-01 15:15:17.690 Detail, Node020, Notification: ValueAdded CC: COMMAND_CLASS_COLOR Instance: 1 Index: 2
> 2020-05-01 15:15:17.690 Detail, Node020, Notification: ValueAdded CC: COMMAND_CLASS_COLOR Instance: 2 Index: 2
> 2020-05-01 15:15:17.690 Detail, Node020, Notification: ValueAdded CC: COMMAND_CLASS_COLOR Instance: 3 Index: 2
> 2020-05-01 15:15:17.690 Detail, Node020, Notification: ValueAdded CC: COMMAND_CLASS_COLOR Instance: 4 Index: 2
> 2020-05-01 15:15:17.690 Detail, Node020, Notification: ValueAdded CC: COMMAND_CLASS_COLOR Instance: 5 Index: 2
> 2020-05-01 15:15:17.883 Detail, Node020, Notification: ValueAdded CC: COMMAND_CLASS_COLOR Instance: 1 Index: 0
> 2020-05-01 15:15:17.884 Detail, Node020, Notification: ValueAdded CC: COMMAND_CLASS_COLOR Instance: 1 Index: 1

.. this matches what I see in Z2M, that only instance 1 has a color selector / setter .. which in turn controls button no. 1. The other three buttons get the color command class as seen here, but they don't have the critical "Index: 0" and "Index: 1" ... which apparently are necessary.

[image]

I also loaded up the device in Zensys Z-Wave controller, (moved my z-wave stick to a PC) to see what it reported for this device. In the controller, we see the wall switch has 5 endpoints (?), the 4 representing the push buttons, and the fifth the dimmer output. Each of the 4 buttons has a Color switch CC, which I've invoked for buttons 3 and 4 .. so they're now in some colors, which means it all works.

[image]

What's missing? Is it a config issue, a bug, or something else entirely?

Note: I'm prepared to ship one of these to @Fishwaldo :)

Note: I know the log above doesn't match the guidelines, so I made a new one where I cleared the ozwcache files. I couldn't find a zwcfg* file ... maybe its hidden within the docker container I'm using.. 

[Scanned log](http://www.openzwave.com/log-analyzer/public/4b29bb84-8abe-48ee-9057-f514254c9c3a/).

LordMike, 2020-05-05:
I've run ozw-daemon for a little while, to see if I could get more CC's out of that, to see if it was a Zwave2Mqtt issue.

In the picture below, ozw-daemon has produced a number of topics related to the 0x33 command class, but only instance 1 in my case has the "Instance 1: Color" aspect (feature?). 

[image]

Fishwaldo, 2020-05-11:
I would need to see the full log file from OZW where ozwcache_*.xml is removed to see what is discovered.

LordMike, 2020-05-11:
Cool. I'll see if I can make that happen later today. I should be able to do it with ozw-daemon.

LordMike, 2020-05-11:
I wasn't able to run ozwdaemon and produce an `OZW_Log.txt`, so I ran a new fresh instance of `zwave2mqtt`, which produced the log file. After some time where I think it got all the info it could (besides the battery powered devices), I sent the log to the analysis site. How does one rid oneself of the `zwcfg*.xml` files?

I've made a few runs, each one with this warning. I finally settled on clearing all the device config files, as they're in a folder with a `zwcfg.xsd` file.. which is kinda close enough. Perhaps the `zwcfg*.xml` refers to all the device configs?

In any case, there's more findings now. More internal errors.. :)

http://www.openzwave.com/log-analyzer/public/ef879f2a-a844-4f2a-817e-e046c4f1ff0f/

Fishwaldo, 2020-05-22:
Firstly - You need to fix this error:
>2020-05-11 22:27:58.043 Info, Unable to load config file /usr/local/etc/openzwave/2gig/ct50e.xml

Something is wrong with the paths setup in zwave2mqtt. (all config files are unable to be opened, including the config file for your device:
>2020-05-11 22:27:58.063 Info, Unable to load config file /usr/local/etc/openzwave/logicsoft/ZDB5100.xml

Anyway, what I can see:
For Instance 1 - The device reports its capabilities correctly:
>2020-05-11 22:29:51.552 Info, Node020, Sending (Query) message (Callback ID=0x8b, Expected Reply=0x04) - ColorCmd_CapabilityGet (Node=20): 0x01, 0x09, 0x00, 0x13, 0x14, 0x02, 0x33, 0x01, 0x25, 0x8b, 0x6f
2020-05-11 22:29:51.552 Info, Node020, Encrypted Flag is 0
2020-05-11 22:29:51.560 Detail, Node020,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2020-05-11 22:29:51.560 Detail, Node020,   ZW_SEND_DATA delivered to Z-Wave stack
2020-05-11 22:29:51.576 Detail, Node020,   Received: 0x01, 0x07, 0x00, 0x13, 0x8b, 0x00, 0x00, 0x02, 0x62
2020-05-11 22:29:51.576 Detail, Node020,   ZW_SEND_DATA Request with callback ID 0x8b received (expected 0x8b)
2020-05-11 22:29:51.576 Info, Node020, Request RTT 24 Average Request RTT 23
2020-05-11 22:29:51.576 Detail, Node020,   Expected callbackId was received
2020-05-11 22:29:51.591 Detail, Node020,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x14, 0x04, 0x33, 0x02, 0x1c, 0x00, 0xcc
2020-05-11 22:29:51.591 Detail, 
2020-05-11 22:29:51.591 Info, Node020, Response RTT 39 Average Response RTT 40
2020-05-11 22:29:51.591 Info, Node020, Received an Color Capability Report: Capability=1c
2020-05-11 22:29:51.591 Info, Node020, Red (0x02)
2020-05-11 22:29:51.591 Info, Node020, Green (0x03)
2020-05-11 22:29:51.591 Info, Node020, Blue (0x04)

But there was a bug in OZW where we didn't ask the for capabilities for the correct instance when Color CC is present on more than one instance. I'm about to push a fix for that. 

Please report if thats fixed this issue.

Fishwaldo, 2020-05-22:
(next time you can just attach the log file to Github)

LordMike, 2020-05-22:
Yea, I was attempting to remove the `zwcfg*.xml` files, so I removed all configs for all products. It helped, as the logs became longer (I assume more data was queried rather than being loaded from configs)..

But awesome you found something :)

LordMike, 2020-05-22:
Which version can I aim to try, w/ Z2M? 

I'm on `1.6.1123`

Fishwaldo, 2020-05-23:
1.6.1133

LordMike, 2020-05-26:
Eureka!.. It works :D

Now I'm ready for the electrician to install the rest :D

closed 2020-05-26 by LordMike