OpenZWave/open-zwave #2205: OpenZWave reinterviewing all devices at every restart

akamming, 2020-05-01:
on request of @gizmocuz i  post this issue here...

for some unknown reason for some users at domoticz OpenZWave is re-interviewing all devices at every restart, even if the information is in the ozwcache.xml file (as far as i can see, but my knowledge does not go very far). This is an issue, cause at every reboot you have to wake up all battery powered devices manually before they work again. 

additional information can be found in the [issue on domoticz](https://github.com/domoticz/domoticz/issues/4119) and [the forum thread](https://www.domoticz.com/forum/viewtopic.php?f=24&t=32401&sid=711107c8048a9eff304fc5dac678210a)

Let me know what additional info you need for analysis, then i can try to gather it...

akamming, 2020-05-04:
Additional info. I just found out that
- After a power cycle (taking my pi of the power): all devices are loaded from ozwcache
- after a restart of the application without power cycle: 3 devices are loaded from cache. domoticz (with openzave 1.6) starts re-interviewing all other devices.

Fishwaldo, 2020-05-11:
Sleeping devices need to be woken up after restart to get the initial data that is needed to validate the cache is valid. They will not "operate" correctly till they are woken up (and proceed past QueryStage_CacheLoad.

How are you determining that devices are being re-interviewed. For any device that has proceeded past QueryStage_CacheLoad, we still get the "dynamic" values from the device - That is the latest value (eg, temp, status of a switch) after restarting.

Fishwaldo, 2020-05-11:
By the way - Looking at the domoticz issue - If you are experiencing a crash - Then it might be at a point where caching is not completed. I can't diagnose that issue.

akamming, 2020-05-12:
> How are you determining that devices are being re-interviewed. For any device that has proceeded past QueryStage_CacheLoad, we still get the "dynamic" values from the device - That is the latest value (eg, temp, status of a switch) after restarting.

i see this activity happening in the control panel.  most devices start with status "probe" and get several statuses before they are "ready"  again. The battery powered devices start with "probe (sleeping). After i manually wake them up them up, they go through several statuses till they are " sleeping"  again

> By the way - Looking at the domoticz issue - If you are experiencing a crash - Then it might be at a point where caching is not completed. I can't diagnose that issue.

I already did some more testing on this. The crash is unrelated. I found out that
1 If i completely power down my system and then restart domoticz. All my devices are read from the cache and everything works as it should
2  If i reboot or restart without a  power cycle, i have this issue. on the domoticz forum this behaviour was confirmed by another user. (to be very specific: If i do a shutdown and restart the pi without it taking it off the power, i still have this issue. if i  switch off the power for 10 sec after the shutdown the issue does not occur)
3 The crash itself is unrelated to the problem, but does cause a restart of domoticz without a power cycle of the pi, which makes the issue occur on my system.

Jofreathome, 2020-05-16:
I see a similar behaviour, after a restart I see that there are 3 powered nodes added. One of them is the AEON Labs ZW090 Z-Stick Gen5+ EU which then gets the description "Static PC Controller" written in red on the hardware page > nodes. 
After a complete shut down of the Pi3b and a waiting time before power up the description is back to "ZW090 Z-Stick Gen5 EU" and there are no new nodes added.

Maybe it is related to this issue #2219

akamming, 2020-05-16:
Hi might be related. But my zwave device is not a z-stick gen 5. I use a zwave.me   zme-uzb1 stick

akamming, 2020-05-18:
i did a lot of research today. I know why domoticz re-interviews: the ozwcache file is incomplete. So this issue probably not in the initialisation fase op OZW, but at the closing phase when there is a graceful shutdown

-  if i use an old ozwcahe file which was still complete: No interviewing takes place at the startup as it should
- as long as domoticz is running the ozwcach file is ok (filesize is around 980k.)
- i even checked by adding a device: It is also added to the ozwcache file. This is a correct ozwcache file, cause if i restart domoticz with manually putting this file in place all works fine.
- but as soon as i do a graceful stop of domoticz, the ozwcache file is reduced to around 47k and misses almost all devices.

The behaviour that it works OK after a power cycle i cannot reproduce anymore

akamming, 2020-05-19:
Please look at the info in [this issue](https://github.com/domoticz/domoticz/issues/4119). There is a bug in the code when polling is enabled which is now confirmed by more users.

I traced it down to [this commit](https://github.com/OpenZWave/open-zwave/commit/f3920e88d731577dbe058e64ebb91aa8e386bd8c) as a result of [this issue](https://github.com/OpenZWave/qt-openzwave/issues/45).

This commit ensures that on more occassions the cache is saved, but as a result WriteCache is now also called in the destructor of node.cpp (the distructor calls Driver->DisablePoll() which rewrites the cache when already several nodes are "destructed"), leaving an incomplete cache file, which causes the re-interviewing at next start.

Can you please fix this? i already tested a quickfix (removing the Writecache() statement from DisablePoll ) which fixed the issue. but i'm not sure if this is the right fix, cause i don't know the full impact

Fishwaldo, 2020-05-22:
Thanks - I put in a alternative fix as there may be other code paths that call WriteCache during shutdown.

closed 2020-05-22 by Fishwaldo

akamming, 2020-05-22:
@Fishwaldo tx for fixing!! just recompiled domoticz with the new version. i can confirm the alternate fix works as well!