OpenZWave/open-zwave #2208: FGDW002 Door Opening Sensor 2 (FIBARO System) not show open/close

poudenes, 2020-05-04:
Hi All,

I have add  2 FGDW002 Fibaro Door Sensors in Zwave2MQTT with some other nodes as well.
They don't have any MQTT part to send OPEN/CLOSE

Version: Openzwave 1.6.974

What can I do to let those 2 devices work correct?

Fishwaldo, 2020-05-11:
Please update to the latest version of OZW (1.6.1xxx) - there was a fix related to not loading specific config files. (in this case, NotificationCCTypes.xml - Please check the log files at the start for messages about unable to load this file)

closed 2020-05-11 by Fishwaldo

poudenes, 2020-05-11:
@Fishwaldo Im so sorry, I solved the issue. The Path was not correct to config files:
Add this: "/etc/openzwave"

Removed the 2 sensors, add them again and got all the missing options.

Forgotten to post it here and close it myself.