OpenZWave/open-zwave #2214: ZW095 Home Energy Meter Gen5 (AEON Labs) not updating / wrong initialization

magpern, 2020-05-11:
**Version**
Z2M docker dev

Build/Run method
- [X] Docker
- [ ] PKG
- [ ] Manually built (git clone - npm install - npm run build ) 

Zwave2Mqtt version: 3.1.0
OpenZwave 1.6.1115

**Describe the bug**
I have just bought a ZW095 Home Energy Meter Gen5 (AEON Labs)
It registers fine in Z2M, but it never updates. It only updates if I manually select to refresh node info. Nothing in the log, nothing in MQTT. Total silence.

**To Reproduce**
Steps to reproduce the behavior:
Just include the node and wait. Nothing happens

**Expected behavior**
The device to update as expected according to settings.

**Additional context**
I have found how to make the device update. After hours and hours of googling.
I found it [here](https://community.openhab.org/t/solved-home-energy-meter-hem-g3-aeon-labs-gen5-zw095-do-not-report-automatically/44579/22) Austris_V post from Dec '18 (now called @billiaz )

It is something with the association groups that is not set up correctly.
I downloaded Z-Wave PC Controller and manually set the association groups (I think that is what I did)

[image]

After sending the command "85 01 0X 01" (where X is the Grouping identifyer) the device started updating according to the set number of seconds in setting 111 to 113

Somehow the device is not initializing correctly or if the xml file for the device is wrong.
I figured, if I need to send ASSOCIATION_SET on group 1,2 and 3 should not the xml file contain these groups?
It just contains a LIFELINE
[Link to xml settings](https://github.com/OpenZWave/open-zwave/blob/master/config/aeotec/zw095.xml)

Link to device inclusion process [here](https://paste.ubuntu.com/p/vJDSnBXwTr/)

magpern, 2020-05-12:
@Fishwaldo  It just struck me... if sending "85 01 0X 01" helps, HEX 85 is DEC 133
133 is the command class for associations. Doesn't that simply mean the definition of the device is wrong.

I can change the xml file for the device, if I just can figure out what is means.

"85 01 00 01"
"85 01 01 01"
"85 01 02 01"
"85 01 03 01"

What in above is assosiation num_group and what is index.
The label? Is that important?

```json
  <CommandClass id="133">
    <Associations num_groups="1">
      <Group index="1" label="Lifeline" max_associations="5"/>
    </Associations>
  </CommandClass>`

Fishwaldo, 2020-05-22:
If the number of groups in the config file is incorrect, please send a PR adding the missing groups. 
>What in above is assosiation num_group and what is index.

The 3rd param is the group "index" and the forth is the target node id.

closed 2020-05-22 by Fishwaldo

alex-savin, 2020-05-26:
@magpern have you had a chance to find a solution to fix the issue?

magpern, 2020-05-26:
> @magpern have you had a chance to find a solution to fix the issue?

Well, the solution is to add more associations to rhe definition of the device. I can do that.
But will it work? Hard to test i think. My device is in production now and I don't want to screw it up.
The queation remains, that I have no answer too, will the initialization/inclusion of the device set the values needed if more assosiations are added.
Only answer I get is create a PR, but no real answer on how or what.

alex-savin, 2020-05-28:
@Fishwaldo Would you mind please to help us with that issue!
I would do anything for testing, but I can't fix it on my own, because of lack of zwave knowledge

alex-savin, 2020-06-13:
@magpern Would you mind to share you ozwcache_0xXXXXXX.xml file, just only records for the node of ZW095 Meter, please.

Thank you in advance

magpern, 2020-06-13:
@alex-savin sure. Let me know if this helps in any way. I did set the parameters needed with another zwave program. So, keep in mind. I have not altered anything in OpenZwave.

I have a new energy meter monitoring the whole house. I have to rewire it though, but when that is done, I could experiment with changing OpenZwave files to make this device work.
It will be truly trail-and-error. The problem with the OpenZwave files is that the number of association is wrong and they need to be set default to report to the zwave controller.

If anyone with knowledge of zwave just could explain how to translate
"85 01 00 01"
"85 01 01 01"
"85 01 02 01"
"85 01 03 01"

into xml for OpenZwave, this issue would be solved in no-time.

```json
<Node id="16" name="" location="" basic="4" generic="49" specific="1" roletype="5" devicetype="4096" nodetype="0" type="Simple Meter" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="100000" version="4" configrevision="5" query_stage="Complete">
		<Neighbors>1,30,128,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0</Neighbors>
		<Manufacturer id="86" name="AEON Labs">
			<Product type="2" id="5f" name="ZW095 Home Energy Meter Gen5">
				<MetaData>
					<MetaDataItem name="OzwInfoPage">http://www.openzwave.com/device-database/0086:005F:0102</MetaDataItem>
					<MetaDataItem name="ProductPic">images/aeotec/zw095.png</MetaDataItem>
					<MetaDataItem name="Description">Aeon Labs Home Energy Meter Gen5 is a Z-Wave device that can monitor the total amount of electricity of your home. It can report the wattage energy usage and KWH usage to a Z-Wave Controller (especially a gateway). In near real-time you’ll have a full understanding of how much electricity you use and when you use it. Products that are Z-Wave certified can be used and communicate with other Z-Wave certified devices. Also a tool that can be programmed, scheduled, controlled and communicated with from anywhere in the world.</MetaDataItem>
					<MetaDataItem name="ProductManual">https://Products.Z-WaveAlliance.org/ProductManual/File?folder=&amp;filename=Manuals/2046/Home Energy Meter Gen5 manual.pdf</MetaDataItem>
					<MetaDataItem name="InclusionDescription">Turn the primary controller of Z-Wave network into inclusion mode, short press the product’s Action button that you can find on the product&apos;s housing.</MetaDataItem>
					<MetaDataItem name="ExclusionDescription">Turn the primary controller of Z-Wave network into exclusion mode, short press the product’s Action button that you can find on the product&apos;s housing.</MetaDataItem>
					<MetaDataItem name="ResetDescription">Press and hold the Action button that you can find on the product&apos;s housing for 10 seconds and then release. This procedure should only be used when the primary controller is inoperable.</MetaDataItem>
					<MetaDataItem name="WakeupDescription">N/A</MetaDataItem>
					<MetaDataItem name="Name">Home Energy Meter Gen5</MetaDataItem>
					<ChangeLog>
						<Entry author="Justin Hammond - Justin@dynam.ac" date="08 May 2019" revision="3">Initial Metadata Import from Z-Wave Alliance Database - https://products.z-wavealliance.org/products/1289/xml</Entry>
						<Entry author="Justin Hammond - Justin@dynam.ac" date="23 May 2019" revision="4">Updated Metadata Import from Z-Wave Alliance Database - https://products.z-wavealliance.org/products/1596/xml</Entry>
						<Entry author="Justin Hammond - Justin@dynam.ac" date="02 Jun 2019" revision="5">Updated Metadata Import from Z-Wave Alliance Database - https://products.z-wavealliance.org/products/2046/xml</Entry>
					</ChangeLog>
				</MetaData>
			</Product>
		</Manufacturer>
		<CommandClasses>
			<CommandClass id="32" name="COMMAND_CLASS_BASIC">
				<Compatibility>
					<Mapping>50</Mapping>
				</Compatibility>
				<State>
					<AfterMark>true</AfterMark>
					<CCVersion>1</CCVersion>
					<StaticRequests>5</StaticRequests>
				</State>
				<Instance index="1" />
				<Instance index="2" endpoint="1" />
				<Instance index="3" endpoint="2" />
				<Instance index="4" endpoint="3" />
			</CommandClass>
			<CommandClass id="50" name="COMMAND_CLASS_METER">
				<Compatibility />
				<State>
					<CCVersion>4</CCVersion>
					<InNif>true</InNif>
					<StaticRequests>3</StaticRequests>
				</State>
				<Instance index="1" />
				<Instance index="2" endpoint="1" />
				<Instance index="3" endpoint="2" />
				<Instance index="4" endpoint="3" />
				<Value type="decimal" genre="user" instance="1" index="0" label="Electric - kWh" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="141.041">
					<Help></Help>
				</Value>
				<Value type="decimal" genre="user" instance="3" index="0" label="Electric - kWh" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="17.799">
					<Help></Help>
				</Value>
				<Value type="decimal" genre="user" instance="4" index="0" label="Electric - kWh" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="113.287">
					<Help></Help>
[... 323 lines cut ...]
				</Value>
				<Value type="string" genre="system" instance="1" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="4.54">
					<Help>Z-Wave Protocol Version</Help>
				</Value>
[... cut ...]
			<CommandClass id="142" name="COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION">
				<Compatibility />
				<State>
					<CCVersion>1</CCVersion>
					<InNif>true</InNif>
					<StaticRequests>5</StaticRequests>
				</State>
				<Instance index="1" />
				<Associations num_groups="1">
					<Group index="1" max_associations="5" label="Lifeline" auto="true" multiInstance="true">
						<Node id="1" />
						<Node id="1" instance="1" />
					</Group>
				</Associations>
			</CommandClass>
		</CommandClasses>
	</Node>`

magpern, 2020-06-13:
@Fishwaldo please reopen this, it is clearly a very common problem

Fishwaldo, 2020-06-14:
As already commented (https://github.com/OpenZWave/open-zwave/issues/2214#issuecomment-632477279)  someone needs to update the config file and test it, then submit a PR.

marin849, 2020-11-10:
The number of association groups is correct. It is only one group.

The problem is that OpenZWave configures a multi channel lifeline association, using MULTI_CHANNEL_ASSOCIATION_SET(groupId=1, multichannelnodeId=1, endpoint=1).
OpenZwave should have configured an ordinary lifeline association, using ASSOCIATION_SET(groupId=1, nodeId=1).

In Domoticz the Group 1 association for the HEM Gen5 shows up as "1.1" and the sensor does not update.
When connecting the Z-stick to PC and using the software mentioned above to send ASSOCIATION_SET GROUP 01 ID 01 the meter directly starts to send reports. In Domoticz I now have two associations, both "1.1" and "1". But it works.

How should the configuration file zw095.xml be changed so that OZW does not make a multi_channel_association?
There is no command class 142 section in zw095.xml so I don't understand why OZW send a multichannel association.
Googling shows that 
` <Compatibility>
<ForceInstances>false</ForceInstances>
</Compatibility>`
should be entered either under Command Class 133 or 142, but I'm not sure what it means.

DanielBaulig, 2021-01-10:
Just flagging that I had that same problem and it took me multiple days of googling, debugging and talking to Aeotec support to get this fixed using the exact same steps that @marin849 outlined (that I did upon instruction of Aeotec support, because I don;t know any Zwave protocol specifics at all).

magpern, 2021-01-10:
@DanielBaulig Mister Someone needs to send a PR to get this updated. We are waiting for Mister Someone as stated in the comment by @Fishwaldo  on 14 Jun 2020

cobryan05, 2021-01-12:
I just got a ZW095 today, ran into the same issue with not updating.

I'd be happy to send a PR if I knew what to change.

I can fix the issue by editing my ozwcache_xxx.xml:

```
<CommandClass id="142" name="COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION">
  <Compatibility>
    <ForceInstances>false</ForceInstances>
  </Compatibility>
```

but I don't know how to edit the XML files to make this 'permanent'. I tried adding

```
  <CommandClass id="144">
    <Compatibility>
      <ForceInstances>false</ForceInstances>
    </Compatibility>
  </CommandClass>
```
to  zw095.xml, restarting OZW, and removing and re-adding the Energy Meter to my network. No updates, and my Compatibility flags didn't make it into my ozw cache.

If someone lets me know *how* to properly make this change, I'd be happy to test it and send a pull request.

hambudi, 2021-01-18:
I've tested this fix, and though I don't know at all what this change does, I am submitting it for review, because it fixes the issue I'm having. Here is the PR https://github.com/OpenZWave/open-zwave/pull/2546

FYI: It would have helped to keep this issue open, as I would have found this thread much sooner. I've had this device for months and have been beating my head against the wall.