OpenZWave/open-zwave #2215: CT101 Humidity sensor requires polling

aav7fl, 2020-05-11:
## Bug overview

There is an issue where the CT101 (`<Product type="6501" id="c" name="CT101 Thermostat (Iris)">`) requires polling to see any updates provided by the humidity sensor. This bug was _briefly_ mentioned in https://github.com/OpenZWave/open-zwave/issues/1179#issue-214492788, but was overshadowed by the fact that the response object was invalid.

## Documentation

### Manual

On page 23 of the [CT101 Manual.pdf](https://github.com/OpenZWave/open-zwave/files/4611080/CT101.Manual.pdf), it states that ` Encapsulation is the only way to configure the thermostat to send humidity sensor multilevel association reports.`.  

### Configuration

There is also a configuration option to set the `Humidity Reporting Threshold` through configuration option #5 for the device.

| Value 	        | Threshold         	|
|-------         	|-------------------	|
| 0     	        | Function disabled 	|
| 1     	        | 3% RH             	|
| 2  (Default)   	| 5% RH             	|
| 3             	| 10% RH            	|

> The PDF manual is missing the table, but the table values can be sourced from here: http://devel.pepper1.net/zwavedb/device/700

## Troubleshooting

The CT 101 is a troubled device. For the better part of the last year or so, I thought the humidity sensor in my unit was bad. It wasn't until I noticed the value was being updated each time I restarted the network that I figured something was wrong with the device firmware.

In order to resolve this, I dug into the specs and configuration linked above to see if I could make any device corrections that would be breaking the sensor updates.

First I tried changing the configuration for the humidity reporting threshold. I updated and validated changing the humidity reporting threshold from its default value (`2` or 5% RH) to `1` (3% RH). 

After boiling a pot of water nearby and waiting 30 minutes, I found there were no multilevel sensor reports sent from the thermostat over the network. After manually polling against the sensor instance, a report was finally sent. The report showed the the RH% had increased by over 15% from its last known value, but the thermostat never sent a report unsolicited. 

This conclusion led me to believe there is a bug in the firmware that prevents reports from being sent out unless manually polled. Given the issues this device had in #1179, I strongly believe this to be the case. It was also mentioned in the original post of #1179 that the humidity sensor required manual polling as well. 

## Attempted Solution

When I enable polling of the value I get an accurate up-to-date data of the humidity sensor. Otherwise the thermostat _never sends any_ multilevel sensor reports (for the associated nodes). 
While I have manually changed the `poll_intensity` in the value entry in my ozwcache file from:

```xml
<Value type="decimal" genre="user" instance="3" index="5" label="Humidity" units="%" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="35">
	<Help>Humidity Sensor Value</Help>
</Value>
```

to

```xml
<Value type="decimal" genre="user" instance="3" index="5" label="Humidity" units="%" read_only="true" write_only="false" verify_changes="false" poll_intensity="96" min="0" max="0" value="35">
	<Help>Humidity Sensor Value</Help>
</Value>
```

Clearly this isn't supposed to be modified since it will be overwritten, but doing so allowed me to receive updates from my humidity sensor in the unit.

<details>
<summary>
Full _unchanged_ ozcache XML for CT 101
</summary>

```xml
<Node id="3" name="" location="" basic="4" generic="8" specific="6" type="General Thermostat V2" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="3" configrevision="2" query_stage="Complete">
	<Neighbors>1,128,127,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0</Neighbors>
	<Manufacturer id="98" name="2GIG Technologies">
		<Product type="6501" id="c" name="CT101 Thermostat (Iris)">
			<MetaData />
		</Product>
	</Manufacturer>
	<CommandClasses>
[... 222 lines cut ...]
				<CCVersion>1</CCVersion>
				<InNif>true</InNif>
				<StaticRequests>5</StaticRequests>
			</State>
			<Instance index="1" />
			<Value type="byte" genre="user" instance="1" index="0" label="Indicator" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="0">
				<Help>Current Indicator State</Help>
			</Value>
		</CommandClass>
	</CommandClasses>
</Node>
```

</details>

## Logs

- Node logs _without_ polling of the humidity sensor (current behavior). Reports are being "sent" on sensor changes (and founds on initialization)
  - [NodeLogs_NoPoll.txt](https://github.com/OpenZWave/open-zwave/files/4611239/NodeLogs_NoPoll.txt)
- Node logs when manully polling is enabled for humidity sensor (`MultiChannelEncap` is sent multiple times)
  - [NodeLogs_Polling.txt](https://github.com/OpenZWave/open-zwave/files/4611237/NodeLogs_Polling.txt)

## Conclusion

I would like to propose polling in enabled on this device for this sensor. Assuming this is the correct way forward, I would be happy to open a pull request, but I'm not sure if I should be approaching this by adding an entry to the `CT101.xml` file, or if I should be doing something else?

Fishwaldo, 2020-05-22:
1st comment - Polling is Bad. :) - In reality - any device that has passed the Z-Wave Plus Certification should be able to provide "instant" updates via the Lifeline Association Groups (as indicated via the extract you provided above). 

2nd comment - Enabling Polling is exposed to the Application via a API call as its very dependant upon the end user. We do not enable/disable polling via config files as its a "policy" (some users might want it, others dont). If you *need* to use polling, please consult whatever application you are using to enable/disable polling.

3rd - If the Instant Updates are not working, you may need to try enabling ForceInstances in the config file. See [here](https://github.com/OpenZWave/open-zwave/wiki/CommandClass-Compatibility-Flags#multichannelassociation-commandclass) and grep other config files for examples. (after changing the config file, you should call RefreshNodeInfo, or shutdown your app, remove zwcache_*.xml and restart). You should see in Association Group 1 (lifeline) a target node of 1.1 (if forceinstances was successful) instead of "1" (where 1 is your Controller NodeID)

If after trying the ForceInstances and you get 1.1 in the association group (and there are no other groups that might be applicable to the Humidity Reporting) then this is a device bug and needs to be reported to the vendor.

aav7fl, 2020-05-22:
Thanks for your helpful response 🦸! I will follow up with testing your second third comment over the weekend.

aav7fl, 2020-05-23:
I've enabled polling inside `qt-openzwave`. Given the mess off the outdated CT101, I have no hopes for any firmware fix to be rolled out. I'll just continue polling it for now.

Thanks!

closed 2020-05-23 by aav7fl