OpenZWave/open-zwave #2218: doorlock.xml - Missing feature in danalock / polylock control (HW: Danalock V3 BTZE)

upwindanderl, 2020-05-13:
Meanwhile it's possible to unlock the lock without "unlocking" itself. 
More like opening and not twist the motor the last quarter for opening the door. 
This feature is only available while using the danalock or polycontrol app.
I think it would be a big benefit to have this also in ozw.
I guess in  homeassistant its called lock.open instead of lock.unlock (for example)

https://community.home-assistant.io/t/danalock-v3-official-support-for-ha/84778/11?u=upwind_anderl

upwindanderl, 2020-05-14:
Parallel i opened a ticket at danalock.com and few minutes ago i got this answer:

> 

  | Support (Danalock)                                                                                      May 14, 2020, 8:33:04 AM GMT+2                                                Hello xxxxxx, Thank you for your feedback. Many users have already requested this feature via Z-wave, so it is in the roadmap but not until end this year. Best regardsDanalock support
-- | --

So maybe its still not possible to realize this by ozw without a new firmware on the lock?! 
(Of course its possible while using the BT connectivity via app?!) 

I guess I've to close here until the lock is able to do it? @Fishwaldo

Fishwaldo, 2020-05-22:
If its not available via the Z-Wave interface, obviously I can't do anything about this.

closed 2020-05-22 by Fishwaldo