OpenZWave/open-zwave #2219: Wrong metadata for the AEON Labs ZW090-Z-Stick Gen5+ EU.

Jofreathome, 2020-05-14:
The metadata used in the Open-Zwave config file for this device is for the ZW090-Z-Stick Gen5, the one without the plus sign.
https://products.z-wavealliance.org/products/1355

I think the metadata for the Gen5+ version should be this one:
https://products.z-wavealliance.org/products/3753

Can someone please investigate this?

Regards,

Fishwaldo, 2020-05-22:
Unfortuantly Aeotec have used the same Product Type and ID numbers for the old Gen5, new Gen5 and Gen5+ Sticks, so its not possible to differentiate between them. I've raised it with Aeotec and they responded they will not be changing these ID's. So we can't do anything.

closed 2020-05-22 by Fishwaldo

Jofreathome, 2020-05-22:
I suggest we go for the latest meta/config data version of this stick, they probably sold a lot more of them and after all we are now talking about OpenZwave 1.6!

broglea, 2020-10-14:
@Jofreathome did you find a way to override this to enable zwave plus functionality out of this device?

Jofreathome, 2020-10-14:
@broglea no, the Z-stick does the job and I decided to just live with it.