OpenZWave/open-zwave #2222: Aeotec Limited ZW141 Nano Shutter new firmware update v3.01

don66, 2020-05-18:
v 3.01 release of firmware: 
https://products.z-wavealliance.org/products/3742

I am not experienced to update existing config/aeotec/zw141.xml myself, can anyone help?

In short the big difference is new Manufacturer ID change from "86" to "371". (both are listed in Device Database)

As well changes to parameters 20,22 and 85

https://products.z-wavealliance.org/products/3742/configs

don66, 2020-05-19:
<MetaDataItem name="OzwInfoPage">http://www.openzwave.com/device-database/0371:008D:0103</MetaDataItem>
    <MetaDataItem ManufacturerId="0371" id="008D" name="ZWProductPage" type="0003">https://products.z-wavealliance.org/products/3742/MetaDataItem>

**Propose changes to config parameter:**

  
  <CommandClass id="112">
    <Value genre="config" index="20" label="Motor behavior after powering on" size="1" type="list" units="" value="0">
      <Help>Set the motor behavior after powering on</Help>
      <Item label="Stay the same position" value="0" />
      <Item label="Move to 99%" value="1" />
      <Item label="Move to 0%" value="2" />
    </Value>
    <Value genre="config" index="22" label="Motor direction" size="1" type="list" units="" value="2">
      <Help>
        Set to toggle the motor running direction
        0 (Default) Shutter mode / Venetian mode :Out 1= Up moving/ Multilevel switch = 99%, Out 2= Down moving/ Multilevel switch = 0%
        1 Shutter mode : Out 1= Down moving/ Multilevel switch = 0%, Out 2= Up moving/ Multilevel switch = 99%. Venetian mode :Out 1= Up moving/ Multilevel switch = 99%, Out 2= Down moving/ Multilevel switch = 0%
        2 Shutter mode : Out 1= Up moving/ Multilevel switch = 99%, Out 2= Down moving/ Multilevel switch = 0% Venetian mode: Out 1= Down moving/ Multilevel switch = 0%, Out 2= Up moving/ Multilevel switch = 99%
        3 Shutter mode / Venetian mode : Out 1= Down moving/ Multilevel switch = 0%, Out 2= Up moving/ Multilevel switch = 99%
      </Help>
      <Item label="Shutter default Venetian default" value="0" />
      <Item label="Shutter reverse Venetian default" value="1" />
      <Item label="Shutter default Venetian reverse" value="2" />
      <Item label="Shutter reverse Venetian reverse" value="3" />
    </Value>
    <Value genre="config" index="34" label="Blade turn time" size="2" max="32767" min="10" type="decimal" units="0.01 seconds" value="150">
      <Help>Set the blade turn time for Venetian mode. Details can be found in section 4.2 of Advanced information Product Manual.</Help>
    </Value>
    <Value genre="config" index="35" label="Curtain trip time" size="2" max="32767" min="500" type="decimal" units="0.01 seconds" value="15000">
      <Help>Set the run time of the curtain. Details can be found in section 4.2 of Advanced information Product Manual.</Help>
    </Value>
    <Value genre="config" index="36" label="Enter/exit calibration mode" size="1" type="list" units="" value="0" write_only="true">
      <Help>Set to enter/exit calibration mode</Help>
      <Item label="Set: Exit calibration mode/Report: Not in calibration mode" value="0" />
      <Item label="Set: Enter shutter mode calibration/Report: in shutter mode calibration" value="1" />
      <Item label="Set: Enter Venetian mode calibration/Report: in Venetian mode calibration" value="2" />
    </Value>
    <Value genre="config" index="37" label="User confirmation for calibration" size="1" type="list" units="" value="0" write_only="true">
      <Help>Set user confirmation for calibration</Help>
    </Value>
    <Value genre="config" index="38" label="Calibration mode status report" size="1" type="list" units="" value="0" read_only="true">
      <Help>Get calibration mode status report</Help>
      <Item label="Calibration mode complete" value="0" />
      <Item label="Start calibration mode, going to reference point A" value="1" />
      <Item label="Reach reference point A, going to reference point B" value="2" />
      <Item label="Reach reference point B, waiting for blade turn calibration" value="3" />
      <Item label="Going to reference point C" value="4" />
      <Item label="Calibration mode terminated" value="5" />
    </Value>
    <Value genre="config" index="39" label="Curtain mode opeation" size="1" type="list" units="" value="0">
      <Help>Set the curtain mode operation</Help>
      <Item label="Shutter mode" value="0" />
      <Item label="Venetian mode" value="1" />
    </Value>
    <Value genre="config" index="40" label="Repositioning" size="1" type="list" units="" value="1" write_only="true">
      <Help>Set to begin repositioning</Help>
    </Value>
    <Value genre="config" index="41" label="Repositioning status report" size="1" type="list" units="" value="0" read_only="true">
      <Help>Get repositioning status report</Help>
      <Item label="Repositioning complete" value="0" />
      <Item label="In repositioning" value="1" />
    </Value>
    <Value genre="config" index="42" label="Calibration lock" size="1" type="list" units="" value="1">
      <Help>Enable/disable calibration lock</Help>
      <Item label="Disable calibration" value="0" />
      <Item label="Enable calibration" value="1" />
      <Item label="Enable S1/S2 calibration" value="2" />
    </Value>
    <Value genre="config" index="48" label="S1 scene activation" size="1" type="list" units="" value="0">
[... 38 lines cut ...]
      <Item label="Push button" value="3"/>
      <Item label="Automatic identification mode" value="4"/>
    </Value>
    <Value genre="config" index="121" label="Set external switch mode for S2" size="1" type="list" units="" value="0">
      <Help>When the mode is determined, this mode value will not be reset after exclusion</Help>
      <Item label="Unidentified" value="0"/>
      <Item label="2-state switch" value="1"/>
      <Item label="3-way switch" value="2"/>
      <Item label="Push button" value="3"/>
[... cut ...]
      <Item label="Enable network inclusion/exclusion" value="1"/>
      <Item label="Enable reset to factory defaults" value="2"/>
    </Value>	
    <Value genre="config" index="255" label="Reset to factory defaults" size="4" type="list" value="0" write_only="true">
      <Help>Reset to factory defaults</Help>
      <Item label="Reset to factory default setting" value="0"/>
      <Item label="Reset to factory default setting and remove from the z-wave network" value="1431655765"/>
    </Value>
  </CommandClass>

don66, 2020-05-19:
start draft
[zw141.xml.txt](https://github.com/OpenZWave/open-zwave/files/4651074/zw141.xml.txt)

Fishwaldo, 2020-05-22:
As the config database is mainly maintained by the community, please open a Pull Request with the suggested fixes. Please see [here](https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices)

closed 2020-05-22 by Fishwaldo