OpenZWave/open-zwave #2223: Openzwave docker base image

robertsLando, 2020-05-19:
In order to speed up builds of projects that are using OZW, @chrisns has created https://github.com/chrisns/openzwave-docker 

Would it be possibile to add this repo to openzwave org? 

I'm just a memeber of the org so I don't have enought permissions to create a new repo

chrisns, 2020-05-19:
i'm using git submodules and dependabot to stay up to date, would be great to get this added to the org, and I'll PR this repo so i get a webhook on push so can build and also be more aware of tags to build them too?
also would be great to push to https://hub.docker.com/u/openzwave happy to have you create docker token secrets in my repo so I don't have direct access to them

Thanks
Chris

Fishwaldo, 2020-05-22:
1) @robertsLando I've already added you to the OZW org on Docker. Can you not create a repo? 

2) I recommend we just "update" the docker-openzwave repository here on Github. I've sent a invite to @chrisns and added @robertsLando to that repo. 

3) I'm using docker for the ozwdaemon for the HA integration, but due to needing very recent versions of QT, I have to use Debian SID as the base image (Or Ubuntu Focal when [this](https://github.com/moby/moby/issues/40734) issue is fixed one way or another)

4) As buildx takes forever (and often times out on github actions for some of my builds) I can do builds on my hardware at home (native arm64, armhf, and amd64) after the CI tests for OZW passes (so this addresses any issues with regards to commits that break the core library rather than just going with a random commit if your pulling direct from the repo and not the validated builds)

when 4 completes we can push the images to a new repo on Docker Hub. 

Ok?

chrisns, 2020-05-22:
hey @Fishwaldo !
2. by update are you implying i takeover the repo and push basically what I built there (making sure I don't miss any features)? (want to be sure before I trample someone elses work) it's currently marked as a fork though which might be misleading, can we maybe rename+archive and make a new repo?

3. very happy to add other base images if it helps to keep things in one place?

4. agree, buildx is slow and github aren't _that_ generous with the compute resource they give actions. if you've got hardware we could maybe add it as a self hosted github runner, i can probably allocate some compute to that too.

+5 Am I ok to PR the open-zwave repo to get a github action to ping the docker one on push so i can build sha's and tags better rather than nightly polling?

+6 Spoken a bit to @robertsLando, wondered if there'd be interest in setting up slack/gitter with semi-regular catchups maybe google meet or whatever (beers optional depending on timezones) so we can discuss visions and be a bit more collaborative across the spread of repos, I'm happy to organise if you think this would be a good idea?

related to ^ @robertsLando I think mentioned in an email to you about testing, your mailout said about you building a test rig, i'd been considering doing the same, would be great to join forces on that and wire it up for shared use rather than us both reinvent a wheel

robertsLando, 2020-05-22:
Everything is ok for me! 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>