OpenZWave/open-zwave #2233: Leviton DZ15S switch config has metadata for DZPA1

kaijk, 2020-05-28:
The DZ15S at: [https://github.com/OpenZWave/open-zwave/blob/master/config/leviton/dz15s.xml](url)
Product ID: 3401, includes the metadata description and some Product ID references for Product ID: 3601 which is the DZPA1 switch [https://products.z-wavealliance.org/products/1959?selectedFrequencyId=2](url)

Thus, the image, manual, description, etc., for the DZ15S are linked incorrectly to the DZPA1 metadata.
I suggest: 
1) Cloning the DZ15S to create the DZPA1 config by changing the Description in lines 2-15 of the config file to the DZPA1 and product ID 3601
2) Fix the DZ15S config by changing the metadata links and references and verbiage from DZPA1 3601 to DZ15S 3401 at various places in lines 16-51

Note: I have both devices and can test their operation when appropriate.

kaijk, 2020-05-29:
@Fishwaldo It appears that you may have started cloning the leviton DZ15S wall switch config to the DZPA1 plug-in switch, but saved the changes back to the DZ15S config before making all the changes from its product id 3401 to the DZPA1's 3601 ID.  The image/dz15s.png image is actually the dzpa1.png image.  

The device database shows two dz15s config entries, one for 3401 and one for 3601.

Notwithstanding my Note above, I don't have a DZPA1. I have the VRCOP which looks the same, but is a serial interface.

I don't have push rights to upload the real dz15s.png image, so its uploaded here.
[image]

Fishwaldo, 2020-06-08:
If you create a fork, make the changes on your local copy and submit a PR back to us we will merge it. 

(sorry, the database is maintained by the community mainly - I don't have any of these devices so I rely on the community to help me maintain it)

closed 2020-06-22 by no-response[bot]