OpenZWave/open-zwave #2237: Yale Keyfree Connected Smart Lock config is messed up as a result of bad commits

nickperry, 2020-06-06:
The data for the 0006 Keyfree lock was replaced with an import of the data for the 0007 Keyless lock.

See my comment on the bad commit here - 

https://github.com/OpenZWave/open-zwave/commit/6a0fdfdee98d589fa0e4d366e9d49b964020db9f#commitcomment-39715112

Additionally, the description for the Keyfree 0006 product type was changed to the wrong string. See my commit on the offending commit here -

https://github.com/OpenZWave/open-zwave/commit/daedc9f9d13fb56e1fd9be54f886eeac0c0e9cc6#commitcomment-39715310

@Fishwaldo @nechry

nickperry, 2020-06-06:
I have noticed https://github.com/OpenZWave/open-zwave/blob/master/config/assa_abloy/KeyfreeConnected-plus.xml is broken too. Similar problem - it's got Keyless 0007 metadata imported instead of Keyfree 0006 metadata.

nickperry, 2020-06-06:
config/assa_abloy/KeyfreeConnected.xml content should be:
`<Product Revision="2" xmlns="https://github.com/OpenZWave/open-zwave">
  <MetaData>
    <MetaDataItem name="OzwInfoPage">http://www.openzwave.com/device-database/0129:0000:0006</MetaDataItem>
    <MetaDataItem name="ProductPic">images/assa_abloy/KeyfreeConnected.png</MetaDataItem>
    <MetaDataItem id="0000" name="ZWProductPage" type="0006">https://products.z-wavealliance.org/products/1358/</MetaDataItem>
    <MetaDataItem name="Name">Yale Keyfree Connected</MetaDataItem>
    <MetaDataItem name="Description">Keyfree Connected with Z-Wave for multipoint lock</MetaDataItem>
    <MetaDataItem id="0000" name="Identifier" type="0006">YKFCON</MetaDataItem>
    <MetaDataItem id="0000" name="FrequencyName" type="0006">CEPT (Europe) (EU)</MetaDataItem>
    <ChangeLog>
      <Entry author="Nick Perry - nick@nickperry.co.uk" date="06 June 2020" revision="3">Updating from the correct entry in the Z-Wave Alliance Database - https://products.z-wavealliance.org/Products/1358/XML</Entry>
    </ChangeLog>
  </MetaData>
  <CommandClass id="113">
    
    <TriggerRefreshValue Genre="user" Index="0" Instance="1">
      <RefreshClassValue CommandClass="98" Index="1" Instance="1" RequestFlags="0"/>
    </TriggerRefreshValue>
  </CommandClass>
</Product>`

Fishwaldo, 2020-06-08:
Please submit a PR to fix this - as the database is primarly maintained by the community (and I don't own any of these devices) there is no way for me to test these types of changes and I have to rely on the community here.

closed 2020-06-08 by Fishwaldo

nickperry, 2020-06-08:
But I am concerned about breaking other people's devices who are now relying on the erroneous config.

I would appreciate some help from you @Fishwaldo in
a) how to update the png image correctly.
b) how is www.openzwave.com/device-database updated?
c) Where do you want to put the config that you put in KeyfreeConnected.xml, when the contents of KeyfreeConnected.xml is replaced with the correct config?