OpenZWave/open-zwave #2248: Check why TriggerRefreshValue is appearing twice in the cache

Fishwaldo, 2020-06-11:
https://github.com/OpenZWave/open-zwave/pull/2247#issuecomment-642856516

mike240se, 2020-06-24:
This bug also causes the parent topic under /instance/1/ to disappear for all the devices on ozw daemon restart.  that then causes HA to make the entities unavailable.

removing the duplicate from the cache fixes the issue as a work around.

steve28, 2020-06-24:
I am seeing this in my cache file as well. I also have issues where the cache file seems to be missing chunks on some ozwd restarts:
```xml
<TriggerRefreshValue Genre="user" Instance="1" Index="6">
        <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
</TriggerRefreshValue>
<TriggerRefreshValue Genre="user" Instance="1" Index="6">
        <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
</TriggerRefreshValue>
```

Fishwaldo, 2020-07-02:
Fixed

closed 2020-07-02 by Fishwaldo