OpenZWave/open-zwave #2249: 2Gig ct200x / vivint element thremostat not updating operating state

srirams, 2020-06-12:
I have a Radio Thermostat Company of America CT200x aka 2gig ct200x aka Vivint Elements thermostat. Some information is found [here](https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/938). I'm getting temperature updates, but the operating state and fan state don't update. According to the [docs](https://www.cd-jackson.com/zwave_device_uploads/938/CT200-UserGuide-20160308.pdf):

> The following commands can be sent to lifeline:
• Command Class Sensor Multilevel - Sensor Multilevel Report
• Command Class Basic - Basic Report
• Command Class Thermostat Mode - Thermostat Mode Report
• Command Class Thermostat Operating State - Thermostat Operat-
ing State Report
• Command Class Thermostat Fan Mode - Thermostat Fan Mode
Report
• Command Class Thermostat Fan State - Thermostat Fan State
Report
• Command Class Thermostat Setpoint - Thermostat Setpoint Report
• Command Class Multichannel - Multi Channel Command Encap-
sulate
These can be accessed via an Association Group Info Command List
Get for group 1 (lifeline)

So it should work, I believe. I've tried it with this [config](https://pastebin.com/rWqPMuPd) as well as no config.

[Log file](https://controlc.com/5cc005ab)

Fishwaldo, 2020-06-24:
>[20200611 19:44:15.654 CDT] [ozw.library] [info]: Info - Node: 10 Received MULTI_CHANNEL_ASSOCIATION_REPORT from node 10, group 1
[20200611 19:44:15.654 CDT] [ozw.library] [info]: Info - Node: 10   The group contains:
[20200611 19:44:15.654 CDT] [ozw.library] [info]: Info - Node: 10     Node 1
[20200611 19:44:15.654 CDT] [ozw.library] [info]: Info - Node: 10 Querying associations for node 10 is complete.

>[20200611 19:44:21.617 CDT] [ozw.library] [info]: Info - Node: 11 Received MULTI_CHANNEL_ASSOCIATION_REPORT from node 11, group 1
[20200611 19:44:21.617 CDT] [ozw.library] [info]: Info - Node: 11   The group contains:
[20200611 19:44:21.617 CDT] [ozw.library] [info]: Info - Node: 11     Node 1
[20200611 19:44:21.618 CDT] [ozw.library] [info]: Info - Node: 11 Querying associations for node 11 is complete.

So the associations are setup correctly - But I can't actually see any "updates" after OZW has completed the startup. I can't see anything "wrong" that OZW is doing regarding setting the associations, although you might want to enable AltTypeIntepretation on this device as its using the alternative encoding:
https://github.com/OpenZWave/open-zwave/wiki/CommandClass-Compatibility-Flags#thermostatsetpoint-commandclass

srirams, 2020-06-27:
I've made the suggested change, and ran it for a few days. The operating state and fan state still don't update reliably. While searching around (I've lost the link now) it may be the device is running in some sort of compatibility mode if the first request is not a multi channel association? 

I've include a log file with a longer run time (please let me know if I should filter it for some particular lines)

[ozwlog.zip](https://github.com/OpenZWave/open-zwave/files/4839919/ozwlog.zip)

kpine, 2020-08-04:
What happens if you change the thermostat mode or setpoints on the the device itself? I have a CT32 and experience the same behavior. When I send z-wave commands to change thermostat mode or setpoints, the operating state is never updated. However, if the mode and setpoints are changed on the device itself, it updates instantly every time. All the messages to and from the thermostat are multi-channel encapsulated, AFAIK.

Sadly (and I hope to be wrong) this is apparently a problem with several Radio Thermostat models. The CT32 manual also has the same verbiage about the operating state updates.  For example, here's a SmartThings [forum post](https://community.smartthings.com/t/thermostat-differences-ct100-ct100-ct101-gocontrol/96373) about the CT101 in which users describe the same behavior. 

You can see that SmartThings [device handler](https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/94885a1a066a19debe9117d21f386f20f1080c9c/devicetypes/smartthings/ct100-thermostat.src/ct100-thermostat.groovy#L279) for the CT100 manually refreshes the operating state and set points, so it is a known problem and sounds like it applies to multiple CT-XXX models.

I was starting to look at using the `TriggerRefresh` compatibility flag, but I'm not sure it's going to work quite right. I only want it to trigger on solicited messages (AKA Gets from OZW). For example, if OZW does a Set Thermostat Mode followed by a Get Thermostat Mode, I'd want that Get to trigger a refresh of the operating state. I haven't experimented enough to know if that's true or not, but I don't think it is based on the usage for locks. I think OZW will trigger on any message marked for refresh.

srirams, 2020-08-04:
When I change the setpoint on the thermostat, the operating state updates, exactly like you describe.

So either these devices simply don't send out operating state updates without an associated temperature / mode change or they are operating in a different mode with ozw / smartthings (and I guess the only way to check would be to compare the behavior with the official hub).

Fishwaldo, 2020-08-05:
If the device doesn't send the operating state, then you are going to have send a refreshvalue command for the operating state valueid. This should be handled via the App somehow (a automation?)

Looking at the smartthings handler above, its doing essentially the same thing

Fishwaldo, 2020-08-05:
Closing this as its not really a OZW bug in that case.

closed 2020-08-05 by Fishwaldo