OpenZWave/open-zwave #2253: 2 problems concerning make install

dgrostoto, 2020-06-14:
I use make for userland installs  and i noticed two problems :
1. the make install does not copy the static library to destination (.a file) it only deals the dynamic library (.so file)
2. the destination dir for .pc file does not take into account the PREFIX variables (it tries to to go in /usr/local which requires special privileges). Usually (for the vast majority of builds i've done)  the pc file comes into $(PREFIX)/lib/pkgconfig directory.

thanks in advance.

Fishwaldo, 2020-06-24:
1. Very few libraries do install archive files... only dynamic librarys
2. pc files are installed to the path that pkg-config says its database is in. (pkg-config --variable pc_path pkg-config) which might be outside of the prefix. You can override it by setting pkgconfigdir variable passed to Make.

closed 2020-06-24 by Fishwaldo

dgrostoto, 2020-06-24:
OK for the pkgconfigdir variable even if it is not the standard way to do it.
Concerning the .a files not installed, the argument "very few libraries" (which does not take into account the programs like domoticz for example that are not libraries they do use static libs by default) is a really bad reason to not do it.

gizmocuz, 2020-07-01:
When we can specify to build also a static library (or when it is done by default), it would be nice if this was also installed

For example when you build the boost libraries, the static libraries are installed in /usr/local/lib

/usr/local/lib/libboost_system.a
/usr/local/lib/libboost_thread.a

Currently, if one compiles openzwave from folder 'a', and a application that would like to use the static library from folder 'b', it is currently for 'b' not possible to find the static library

On debian when installing most '-dev' libraries (via apt) they do install a static library (.a) version as well

Would be great if this could be installed