OpenZWave/open-zwave #2259: Additional Kwikset Lock Types

peng1can, 2020-06-17:
My Kwikset 912 Lock was not recognized.  I have added it's information, but while I was at it, I added entries for other known Kwikset Lock variants if it would be helpful to add them while you're at it.  The new Kwikset block looks like this:

  <Manufacturer id="0090" name="Kwikset (Spectrum Brands)">
    <Product config="kwikset/smartcode.xml" id="0001" name="Touchpad Electronic Deadbolt" type="0001"/>
    <Product config="kwikset/smartcode.xml" id="0642" name="SmartCode 916" type="0003"/>
    <Product config="kwikset/smartcode.xml" id="0642" name="SmartCode 916" type="0001"/>
    <Product config="kwikset/smartcode.xml" id="0440" name="SmartCode 914" type="0003"/>
    <Product config="kwikset/smartcode.xml" id="4006" name="SmartCode 914" type="0003"/>
    <Product config="kwikset/smartcode.xml" id="0236" name="SmartCode 912" type="0003"/> 
    <Product config="kwikset/smartcode.xml" id="0339" name="SmartCode 912" type="0003"/>
    <Product config="kwikset/smartcode.xml" id="0236" name="SmartCode 910" type="0001"/>
    <Product config="kwikset/smartcode.xml" id="0238" name="SmartCode 910" type="0003"/>
    <Product config="kwikset/smartcode.xml" id="0001" name="SmartCode 910" type="0001"/>
    <Product config="kwikset/smartcode.xml" id="0541" name="SmartCode 888" type="0003"/>
    <Product config="kwikset/smartcode.xml" id="0742" name="Obsidian" type="0003"/>
    <Product config="kwikset/smartcode.xml" id="0440" name="SmartCode 10" type="0006"/>
    <Product config="kwikset/914c.xml" id="0446" name="Convert 914" type="0003"/>
  </Manufacturer>

Thanks

firstof9, 2020-06-19:
Probably best to submit a PR with the modified file.

mike240se, 2020-06-19:
I have been trying to get the 888 lock added and no matter what i do openzwave crashes when i refreshnode after creating 888.xml and attaching it to the device id in manufcaturer_specific.xml.   All I have is 1 command class to fix the alarm event instead of door lock event when manually locking/unlocking.  

I then removed all the command classes and just have a blank 888.xml with just the meta data and the proper structure and it still crashes every time.  Problem is nothing helpful in the crash dump at all its just 1 line of sigwit.  

not really sure what is wrong.

Fishwaldo, 2020-06-24:
Thanks - I added all except the 888 as I know @mike240se is working on that one.

closed 2020-06-24 by Fishwaldo

mike240se, 2020-06-24:
this is causing an issue for someone with a 916 in the HA #zwave discord.  I was going to mention that while very similar each of these locks have slightly different firmware and a lot of the config options in smartcode.xml dont work for some of the models.  for example my 888 has none of those config options at all.   The person with the 916 is having the same issue I am where the instance/1 parent topics disappear on restart of the ozwdaemon and requires deleting the ozwcache before each restart.

perhaps related, all these trigger refresh show twice in the ozwcache file:

https://i.imgur.com/VMGZiuX.png