OpenZWave/open-zwave #2262: Kwikset Lock status not updating

firstof9, 2020-06-19:
The lock status doesn't update when manually locking/unlocking the lock.

I attempted the adjustment from the 914c.xml:
```
                        <CommandClass id="113" name="COMMAND_CLASS_NOTIFICATION">
                                <Compatibility />
                                <State>
                                        <CCVersion>1</CCVersion>
                                        <Encrypted>true</Encrypted>
                                        <StaticRequests>2</StaticRequests>
                                </State>
                                <Instance index="1" />
                                <Value type="byte" genre="user" instance="1" index="512" label="Alarm Type" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="21">
                                        <Help>Alarm Type Received</Help>
                                </Value>
                                <Value type="byte" genre="user" instance="1" index="513" label="Alarm Level" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="1">
                                        <Help>Alarm Level Received</Help>
                                </Value>
                                <Value type="int" genre="config" instance="1" index="514" label="Automatically Clear Events" units="ms" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="-2147483648" max$
                                        <Help>Automatically Clear Events sent by a device after a number of Milliseconds</Help>
                                </Value>
                                <TriggerRefreshValue Genre="user" Instance="1" Index="6">
                                        <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
                                </TriggerRefreshValue>
                        </CommandClass>
```

Something tells me ```Index="6"``` isn't right.

firstof9, 2020-06-19:
As requested the log from an unlock / lock.

```
[20200619 15:55:28.480 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x0e, 0x02, 0x98, 0x40, 0x27
[20200619 15:55:28.486 UTC] [ozw.library] [info]: Info - Node: 14 Received SecurityCmd_NonceGet from node 14
[20200619 15:55:28.491 UTC] [ozw.library] [info]: Info - Node: 0 NONCES: 0x16, 0xc7, 0xc9, 0xaf, 0xe9, 0xde, 0x10, 0x0c
[20200619 15:55:28.497 UTC] [ozw.library] [info]: Info - Node: 0 NONCES: 0x84, 0x60, 0x80, 0xb1, 0x04, 0xa6, 0x9e, 0xb1
[20200619 15:55:28.505 UTC] [ozw.library] [info]: Info - Node: 0 NONCES: 0x85, 0xe5, 0x3d, 0xac, 0xb9, 0x76, 0xc9, 0x2d
[20200619 15:55:28.513 UTC] [ozw.library] [info]: Info - Node: 0 NONCES: 0x37, 0x62, 0xc7, 0x21, 0x03, 0x25, 0xbe, 0x87
[20200619 15:55:28.522 UTC] [ozw.library] [info]: Info - Node: 0 NONCES: 0xbd, 0x7a, 0x17, 0x76, 0xf0, 0xe1, 0xa3, 0x44
[20200619 15:55:28.532 UTC] [ozw.library] [info]: Info - Node: 0 NONCES: 0x53, 0x61, 0xde, 0x24, 0xda, 0x63, 0xee, 0xfb
[20200619 15:55:28.541 UTC] [ozw.library] [info]: Info - Node: 0 NONCES: 0x75, 0x6f, 0xa7, 0x52, 0x99, 0x21, 0x14, 0x60
[20200619 15:55:28.551 UTC] [ozw.library] [info]: Info - Node: 0 NONCES: 0x50, 0xf1, 0xc4, 0x1f, 0x09, 0x83, 0x66, 0x36
[20200619 15:55:28.560 UTC] [ozw.library] [info]: Info - Node: 14 Sending (Send) message (Callback ID=0x01, Expected Reply=0x00) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x0e, 0x0a, 0x98, 0x80, 0x53, 0x61, 0xde, 0x24, 0xda, 0x63, 0xee, 0xfb, 0x05, 0x01, 0x81:
[20200619 15:55:28.577 UTC] [ozw.library] [debug]: Detail - Node: 0   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
[20200619 15:55:28.587 UTC] [ozw.library] [debug]: Detail - Node: 0   ZW_SEND_DATA delivered to Z-Wave stack
[20200619 15:55:28.597 UTC] [ozw.library] [debug]: Detail - Node: 0   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x03, 0xe9
[20200619 15:55:28.604 UTC] [ozw.library] [debug]: Detail - Node: 0   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
[20200619 15:55:28.621 UTC] [ozw.library] [debug]: Detail - Node: 14   Received: 0x01, 0x1e, 0x00, 0x04, 0x00, 0x0e, 0x18, 0x98, 0x81, 0xd8, 0xc8, 0x6d, 0xf0, 0x74, 0x56, 0x83, 0x96, 0xb0, 0xff, 0x7b, 0x8b, 0x17, 0x53, 0x88, 0x6c, 0x79, 0xd8, 0xf8, 0x65, 0x02, 0xed, 0x9c
[20200619 15:55:28.627 UTC] [ozw.library] [info]: Info - Node: 0 Raw: 0x98, 0x81, 0xd8, 0xc8, 0x6d, 0xf0, 0x74, 0x56, 0x83, 0x96, 0xb0, 0xff, 0x7b, 0x8b, 0x17, 0x53, 0x88, 0x6c, 0x79, 0xd8, 0xf8, 0x65, 0x02, 0xed, 0x9c
[20200619 15:55:28.633 UTC] [ozw.library] [debug]: Detail - Node: 14 Decrypted Packet: 0x00, 0x71, 0x05, 0x16, 0x01
[20200619 15:55:28.639 UTC] [ozw.library] [info]: Info - Node: 14 Got a AlarmCmd_Report Message....
[20200619 15:55:28.645 UTC] [ozw.library] [info]: Info - Node: 14 Received Alarm report: type=22, level=1
[20200619 15:55:28.652 UTC] [ozw.library] [debug]: Detail - Node: 14 Refreshed Value: old value=18, new value=22, type=byte
[20200619 15:55:28.658 UTC] [ozw.library] [debug]: Detail - Node: 14 Changes to this value are not verified
[20200619 15:55:28.664 UTC] [ozw.library] [debug]: Detail - Node: 14 Refreshed Value: old value=1, new value=1, type=byte
[20200619 15:55:28.670 UTC] [ozw.library] [debug]: Detail - Node: 14 Changes to this value are not verified
[20200619 15:55:28.676 UTC] [ozw.library] [debug]: Detail - Node: 14 Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 512
[20200619 15:55:28.682 UTC] [ozw.library] [debug]: Detail - Node: 14 Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 513
[20200619 15:55:28.683 UTC] [ozw.notifications] [debug]: Notification pvt_valueChanged:  144115188316782609 Thread:  0x76f0a410
[20200619 15:55:28.688 UTC] [ozw.mqtt.publisher] [debug]: Publishing Event valueChanged: 144115188316782609
[20200619 15:55:28.690 UTC] [ozw.notifications] [debug]: Notification pvt_valueChanged:  144396663293493265 Thread:  0x76f0a410
[20200619 15:55:29.941 UTC] [default] [debug]: QVariant(int, 1)
[20200619 15:55:29.945 UTC] [default] [debug]: QVariant(int, 2)
[20200619 15:55:29.950 UTC] [default] [debug]: QVariant(int, 6)
[20200619 15:55:29.954 UTC] [default] [debug]: QVariant(int, 8)
[20200619 15:55:29.957 UTC] [default] [debug]: QVariant(int, 9)
[20200619 15:55:29.960 UTC] [default] [debug]: QVariant(int, 10)
[20200619 15:55:29.965 UTC] [default] [debug]: QVariant(int, 11)
[20200619 15:55:29.969 UTC] [default] [debug]: QVariant(int, 12)
[20200619 15:55:29.973 UTC] [default] [debug]: QVariant(int, 14)
[20200619 15:55:29.977 UTC] [default] [debug]: QVariant(int, 15)
[20200619 15:55:29.980 UTC] [default] [debug]: QVariant(int, 16)
[20200619 15:55:29.982 UTC] [default] [debug]: QVariant(int, 17)
[20200619 15:55:29.984 UTC] [default] [debug]: QVariant(int, 18)
[20200619 15:55:29.987 UTC] [default] [debug]: QVariant(int, 19)
[20200619 15:55:29.989 UTC] [default] [debug]: QVariant(int, 20)
[20200619 15:55:29.991 UTC] [default] [debug]: QVariant(int, 21)
[20200619 15:55:29.993 UTC] [default] [debug]: QVariant(int, 22)
[20200619 15:55:29.995 UTC] [default] [debug]: QVariant(int, 23)
[20200619 15:55:29.997 UTC] [default] [debug]: QVariant(int, 24)
[20200619 15:55:29.999 UTC] [default] [debug]: QVariant(int, 26)
[20200619 15:55:30.001 UTC] [default] [debug]: QVariant(int, 27)
[20200619 15:55:30.003 UTC] [default] [debug]: QVariant(int, 28)
[20200619 15:55:30.005 UTC] [default] [debug]: QVariant(int, 29)
[... cut ...]
[20200619 15:55:33.216 UTC] [ozw.library] [debug]: Detail - Node: 24 Notification: ValueChanged CC: COMMAND_CLASS_METER Instance: 2 Index: 2
[20200619 15:55:33.225 UTC] [ozw.library] [debug]: Detail - Node: 24 Notification: ValueChanged CC: COMMAND_CLASS_METER Instance: 2 Index: 256
[20200619 15:55:33.225 UTC] [ozw.notifications] [debug]: Notification pvt_valueChanged:  562950361088034 Thread:  0x76f0a410
[20200619 15:55:33.237 UTC] [ozw.notifications] [debug]: Notification pvt_valueChanged:  72057594445594656 Thread:  0x76f0a410
```

rtrox, 2020-06-20:
Not a maintainer, but glad to be. I'm troubleshooting the same thing, but with a Kwikset Obsidian. Looking at your logs here, they look similar to mine -- It doesn't look like Instance 1 Index 6 is updated on your device when you lock the door manually. I'm seeing this here:
```
[20200619 15:55:30.698 UTC] [ozw.library] [info]: Info - Node: 14 Received Alarm report: type=21, level=1
[20200619 15:55:30.704 UTC] [ozw.library] [debug]: Detail - Node: 14 Refreshed Value: old value=22, new value=21, type=byte
[20200619 15:55:30.709 UTC] [ozw.library] [debug]: Detail - Node: 14 Changes to this value are not verified
[20200619 15:55:30.714 UTC] [ozw.library] [debug]: Detail - Node: 14 Refreshed Value: old value=1, new value=1, type=byte
[20200619 15:55:30.722 UTC] [ozw.library] [debug]: Detail - Node: 14 Changes to this value are not verified
[20200619 15:55:30.730 UTC] [ozw.library] [debug]: Detail - Node: 14 Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 512
[20200619 15:55:30.739 UTC] [ozw.library] [debug]: Detail - Node: 14 Notification: ValueChanged CC: COMMAND_CLASS_NOTIFICATION Instance: 1 Index: 513
```
It *does* look like Index 512 and 513 notify. Hopefully maintainers can verify I'm right here, but if so, you may end up wanting to watch #2264, as I tried to update the TriggerRefreshValue Index to 512, and OpenZwave seems to revert that value to zero.

firstof9, 2020-06-20:
Yes it seems we're in the same boat. If I manually change the value in the cache file to 512 and restart the container, the whole COMMAND_CLASS_NOTIFICATION CC vanishes from ALL devices.

firstof9, 2020-06-20:
I set my index to 513 in the ozw cache XML and it started working.
```
                        <CommandClass id="113" name="COMMAND_CLASS_NOTIFICATION">
                                <Compatibility />
                                <State>
                                        <CCVersion>1</CCVersion>
                                        <Encrypted>true</Encrypted>
                                        <StaticRequests>2</StaticRequests>
                                </State>
                                <Instance index="1" />
                                <Value type="byte" genre="user" instance="1" index="512" label="Alarm Type" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="21">
                                        <Help>Alarm Type Received</Help>
                                </Value>
                                <Value type="byte" genre="user" instance="1" index="513" label="Alarm Level" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="255" value="1">
                                        <Help>Alarm Level Received</Help>
                                </Value>
                                <Value type="int" genre="config" instance="1" index="514" label="Automatically Clear Events" units="ms" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="-2147483648" max$
                                        <Help>Automatically Clear Events sent by a device after a number of Milliseconds</Help>
                                </Value>
                                <TriggerRefreshValue Genre="user" Instance="1" Index="513">
                                        <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
                                </TriggerRefreshValue>
                        </CommandClass>
```

firstof9, 2020-06-23:
Just for record this is a 910 SmartCode Traditional Electronic Deadbolt

Fishwaldo, 2020-06-24:
> Yes it seems we're in the same boat. If I manually change the value in the cache file to 512 and restart the container, the whole COMMAND_CLASS_NOTIFICATION CC vanishes from ALL devices.

Can you post a log of refreshnodeinfo when you set it to 512 and everything goes bye bye?

firstof9, 2020-06-24:
Looks like it didn't blow up as bad in 1149, but some nodes wouldn't expose to Home Assistant. I've attached the whole startup/shutdown log after modifying the value. Figured that'd be helpful.

[ozwdaemon.log](https://github.com/OpenZWave/open-zwave/files/4826615/ozwdaemon.log)

mike240se, 2020-06-24:
I am having the same issue with my 888.xml.  When restarting the daemon the parent topics under instance/1/ dissapear causing the entities to disappear from HA.  I then clear the ozwcache, restart the daemon and everything comes back and works perfect again..... until I restart ozwdaemon docker again, then same issue happens, the parent topics under instance/1/ for each device disappear again and i must delete the ozwcache and restart the daemon.

what is odd is that everything works perfectly until the daemon is restarted.

mike240se, 2020-06-24:
the issue seems to be that this code is duplicated in the ozwcache.  removing the duplicate from ozwcache fixes the bug and now my instance/1/ parent does not disappear on restart of the daemon

i had this:

<TriggerRefreshValue Genre="user" Instance="1" Index="6">
                                        <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
                                </TriggerRefreshValue>
<TriggerRefreshValue Genre="user" Instance="1" Index="6">
                                        <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
                                </TriggerRefreshValue>

firstof9, 2020-07-05:
This seems, at least for the 910 that having the index set to 512/513 resolves the issue @Fishwaldo, would you like me to submit a PR to adjust the XML?

Fishwaldo, 2020-07-06:
Yes Please.

The Duplicate TriggerRefresh Should be fixed btw.

closed 2020-07-06 by Fishwaldo