OpenZWave/open-zwave #2264: [bug] TriggerRefreshValue reverts Index to 0 when set to 512

rtrox, 2020-06-19:
Background, I'm working to solve a similar to #292 in my setup- Manually locking or unlocking does not trigger a DoorLockReport, and therefore the lockstatus is not updated at the controller.

 I have a Kwikset Obsidian 954. I've built a configuration file for it, which I've attached in it's entirety here. I was able to determine that Alarm Type (3-113-1-512) changes whenever the lock is manually moved. Because of this, I added this TriggerRefreshValue block to my config:
```xml
<TriggerRefreshValue Genre="user" Instance="1" Index="512">
  <RefreshClassValue CommandClass="98" Index="1" Instance="1" RequestFlags="0"/>
</TriggerRefreshValue>    
```

However, after clearing my cache, restarting Zwave2Mqtt, and then triggering a RefreshNodeInfo, what appears in ozwcache is:
```xml
<TriggerRefreshValue Genre="user" Instance="1" Index="0">
  <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
</TriggerRefreshValue>
<TriggerRefreshValue Genre="user" Instance="1" Index="0">
  <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
</TriggerRefreshValue> 
```

To test if I'd configured this incorrectly, I tested changing the `Index` to 6 (knowing this value isn't updated on my lock), and this was correctly reflected in the ozwcache after the same operations. I'm guessing something is going wrong that's causing this to revert, but I haven't been able to determine exactly what it is. 

I don't think the duplication is related or problematic, though it's probably also a bug. That duplication happened with the test using `Index` 6 as well.

Any help on why Index is reset to zero when set to 512, but is parsed correctly when set to 6?

Thanks!

[ozwcache_0xf3aeea14.xml.txt](https://github.com/OpenZWave/open-zwave/files/4807038/ozwcache_0xf3aeea14.xml.txt)
[obsidian.xml.txt](https://github.com/OpenZWave/open-zwave/files/4807039/obsidian.xml.txt)
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/4807040/OZW_Log.txt)

rtrox, 2020-06-20:
Interestingly, when I updated the `Index` to 513, this is what appears in ozwcache:
```
<TriggerRefreshValue Genre="user" Instance="1" Index="1">
  <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
</TriggerRefreshValue>
<TriggerRefreshValue Genre="user" Instance="1" Index="1">
  <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
</TriggerRefreshValue>
```
It looks like the int wraps after 511?

firstof9, 2020-06-20:
Think I might have found the offender:

https://github.com/OpenZWave/open-zwave/blob/master/cpp/src/command_classes/CommandClass.cpp#L304

rtrox, 2020-06-20:
@firstof9 glad you found a workaround for you, but editing the ozwcache directly feels a bit fragile to me. I also have a workaround using node-red, but would prefer to eventually rely on the device config once this bug is resolved.

firstof9, 2020-06-20:
Yes I agree having the actual lock status update is preferred, rather than a node-red or other work arounds.

Fishwaldo, 2020-06-22:
@rtrox firstly - Please update - Your running a old version of ozw and you have a ton of problems with incorrect paths (that I believe has been fixed in later releases of z2m)

>2020-06-19 16:45:44.792 Always, OpenZwave Version 1.6-974-ge89d2a16-dirty Starting Up
2020-06-19 16:45:44.793 Warning, Unable to load Localization file /data/openzwave/Localization.xml: Failed to open file
2020-06-19 16:45:44.793 Always, Using Language Localization 
2020-06-19 16:45:44.793 Warning, Unable to load NotificationCCTypes file /data/openzwave/NotificationCCTypes.xml
2020-06-19 16:45:44.793 Warning, Unable to load SensorMultiLevelCCTypes file /data/openzwave/SensorMultiLevelCCTypes.xml

Then please run another test with log files and post them here.

rtrox, 2020-06-23:
@Fishwaldo - I took a look this evening, it looks like this is the latest version in the alpine linux repos:
https://pkgs.alpinelinux.org/packages?name=openzwave&branch=edge
https://pkgs.alpinelinux.org/packages?name=openzwave-dev&branch=edge

Looks like the package isn't officially maintained though, so I guess I'm not surprised it hasn't been updated since 2019. I'll take a look at rebuilding from source this week so I can re-test. 

(I know you don't maintain either of these other packages, sharing for others who stumble here) It'll take me a day or two, right now I'm running the hassio zwave2mqtt add-on, which just uses the open-zwave version in the alpine repos. I took a look at the official zwave2mqtt image, but it looks like it's actually pegged at OpenZwave 1.4.1:
https://hub.docker.com/r/robertslando/zwave2mqtt/dockerfile

My apologies, I should have checked how old the version being used by  this package was before opening an issue :-/. Once I have time to build a new docker image and try this again, I'll update here.

rtrox, 2020-06-23:
Updated in the issue linked here. I've built a new version of the add-on image using the latest stable of OpenZWave, but have done zero testing so far. It's getting a bit late here in PST, will pick this back up tomorrow to test using the newer release. If anyone is interested, the dirty Dockerfile is available over there.

rtrox, 2020-06-23:
@Fishwaldo  - thanks for the pointer that the openzwave version in the hassio z2m addon was out of date. I was able to test my local docker image for it this morning, and confirmed this now behaves as expected. My TriggerRefreshValue makes it into my ozwcache without issue:

```
<TriggerRefreshValue Genre="user" Instance="1" Index="512">
  <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
</TriggerRefreshValue>
<TriggerRefreshValue Genre="user" Instance="1" Index="512">
  <RefreshClassValue CommandClass="98" RequestFlags="0" Instance="1" Index="1" />
</TriggerRefreshValue>
```

And notification is correctly sent when the Alarm is received:
```
2020-06-23 08:28:30.145 Info, Node004, Processing (Send) Nonce Request message (Callback ID=0xa5, Expected Reply=0x04)                 2020-06-23 08:28:30.145 Info, Node004, Sending (Send) message (Callback ID=0x02, Expected Reply=0x04) - Nonce_Get(DoorLockCmd_Get) - 0x2020-06-23 08:28:30.155 Detail, Node004,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8                                                2020-06-23 08:28:30.156 Detail, Node004,   ZW_SEND_DATA delivered to Z-Wave stack                                                      2020-06-23 08:28:30.245 Detail, Node004,   Received: 0x01, 0x07, 0x00, 0x13, 0x02, 0x00, 0x00, 0x09, 0xe0                              2020-06-23 08:28:30.246 Detail, Node004,   ZW_SEND_DATA Request with callback ID 0x02 received (expected 0x02)                         2020-06-23 08:28:30.247 Info, Node004, Request RTT 99 Average Request RTT 268                                                          2020-06-23 08:28:30.329 Detail, Node004,   Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x04, 0x0a, 0x98, 0x80, 0xed, 0x75, 0x1e, 0x68, 0xcb2020-06-23 08:28:30.331 Info, Node004, Received SecurityCmd_NonceReport from node 4                                                    2020-06-23 08:28:30.333 Info, Node004, Encrypted Flag is 1                                                                             2020-06-23 08:28:30.334 Info, Node004, Sending (Send) message (Callback ID=0xa7, Expected Reply=0x04) - DoorLockCmd_Get (Node=4): 0x01,2020-06-23 08:28:30.346 Detail, Node004,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8                                                2020-06-23 08:28:30.348 Detail, Node004,   ZW_SEND_DATA delivered to Z-Wave stack                                                      2020-06-23 08:28:30.464 Detail, Node004,   Received: 0x01, 0x07, 0x00, 0x13, 0xa7, 0x00, 0x00, 0x0c, 0x40                              2020-06-23 08:28:30.465 Detail, Node004,   ZW_SEND_DATA Request with callback ID 0xa7 received (expected 0xa7)                         2020-06-23 08:28:30.466 Info, Node004, Request RTT 319 Average Request RTT 293                                                         2020-06-23 08:28:30.467 Detail, Node004,   Expected callbackId was received                                                            2020-06-23 08:28:30.546 Detail, Node004,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x04, 0x02, 0x98, 0x40, 0x2d                        2020-06-23 08:28:30.548 Info, Node004, Received SecurityCmd_NonceGet from node 4                                                       2020-06-23 08:28:30.548 Info, NONCES: 0xfc, 0x81, 0x20, 0x4b, 0x1b, 0x8b, 0x67, 0xce                                                   2020-06-23 08:28:30.548 Info, NONCES: 0x72, 0x8b, 0xa1, 0x77, 0x80, 0xa7, 0xfa, 0xc1                                                   2020-06-23 08:28:30.548 Info, NONCES: 0xe8, 0x4d, 0xb8, 0xaf, 0x3b, 0xcf, 0x07, 0x81                                                   2020-06-23 08:28:30.548 Info, NONCES: 0xb4, 0xc5, 0xbb, 0xe6, 0x72, 0x05, 0x38, 0xf7                                                   2020-06-23 08:28:30.548 Info, NONCES: 0x19, 0x09, 0xf0, 0x03, 0xbb, 0xd0, 0x52, 0xa0                                                   2020-06-23 08:28:30.549 Info, NONCES: 0xe0, 0x8a, 0x23, 0xf1, 0xfc, 0x63, 0xe5, 0x65                                                   2020-06-23 08:28:30.549 Info, NONCES: 0x8a, 0xa6, 0xc7, 0x16, 0xa4, 0x58, 0x37, 0x00                                                   2020-06-23 08:28:30.549 Info, NONCES: 0x81, 0x60, 0xc9, 0x22, 0xba, 0x6f, 0xc2, 0xda                                                   2020-06-23 08:28:30.549 Info, Node004, Sending (Send) message (Callback ID=0x01, Expected Reply=0x04) - Nonce_Report - 0x01, 0x11, 0x002020-06-23 08:28:30.558 Detail, Node004,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8                                                2020-06-23 08:28:30.560 Detail, Node004,   ZW_SEND_DATA delivered to Z-Wave stack                                                      2020-06-23 08:28:30.669 Detail, Node004,   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x0c, 0xe6                              2020-06-23 08:28:30.671 Detail, Node004,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)                         2020-06-23 08:28:30.671 Info, Node004, Request RTT 524 Average Request RTT 408                                                         2020-06-23 08:28:30.758 Detail, Node004,   Received: 0x01, 0x21, 0x00, 0x04, 0x00, 0x04, 0x1b, 0x98, 0x81, 0x7b, 0xbb, 0xa8, 0xc6, 0x882020-06-23 08:28:30.759 Info, Raw: 0x98, 0x81, 0x7b, 0xbb, 0xa8, 0xc6, 0x88, 0xef, 0x8c, 0xbb, 0x2d, 0xb7, 0xa7, 0x17, 0x98, 0xc6, 0xd02020-06-23 08:28:30.760 Detail, Node004, Decrypted Packet: 0x00, 0x62, 0x03, 0x00, 0xff, 0x02 [... cut ...] 4, Removing current message                                                                      2020-06-23 08:28:30.769 Detail, Node004, Notification: ValueChanged CC: COMMAND_CLASS_DOOR_LOCK Instance: 1 Index: 0                   2020-06-23 08:28:30.770 Detail, Node004, Notification: ValueChanged CC: COMMAND_CLASS_DOOR_LOCK Instance: 1 Index: 1  
```

I'll close this issue and work on a PR for the hassio addon. Thanks again for your patience, and sorry for not catching the old version earlier!

closed 2020-06-23 by rtrox