OpenZWave/open-zwave #2267: MCOhome FCU-thermostat

kostas2911, 2020-06-22:
Manufacturer: McoHome Technology Co Ltd
Product Type: 0803
Product ID: 5102
Product Name: MH8-FC
[mco.zip](https://github.com/OpenZWave/open-zwave/files/4812054/mco.zip)

Fishwaldo, 2020-06-24:
1) Please add a MetaData Block and Image file please (https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices#creating-the-metadata-block)
2) there is several old 1.4 settings mixed in with 1.6. (eg,alttypeintepretation). Please consult the above link and https://github.com/OpenZWave/open-zwave/wiki/CommandClass-Compatibility-Flags for the correct way on 1.6.

kostas2911, 2020-06-24:
Is this ok?
[mcofcu.zip](https://github.com/OpenZWave/open-zwave/files/4823936/mcofcu.zip)

Fishwaldo, 2020-07-07:
Not really - But I fixed up a few things and commited it. Thanks!

closed 2020-07-07 by Fishwaldo

kostas2911, 2020-07-07:
Hi,

I think that there is an error on manufacturer_specific.xml. id should be 5102 not 3102

Also on mh8fceu0803.xml without below lines:

<CommandClass id="67" name="COMMAND_CLASS_THERMOSTAT_SETPOINT" version="2" innif="true" base="1" typeInterpretation="B">
	<Instance index="1" />
	<Value type="decimal" genre="user" instance="1" index="1" label="Heating 1" units="C" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0" />
	<Value type="decimal" genre="user" instance="1" index="2" label="Cooling 1" units="C" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0" />

i cannot set heat temperature

kostas2911, 2020-07-08:
[image]
[image]

kostas2911, 2020-08-02:
There is an error on manufacturer_specific.xml. id should be 5102 not 3102