OpenZWave/open-zwave #2269: TimeOut, network basically unusable

cinhcet, 2020-06-23:
Hi,

First of all, thanks a lot for developing this library!

I had a zwave network running for months now without problems. Then I had to restart the controller (a cubietruck, similar to raspberries) and since then, the network is basically unusable, since every device constantly produces Notification: Notification - TimeOut, which leads to a blocked network for maybe 10 seconds every time I send a command.
I have not changed anything in the network.

Now if I take the zwave controller stick, put it into my laptop and run the identical code from there (I placed my laptop within 1m vicinity of the cubietruck), then everything works, no timeouts, all is responsive etc.

I am using openzwave through the nodejs wrapper in my own node for Node-RED (not node-red-contrib-openzwave).

How can that be and how to debug that?

The two log files were created by starting up and then switching node 3 on and off. The _notWorking is from the cubietruck, which has the timeout issues, and the _working one is from my laptop.

The only thing I notice is that for the working one there are much less hop: 1, but more hop: 0 in the log file. Can 1 m make such a difference? (the devices are about 5-7m away from the stick).
Furthermore, there are these "ERROR: ZW_SEND_DATA could not be delivered to Z-Wave stack", which are only in the notWorking one.
But why has that worked before without problems?

[OZW_Log_notWorking.txt](https://github.com/OpenZWave/open-zwave/files/4822247/OZW_Log_notWorking.txt)
[OZW_Log_working.txt](https://github.com/OpenZWave/open-zwave/files/4822248/OZW_Log_working.txt)

Fishwaldo, 2020-06-24:
You could be on the verge of power issues with your cubietruck - thats the only thing I can think off. Try a externally powered hub.

closed 2020-06-24 by Fishwaldo

cinhcet, 2020-06-24:
It is not a power issue. I cannot rule out a hardware fault of course, but why are you closing this if basically the same code on a different machine ones produces "ERROR: ZW_SEND_DATA could not be delivered to Z-Wave stack" and on the other one not?