OpenZWave/open-zwave #2284: IntervalBetweenPolls is ignored

steve28, 2020-07-02:
The `IntervalBetweenPolls` bool in the `options.xml` is being ignored. I currently have this set as so:
```
<Option name="IntervalBetweenPolls" value="true" />
```
Which should poll a single device from the polling list every `PollInterval`.  That is, with the default value of `PollInterval` of 30000 (30 s).  If I have 4 devices set to poll, it should take 2 min to get through all 4 devices (1 every 30s * 4 devices).

Instead, it's still polling every device equally spaced within `PollInterval` (which is the behavior that is intended with `IntervalBetweenPolls` set to `false`).  So in the above case it's polling a device every 30/4 = 7.5s.

Versions of stuff:
    "OpenZWave_Version": "1.6.1149",
    "OZWDaemon_Version": "0.1.143",
    "QTOpenZWave_Version": "1.2.0",
    "QT_Version": "5.12.5",
    "getControllerLibraryVersion": "Z-Wave 4.05"

chilicheech, 2020-08-12:
I'm having an issue related to polling where polling seems to be set up in the logs but the polling never actually happens.

broyuken, 2020-10-01:
Is there any movement being made on this? Is there a way to modify the poll interval to shorter than 30 seconds?

broyuken, 2020-12-07:
Anything we can do to help move this issue forward?

steve28, 2020-12-07:
Aside from fixing it ourselves (which I do not have the capacity to do), no.